z88dk/z88dk #854: (classic) Add Laser 350/500/700

suborb, 2018-07-18:
Just gathering information:

* Z80 @3.694700 MHz
* 32k ROM
* 16k banking model in all 4 segments
* Banked graphics memory (unknown chip, 6845ish?)
* Cartridges
* Memory mapped io
* 16/64/128k memory available
* MS Basic

Probable available machine specific features:

* [ ] Cartridge support
* [ ] Native console
* [ ] Generic console (text)
* [ ] Generic console (hires)
* [ ] 1 bit sound
* [ ] Graphics (hires)
* [ ] Graphics (lores via gencon)?
* [ ] Inkey support

suborb, 2018-07-19:
I can't get the cartridge to be loaded up by the Mame framework - the m_cartslot member is null - so this port is blocked for the moment.

nippur72, 2018-08-31:
I have a real hardware (Laser 500) and am interested in implementing this platform. But I am very new to z88dk (I've just installed it for the first time), so I don't know how I can effectively contribute.

Unfortunately the MAME emulation is very poor, keyboard support is partial and WAV files from/to real hardware need to be manually speed-corrected.

suborb, 2018-08-31:
Hi, sorry I ended up being distracted and never thanked you for your post on atariage.

I started work on this target a while back - with the aim of building a ROM cartridge - to at least get something up and running. It's sitting in the vz700 branch. Unfortunately I ran into Mame issues and I couldn't get the cartridge to start.

The main stumbling block at the moment is probably the file format - if we can write a cassette file (in any format) then we can probably get a fair amount of stuff up and running even with Mame not being an accurate emulator - certainly we can test out most of the things listed above. 

So if you have any information on the cassette format then that would really useful.

nippur72, 2018-08-31:
Ok, I will investigate the cassette format. The MAME sources contain all the details regarding the raw waveform (duration of the pulses and so on). Hopefully there is also something about the high-level format. As last resort, there is always the BASIC ROM 😱

nippur72, 2018-09-01:
Ok, I've just finished to reverse-engineer the cassette format!

Good news, it's very similar to the VZ200's one, only small changes are needed.

```
LOW LEVEL FORMAT
================

The low level format of the pulses is exactly the same as in VZ200:
- Bit 0: a sequence of HLHHLL (short cycle + long cyle)
- Bit 1: a sequence of HLHLHL (3 short cycles)
- Polarity doesn't seem to affect the loading (e.g. swapping H with L).
- Bytes are turned into bits by sending most significant bits first.

H and L last about 12 samples which corresponds to what's called "bip" 
in z88dk/src/appmake/vz.c. A "bop" is about twice, it's set to 25 in vz.c.

Due to MESS inaccuracy, WAV files feeded into MESS need to be increased in speed by ~15%. 
Similarly, when loading on a real machine a file produced in MESS, a slow down of ~15% is needed.
That can be done in "Audacity": Menu->Effects->Change speed. 

HIGH LEVEL FORMAT
=================

The high level format changes slightly from VZ200. (*) indicate a change.

header:    0x80 repeated 128 times + 0xfe repeated 5 times

file-type: 0xf0 (type "T" aka basic program) or 
           0xf1 (type "B" binary file, saved with MON)           
           (*) VZ200 apparently does not use the code 0xf0,
               resulting in basic programs being seen as binary ("B")

file-name: bytes of the string + 0x00 (terminator)

header2: 0x00 + 0x80 repeated ten times + 0xff

(*) header2 is missing in VZ200, this is (I guess) the cause
    why a VZ200 program can't be loaded on a L500. It raises a "Loading error"

start-address: 2 bytes, usually: 0x95 0x89 (basic programs start at 0x8995)
end-address:   2 bytes e.g. 0xa3 0x89 for 0x89a3
data:          the actual bytes to be loaded
checksum:      2 bytes 
               (*) I have not checked, but I guess it's the same as in VZ200
terminator:    0x00 repeated long enough
```
Let me know if there's something which is not clear.

nippur72, 2018-09-01:
and this is a sort of stub file in order to make programs launch themselves.

Starting address of the generated z88dk code must be 0x89A3

```c
// 2018 A=&H89A3:CALL A

unsigned char stub[14] = { 
   0xA3, 0x89,                         // pointer to next basic line in memory (&H89A3)
   0xE2, 0x07,                         // 2018
   0x41, 0xF0, 0x0C, 0xA3, 0x89, 0x3A, // A=&H89A3
   0xB6,                               // :
   0x20, 0x41,                         // CALL A
   0x00                                // line terminator
};
```
(note that `CALL` command only accepts variables as parameter)

suborb, 2018-09-02:
Thanks for this info - I've updated the branch with:

- a ROM printer using the call
- The default CRT is now RAM with the BASIC stub listed above

The next step should be taking a copy of the vz200 appmake code, making it match your investigation above and actually creating a file that can be tested.

suborb, 2018-09-02:
There's a first version of appmake, if you take the branch, build you should be able to:

    zcc +vz700 world.c -create-app --Cz--audio

Then a wav file is created that may work.

nippur72, 2018-09-02:
I'm having troubles building it, it compiles everything except `appmake`:
```
Errore	LNK2001	simbolo esterno _einstein_exec non risolto	appmake	\GitHub\z88dk\win32\appmake\appmake.obj	1	
Errore	LNK2001	simbolo esterno _einstein_options non risolto	appmake	\GitHub\z88dk\win32\appmake\appmake.obj	1	
Errore	LNK2001	simbolo esterno _mz2500_exec non risolto	appmake	\GitHub\z88dk\win32\appmake\appmake.obj	1	
Errore	LNK2001	simbolo esterno _mz2500_options non risolto	appmake	\GitHub\z88dk\win32\appmake\appmake.obj	1	
```
there are those unresolved externals.

I'm building it (vz700 branch) on Windows 10, VS2017 and GNU make. Is there something else needed?

suborb, 2018-09-02:
Try again - Paulo updated the visual studio build in master and it needed a merge.

nippur72, 2018-09-02:
It goes a lot further but still does not build:
```
--- Building TS2068 (Spectrum clone) Library ---                                                             
make[1]: Nothing to be done for 'all'.                                                                       
make[1]: Nothing to be done for 'all'.                                                                       
make[2]: Nothing to be done for 'all'.                                                                       
TARGET=zx TYPE=z80 z80asm -d -IC:\Users\Nino\Desktop\USB\GitHub\z88dk\lib\config\/../ -DSTANDARDESCAPECHARS -
DFORzx -DFORts2068 -DA80COL -xC:/Users/Nino/Desktop/USB/GitHub/z88dk/libsrc//ts2068_clib @./ts2068.lst       
Error at file 'spectrum/basic/zx_getfloat.asm' line 21: cannot read file 'target/ts2068/def/ts2068fp.def'    
Error at file 'spectrum/basic/zx_getfloat.asm' line 52: symbol 'ZXFP_BEGIN_CALC' not defined                 
Error at file 'spectrum/basic/zx_getfloat.asm' line 52: expected constant expression                         
Error at file 'spectrum/basic/zx_getfloat.asm' line 46: symbol 'ZXFP_STK_STORE' not defined                  
Error at file 'spectrum/basic/zx_getfloat.asm' line 53: symbol 'ZXFP_RE_STACK' not defined                   
Error at file 'spectrum/basic/zx_getfloat.asm' line 54: symbol 'ZXFP_END_CALC' not defined                   
[...omitted...]
Error at file 'spectrum/zxmmc/mmc_write_command.asm' line 24: symbol 'SPI_PORT' not defined                  
Error at file 'spectrum/zxmmc/mmc_write_command.asm' line 28: symbol 'SPI_PORT' not defined                  
Error at file 'spectrum/zxmmc/mmc_write_command.asm' line 32: symbol 'SPI_PORT' not defined                  
96 errors occurred during assembly                                                                           
make: *** [makefile:251: ts2068_clib.lib] Error 1                                                            
make: Leaving directory 'C:/Users/Nino/Desktop/USB/GitHub/z88dk/libsrc'                                      
                                                                                                             
Nino@SIXTYFOUR C:\Users\Nino\Desktop\USB\GitHub\z88dk                                                        
$ if 2 NEQ 0 goto :error                                                                                     
Build script failed                                                                                          ```

suborb, 2018-09-02:
I'm not sure what's happening there, it builds elsewhere ok. I've had success building the libraries on windows with the following method: https://github.com/z88dk/z88dk/wiki/libsrc#rebuilding-the-z88dk-libraries-on-windows

However, see you can build vz700_clib.lib: make vz700_clib.lib - that's the important one here!

suborb, 2018-09-03:
It's progress! I'm impressed that it got that far! Getting the output file onto a new device/emulator is usually the painful bit.

The printf() will be large because it's pulling in all the format conversions - there are ways to reduce the number included and hence the size of the binary - but lets not worry about that at the moment.

Nothing uses PRINTSTR directly, CHROUT is hooked directly into fputc_cons() - so that will be the shortest.

nippur72, 2018-09-03:
The loading issue is due to the wrong crt0 stub being two bytes shorter than it should be. We can fix that, but there's another issue. As of now, the machine code is placed at the end of the Basic program causing it to get overwritten by Basic itself when the command `RUN` is launched. 

We need to insert the whole machine language code into a big fake Basic line, so it is seen as valid Basic code and thus left untouched. In order to do that we need to have a "parametric" stub, passing to it the size of the program to be put into a basic line. Is that possible?

There is always the option of pure binary files without stubs, they can be launched manually with:
```
MON
8995G
```

nippur72, 2018-09-04:
now with #942 it does load, but the generated code doesn't look right to me (it writes in the ROM area) and indeed it does not work. But I don't know how to proceed from here.

suborb, 2018-09-04:
The code has been ORG'd to 0 (implicitly because there's no org statement!). I think the start of the file should be:

```
        defc CRT_ORG_CODE = 0x8995

        INCLUDE "crt/classic/crt_rules.inc"

        org     CRT_ORG_CODE
        ; BASIC header for the vz
basicstart:
   defb 0xFF, 0xFF                           ; pointer to next basic line in memory
   defb 0xE2, 0x07                           ; 2018
   defb 0x41, 0xF0, 0x0C                     ; A=&H
   defw start                                ;
   defb 0x3A                                 ; :
   defb 0xB6, 0x20, 0x41                     ; CALL A
   defb 0x00                                 ; basic line terminator

start:
```

nippur72, 2018-09-04:
Fixed the `org` statement, now it works!

And here is the first program ever made for this machine: 
[image]

nippur72, 2018-09-04:
There is a minor problem with `\n` and/or `\r`, (tried all combs): it doesn't make the new line, it just goes at the beginning of the current line. But I guess we can fix that.

And BTW, I've used `BASCK` on this machine's ROM and it found an extended MS Basic 8080/Z80 with lot of entry points 😄. I've already found a `getk()` candidate.

So what is next step?

suborb, 2018-09-04:
That's brilliant. Congratulations!

There may well be another separate function that does a line feed which we'll have to call.

Next step is the fgetc_cons() + getk() implementation. Then we have a "functioning target".

It's pretty bare bones though. I like to get ports up with the following:

- [x] ROM console out
- [ ] ROM keyboard reading
- [ ] Generic console (a cross target VT52 engine)
- [ ] Lores (block) graphics
- [ ] Hires graphics (if available)
- [ ] Sound support (if available)
- [ ] Inkey input (reading the ports directly and supporting multiple concurrent keypresses)

Having that functionality allows games and other software to be ported over easily.

Oh, and running in some kind of emulator - I'm guessing we have to tweak the .wav file in someway to get it work in mame? Once it's working in mame then I can properly help get the machine up and running.

I started the generic console a while back (it's in libsrc/stdio/vz700/generic_console.asm). The console basically pokes the screen addresses with the character codes and attribute values - you can switch it in and try it out with `-pragma-redirect:fputc_cons=fputc_cons_generic` and then compile examples/console/ansivt52.c to test it out. I'm not expecting it to work first time!

nippur72, 2018-09-05:
if you duplicate the structure from vz200, I can modify the files and adapt them to vz700 (as I am not yet acquainted with z88dk code organization).

As for runnig WAVs on the emulator, we might have a `--fast` flag that would make files 15% smaller/faster so that MESS can load them. I am looking into this issue on the MAME repo too. 

As for the generic console, the machine has two text modes (40/80 cols), should we use the flag in system RAM? or should  the generic console be totally independent?

suborb, 2018-09-05:
I've created placeholders for fgetc_cons() and getk() - getk() is what's needed to be implemented - it should just scan the keyboard and return the ascii value key is pressed (or 0 if nothing).

Re the flag, I've generally not kept the OS up-to-date with the mode that's on screen - usually because we're bypassing it completely to speed things up.

nippur72, 2018-09-06:
great work! Where did you get the constants for `vz700.def` ?

Regarding text in GR modes, the charset ROM is not readable by the CPU, so the 2K font file needs to be included along with the code.

suborb, 2018-09-06:
The constants came from ROM disassembly and interpolation!

Good to know re the font, I normally hunt through the ROMs trying to find the machine one to save a bit a space in the user binary. We've got quite a few fonts available so we can make the VZ700 look like a Spectrum! - https://github.com/z88dk/z88dk/wiki/Classic-GenericConsole

Have you had any luck with keyboard reading?

nippur72, 2018-09-06:
for the fonts, you can use the ROM dump that comes with MAME so there is perfect match. The last 1K is reversed charset that can be omitted and reversed in software. And, just out of curiosity, that ROM dump it is not really an actual dump, Davide Moretti told me that back in the 2000's he grabbed a PAL video frame and encoded the pixels manually.

As regards the keyboard, I've figured out how it works at the low level, but I have not spotted the routine in the ROM (yet). I'm also working on a browser-based emulator as a side project because MESS is too buggy.

suborb, 2018-09-06:
I've added keyboard reading by scanning the IO ports directly so I think we've got enough to do an initial merge.

Before I do that, could you do a sanity test of it on real hardware - just a simple loop as follows would be enough. It should block until a key is pressed. Shift, control, graph won't report anything but the rest should do.

``
#include <stdio.h>
int main()
{
        while ( 1 ) {
                int c = fgetc_cons();
                printf("%d = %c ",c,c);
        }
}
```

nippur72, 2018-09-08:
sorry for the delay, I had to wait for a nightly build because it no longer builds on my Windows machine.

I tried the above program on the real hardware and it does NOT work; it simply hangs. I'm investigating the issue, sadly MESS is of little use because the keyboard is not accurately implemented. 

There is something weird about the keyboard when reading rows from A to D, indeed I have issues in my own emulator ([laser500emu](https://nippur72.github.io/laser500emu/)) and I suspect that MESS has the same issue.

P.S. is it too late to change `vz700` into something else ? That's because commercially there never was a "VZ700".

nippur72, 2018-09-08:
I figured what was wrong: the keyboard is not mapped at 0x4000, it seems that only a smaller portion is mapped. The ROM uses 0x6B00 to 0x6FFF (directly from my emulator, hooking I/O reads). A similar design is the VZ200s. 

So row 1 is 0x6ffe, row 9 is 0x6f7f (moving the "0" bit from right to left).
The other rows are: 
```
6bff (\,[,],ins,del), 
6aff (caps,delline,home,up,left,right,down)
69ff (f10,f9,f8,f7,f6,f5)
68ff (f4,f3,f2,f1)
```
The `Reset` key seems not to be mapped, I wonder whether it's connected to a hardware interrupt in the CPU. (When pressed you almost always get a Ready prompt).

Anyway I am still unable to feed extended keys in the emulator, it looks like setting the bits isn't enough. Perhaps there are precise timings to adhere. This is an hell of keyboard, even on the real hardware typing is unresponsive, if you type too quickly you miss keystrokes!

Re on the +vz700 name change: I'm for `l500` or `laser500` dropping 700 altogether because the 500 was much more common back in the days and it's the machine you still find today. 

Re on compiling on windows: previously I was able to compile with `make vz700_clib.lib` but now it doesn't find the `.def` files even if they are there. I cloned a fresh repo.

suborb, 2018-09-09:
Thank you for testing. It's a bit worrying that some don't respond - the scanning is really simple so if one key in a row works, the other one should.

Maybe it's time to find the ROM routine and lift it.

If you run the program above: https://github.com/z88dk/z88dk/issues/854#issuecomment-419641649 (change the $40fe to $6bfe of course) and press each key do you see a change in the output value for the main keyboard rows?

How do you load programs into your emulator?

nippur72, 2018-09-09:
I tried to find the keyboard ROM routine but failed, I'm still too new to the Z80.

Yes, trying the above program there is change for all the main keys; what doesn't work is `fgetc_cons()` (half of the keys do not work).

BTW, when running the program I seldom got random numbers... so I wrapped the routine around `DI` and `EI` thinking that the interrupt might do some `OUT (41h), 1` in between. That worked but left me puzzled... does that makes sense at all? 

To load a program in my emulator drag & drop a `.bin` file (not wav) and then RUN. From the JavaScript console there are also commands for getting a program in and out from the emulator. They are also saved in `window.localStorage` so that they can be retrieved later. The emu is very incomplete, it only supports `TEXT 40` for now. But it is a lot of fun!

suborb, 2018-09-09:
I think the keyboard routine is around $064b 

Yes, I had exactly the same problem with the keyboard reading - intermittent random values as if we'd been paged out. Which is theoretically possible, however originally we were keeping the system variable updated so after the interrupt we should have had the paging restored.

I do the same thing with the video memory - page into the second segment - but I haven't seen any visual oddities so I'm at a bit of a loss as to what's happening.

nippur72, 2018-09-10:
I managed to emulate the keyboard correctly, I did a full dump of the I/O space (4000-7FFF) on the real machine here's whats happening:
- lowest rows do respond from 6800 to 6fff
- extended row respond only to 68, 69, 6a, 6b
the result from the two kind of rows is joined via OR 

My guess is that they used they keyboard design from the vz200 and then hacked it to add extend keys. Indeed the inital model 350 does not have the extended rows (it's a 500 in a 310 case).

nippur72, 2018-09-12:
yes they are mirrored, bit 0 is leftmost. In GR3 (2 pixels x byte) left pixel is lower nibble (& 0x0F), right is upper. 

My emulator now supports all video modes, you can quickly do tests by opening the browser and dragging bin files there. 

I've also reverse engineered the joystick using an old Basic type-in I wrote back in 1986:
```javascript
/*
Joystick 1:

up    = ~(inp(&h2b) &  1)
down  = ~(inp(&h2b) &  2)
left  = ~(inp(&h2b) &  4)
right = ~(inp(&h2b) &  8)
fire  = ~(inp(&h2b) & 16)
fire2 = ~(inp(&h27) & 16)

Remarks: bits are with negated logic, 0 when button in pressed. 
         Direction and first fire are on port &h2b, the
         other fire button is on port &h27.
         Basic provides the JOY(n) function,
         JOY(0) = direction 1,2,3,4,5,6,8 starting from up and then clockwise
         JOY(1) = 1 fire1 button 
         JOY(2) = 1 fire2 button
*/
```

suborb, 2018-09-12:
Cool. I can deal with that mirroring easily enough - just wanted to check that the emulator was correct before adding the code. 

Thanks for the joystick hint. Might as well add that as well when I get the chance.

suborb, 2018-09-12:
Getting there, just hires graphics and sound to go.

@nippur72 I presume toggling the state of speaker A generates a square wave? What's speaker B for?

Your emulator is a great way to test, having a quick turnaround makes my life so much easier. Have you got any plans to add a debugger?

nippur72, 2018-09-13:
from the vz200 manual (presuming it's the same in Laser500):
```
The speaker is driven in push-pull fashion by bits 0 and 5. To make the speaker sound a note, the software
should toggle bits 0 and 5 alternately at the required rate. ie when bit 0 is a logic "1", bit 5 should be logic
"1" and vice-versa. Note that when this is done the software should not alter the other bits of the latch.
```

As for the debugger, most likely I'll add one, currently this emulator is my toy and it's likely I'll play with it for a while 😄 .

P.S. I've found how to hook to the raster NMI interrupt, it's very easy:
```c
int install_interrupt() {
   #asm
      di
      ld hl, _interrupt  ;
      ld ($8013), hl     ;
      ld a, $c3          ;
      ld ($8012), a      ; 8012: C3 ?? ?? JP interrupt
      ei
   #endasm
}

void interrupt() __z88dk_fastcall __naked {
   #asm   
   ret
   #endasm
}
```

nippur72, 2018-09-15:
re on the speaker: a quick test on the real hardware revealed that only bit 1 of I/O (0x6800) is used for producing sounds.

suborb, 2018-09-16:
Thank you. I've just added 1 bit sound to the laser500 library.

I've tested in Mame and it's making some sound.

nippur72, 2018-09-17:
BTW, my emulator now supports sounds too

suborb, 2018-09-17:
I saw you were working on it. Currently the emulator is broken in Safari:

```
[Error] ReferenceError: Can't find variable: restoreState Global Code (emulator.js:203)
```

suborb, 2018-09-23:
I'm not able to make that routine return anything sensible in terms of KEYASCII

The one that does work is $58f0, though that blocks until a key is pressed.

nippur72, 2018-09-23:
strange, it works for me, see [this live example](https://nippur72.github.io/laser500emu/?load=examples/scankey)

Of course ASCII `0x8030` is valid only if `0x85F0` != 255

nippur72, 2018-09-25:
still can't find a good keyboard routine entry point in the ROM, it's too entangled... jump here then jump there!

I wonder how the Basic does keyboard scanning... I suspect it's a bit flag that it's raised when you "RUN", and the interrupt routine does the rest.