z88dk/z88dk #856: [question] udg at linking time with a .asm file (or any human-readable editable file)?

Fabrizio-Caruso, 2018-07-21:
Hi
User-defined graphics can be done in multiple ways. 
It is nice and useful to have common APIs. This is should be possible on all capable targets.
On targets with udg-data mapped in memory, the most efficient way is probably the linker: you create a binary which includes the UDG data at the right place in memory.
This is what I am doing with the Spectrum but I am doing it the wrong way...
I compile with a .asm file with a following content 
"
SECTION data_user

PUBLIC _udg

_udg:

   BINARY "udg.bin"
"
and a raw .bin file that is supposed to contained default udg.
Then I redefine some of the characters inside my code. This should be avoided.

How can I just link a .asm file (possibly not a raw binary) that I can edit and then just link so that I can udg at zero computational cost?

pauloscustodio, 2018-07-21:
z80asm has a primitive way to define the kind of UDG used by the spectrum. You can write:
```
defb %"--------"
defb %"--####--"
defb %"-#----#-"
defb %"-#----#-"
defb %"-######-"
defb %"-#----#-"
defb %"-#----#-"
defb %"--------"
```

Fabrizio-Caruso, 2018-07-21:
Can't I simply use 0s and 1s?
How do I convert the default font set to a human-readable format?

Fabrizio-Caruso, 2018-07-21:
The final goal for me is two generate all modified font/sprite files for all targets accross multiple devkits. I could even edit the binary if I knew its format but debugging is harder

Fabrizio-Caruso, 2018-07-21:
In my code I am linking the default font set and then I modify it.
I use an extern pointer to unsigned char to point to the font.
If I knew the address of the font set, I could simply modify it without loading it.

pauloscustodio, 2018-07-21:
> Can't I simply use 0s and 1s?

Yes, but less readable, IMHO
```
defb %00000000
defb %00111100
defb %01000010
defb %01000010
defb %01111110
defb %01000010
defb %01000010
defb %00000000
```

Fabrizio-Caruso, 2018-07-21:
@pauloscustodio  how can I get the Spectrum default fonts in a human-readable format?

What is the start address of the fonts for the Spectrum?

pauloscustodio, 2018-07-21:
The start address is at $3D00, and the human readable format is in the ROM disassembly, for example https://skoolkid.github.io/rom/asm/3D00.html

Fabrizio-Caruso, 2018-07-21:
@pauloscustodio that address is the rom address, writing on that address won't produce user-defined graphics.

I need to understand how to get user-defined graphics by :
- either only linking a font binary with all fonts and no need for (run-time) code
- or only modifying the default fonts 

I don't fully understand and like what I am doing, i.e., 
1. link the binary file that contains the default font characters:
```
SECTION data_user
PUBLIC _udg
_udg:
   BINARY "udg.bin"
```
where udg.bin contains "somehow" the default characters; 
2. and then modify some of them with:
```
	for (i = 0; i < sizeof(redefine_map) / sizeof(*redefine_map); ++i)
	{
	   memcpy(udg + (redefine_map[i].ascii - 32)*8, redefine_map[i].bitmap, 8);
	}
```

What my code does in point 2 is clear: it is modifying the content of the memory area at udg + some offset.

Why do I need to link udg.bin? Couldn't I simply write on the right locations?

Fabrizio-Caruso, 2018-07-21:
I got the main idea (more  or less)!

closed 2018-07-21 by Fabrizio-Caruso

pauloscustodio, 2018-07-21:
The spectrum UDGs are at an address pointed by the system variable UDG (23675) and contain initially a copy of the letters 'A' through 'U'.

Fabrizio-Caruso, 2018-08-10:
@pauloscustodio 
Let us say I want to take one of the 8x8 font sets that come with Z88DK in .bin format, and I want to change just a characters? 
How can I do this? 
I guess I need to reconstruct the .bin into a .asm with all 256 characters?

Fabrizio-Caruso, 2018-08-11:
@pauloscustodio THANKS a lot!
Having an easily editable file, can allow me to modify single characters!
How can I convert the .asm into a usable .bin? I guess running the compiler should be enough. Am I right?

pauloscustodio, 2018-08-11:
Just assemble it:
z80asm -b font.asm