z88dk/z88dk #858: [trs80][real hardware][model III] Reports about bugs on real Model III but OK for trs80gp 1.8

Fabrizio-Caruso, 2018-07-22:
In 
https://www.facebook.com/fabrizio.caruso.56/posts/10215941571836654?comment_id=10215980934660700&reply_comment_id=10216681633897743&notif_id=1532173111996400&notif_t=feed_comment

I have got a bug report about my game: it works fine on trs80gp 1.8 but it fails on real hardware (Model III).

Has the trs80 already been tested on some real hardware?

I compile it with:
```
trs80:
	$(Z88DK_PATH)$(MYZ88DK) +trs80 -lndos \
	-lm -create-app \
	-D__TRS80__ \
	-DFULL_GAME -DBETWEEN_LEVEL -DEND_SCREEN \
	-DCONIO_VT52 \
	-DCONIO -DNO_INIT_GRAPHICS \
	-o$(BUILD_PATH)/FULL_trs80.bin \
	...
```

suborb, 2018-07-22:
As far as I know we’ve had no reports either way. 

Finding out why things don’t work on the real hardware means that we have to start from the beginning. 

- does a file load?
- can we print to the console using rom calls
- do extra features work

Etc etc

If the file does load then problems are usually stack/memory usage etc

Fabrizio-Caruso, 2018-07-22:
Timothy (see FB link) says it loads and starts but then some arrow characters are displayed. I have askef him more details on the hardware. It is a Model III, which is compatible, yet not fully to the Model I.

Fabrizio-Caruso, 2018-07-22:
Timothy reports the same failure at the very same point. Under trs32 emulator
http://www.trs-80emulators.com/trs32/

Fabrizio-Caruso, 2018-07-22:
Timothy has also made some videos that show the issue at the FB post (see link above)

I have not been able to reproduce the same issue on the trs32 emulator.

Fabrizio-Caruso, 2018-07-22:
Videos about the issue:
https://www.facebook.com/halloran.tim/videos/2202483339762075/
https://www.facebook.com/halloran.tim/videos/2202484969761912/

showing the in-game issue: you get an ugly "Cass?" prompt after the game is started.

The game fails exactly at the same place on the trs32 emulator
(tested with config as: Moderl III Level II with no disk)

So, we should use trs32 as an emulator reference.
[image]

The very same game works fine on trs80gp 1.8 emulator.

suborb, 2018-07-22:
Can you point me to ROMS? I already had that emulator downloaded but never actually got it working.

Anyway, it looks like quite a lot of it is working well - the console is displaying, the keyboard is reading so that's really promising.

My best guess is that the stack is colliding with the bss area and that's why the reset is happening. It looks like that emulator doesn't have a debugger which is going to make figuring out what's gone wrong a little tricky.

Fabrizio-Caruso, 2018-07-22:
[roms.zip](https://github.com/z88dk/z88dk/files/2217225/roms.zip)

suborb, 2018-07-22:
I can't reproduce it with the .cmd version and for some reason I can't get the .cas version to load.

To disprove my stack hunch, it's probably worth printing the value of sp on startup:

    unsigned int getsp() {
    #asm
        ld hl,0
        add hl,sp
    #endasm
    }

    ...
    printf("%04x\n",getsp());

We don't change the stack pointer so we're inheriting whatever we've been given.

Fabrizio-Caruso, 2018-07-22:
In order to reproduce it you need to:
1. Use the rom above, uncompress them and havebthe trs32 emu point to their folder
2. You need to select Model III Level II
3. Disable disk
4. At Cass? Press L
5. At memory press enter
6. Type SYSTEM and enter
7. Attach the .cas 
8. Show the emulated cassette player 
9.  Type ../BUI and enter (that is the name of file by mistake because I used -o with a path)
10. Click on play
11. Wait some time (counter beyond 300 for the full game)
12. Type / and enter to start the game

suborb, 2018-07-22:
Thank you, I was missing the L in the answer for Cass?

Yes, it's the stack pointer, when loading a .cmd, the entry sp is somewhere around 0xffb5, for a .cas is around 0x4285.

I'd suggest compiling with -pragma-define:REGISTER_SP=0xffb5 and that should fix the .cas compiles - I've just verified that .cas files work correctly with that option.

closed 2018-07-22 by suborb

suborb, 2018-07-22:
Yes, the .cmd seems to have been working fine and it was just the .cas that failed. I don't know the full memory map but I think there's very limited stack space available with the default sp for a .cas load. So I suspect the stack (going downwards) may have overlaid the screen and then some screen printing overwrote a stack address and the return from a function ended up jumping into space (and hence resetting the machine).

Fabrizio-Caruso, 2018-07-22:
How do you load the .cmd files with trs32?

suborb, 2018-07-22:
1. Get to the BASIC prompt
2. File->RUN/CMD file

In trs80gp you don't need to get through to the basic prompt before loading the .cmd file, but in trs32 it is needed.

Fabrizio-Caruso, 2018-07-22:
@suborb Thanks a lot for your support!
I manage to run the .cmd file. It runs fine except for the blinking cursor that remains there.
I do not see it with the .cas file.