z88dk/z88dk #864: z80asm: stringify numeric constant in defm

aralbrec, 2018-08-03:
I've come across the need to stringify a constant in a text string.

```
defc CORE_VERSION = 14010
defm "Requires Core v", CORE_VERSION, ' '+0x80
```

I'd like `CORE_VERSION` to be present as text instead of as numeric bytes.  A leading `#` as in the c pre-processor may be a good method:

```
defm "Requires Core v", #CORE_VERSION, ' '+0x80
```

Another way I can do this is with m4 but being able to do it in the assembler would be more convenient.

zx70, 2018-08-05:
encoding the number with an expression on defb for every single digit is ..uhm.. less practical I suppose :)

pauloscustodio, 2018-08-05:
OK. I'll put this in the preprocessor where I am working at the moment.

The hash is also used as prefix for hexadecimal numbers. I can check if the following word is a valid hexadecimal number, and stringify otherwise.

pauloscustodio, 2019-07-19:
Commit e0d92ea03706a7c8466019742e6218f6d58fe6a2 has a partial fix. It will work if the number is defined as a macro (known at preprocessing time) but not if the constant is a DEFC (known at link time).
I.e. one has to write:
```
#define CORE_VERSION 14010
defm "Requires Core v", # CORE_VERSION, ' '+0x80
```
This commit is still in the z80asm_macros branch.

pauloscustodio, 2021-12-28:
With the newly defined macro capability, this can be written as:
```
		defl CORE_VERSION = 14010
		defm "Requires Core v"
		reptc char, CORE_VERSION
		  defb char
		endr
		defm ' '+0x80
```

See macro documentation at https://github.com/z88dk/z88dk/wiki/Tool---z80asm---preprocessor

commit e0d92ea0 (2019-07-19): Fix #864: Add string-ize operator '#'

Encloses the next word or number in double quotes. If the word is a macro, it is expanded before quoting.
A space is required between the hash and the word.
Removed constant expression operator from z80asm (#) - no longer required and collides with new string-ize operator.
Fixed the libraries still using '#' to start expressions with parenthesis, to differenciate from indirect addressing - replaced by a '+' sign.
commit 28b10d90 (2021-12-29): Fix #864 with macros implementation
closed 2021-12-28 by pauloscustodio