z88dk/z88dk #866: No PSG support for Otrona Attache

ianfinder, 2018-08-07:
The Attache has a PSG chip, and it would be nice to have the set and get PSG register library commands implemented for it. I can provide a code sample of how the PSG on the otrona is used if necessary.

zx70, 2018-08-08:
Yes please, this is the information I can gather from the BIOS.

SPIOA	=	0F9H	;PIO PORT A COMMAND    (possibly  $CFF9 rather than simply 'F9')

DPIOA	=	0F8H	;PIO PORT A DATA
			;  A0-7 = LATCH DATA OUT
			;	L0 = MOTOR ON
			;	L1 = GRAPHICS ENABLE
			;	L2 = EPROM ENABLE
			;	L3 7 DISPLAY BRTNS.
			;  A0-7 = 8910 DATA I/O
                        ;  A0-3 = 5832 D0-3 I/O
			;  A4-7 = 5832 A0-3 OUT
			;  A0-3 = 5101 D0-3 I/O
			;  A4-7 = 5101 A0-3 OUT

zx70, 2018-08-08:
another improvement (a bit tricky) could be to have the graphics driven directly, at the moment the BIOS calls are used.

zx70, 2018-08-09:
to clarify: "yes please"  =  sample code is appreciated  :)

ianfinder, 2018-08-09:
I'm a little too busy at work to do this properly and check it in, but I took a first stab and these routines seem to work.

The otrona does some weird thing where you have to strobe a pin on DPIOB to latch in the stuff you're sending the PSG, as far as I can tell.

I can provide more detail later. Do you have an otrona to test with? I haven't written a get_ yet.

This is just a nasty, crude first step but it seems to work.

[set_psg_callee.txt](https://github.com/z88dk/z88dk/files/2275926/set_psg_callee.txt)
[psg_init.txt](https://github.com/z88dk/z88dk/files/2275927/psg_init.txt)

zx70, 2018-08-10:
(Very) sadly I don't have the real computer, but I fixed the MAME emulation which now produces an acceptable text output.   I have no idea on whether the sound is emulated correctly, time to try..

zx70, 2018-08-14:
Ok, I hooked your routines to the library build process, a new file (attache.lib) will be created which, in the future, could be improved with more target specific routines (e.g. to access to the CMOS RAM).

closed 2018-08-14 by zx70

zx70, 2018-08-14:
Test on MAME succeeded, psgtest plays the melody  :)
The build steps I followed:
zcc +cpm -create-app -lattache -lm -DNODELAY psgtest.c
appmake +cpm -f ATTACHE -b a.com
bin2imd  a.raw a.imd /2 DM=5 N=40 SS=512 SM=1-10
mame-old attache -flop1 oattache.td0 -flop2 A.IMD
...then after booting I moved to drive B and launched 'A'

I'm using an old version of MAME because I miss the latest BIOS version, but I'm sure there won't be problems with the new ones.