z88dk/z88dk #867: [question][udg] Example with portable multi-target code to handle UDG?

Fabrizio-Caruso, 2018-08-10:
I would like to use user-defined graphics on multiple targets by possibly using the same code.

I know this is now possible on multiple targets.

How? Where can I find a simple example to redefine, say, the '*' character?

suborb, 2018-08-11:
[I've been away so just catching up]

UDGs are character code 128 onwards, individual characters aren't redefined, but the entire set is switched in:

     void *param = &udgs;
     console_ioctl(IOCTL_GENCON_SET_UDGS, &param);

Where udgs in my example is defined as:

    unsigned char udgs[] = {
        0xaa ....
    };

Or you use Paulo's tool, link in the resulting .asm file and reference with:

    extern unsigned char udgs[];

The same code will work for all targets that support UDGs.

Fabrizio-Caruso, 2018-08-11:
@suborb Paulo's tool may be a solution for my use-case (i.e., modify just 1 or few characters).

I don't understand how to use console_ioctl if I just want to redefine just some characters and keep all the others unmodified.

Fabrizio-Caruso, 2018-08-11:
@suborb let us say I want to modify character 42 ('*') and keep all other characters as they are in a given font-set. How can I modify it easily with console_ioctl?

Fabrizio-Caruso, 2018-08-11:
@suborb maybe I may use a font-set and then POKE into udgs to modify just one character at run-time as I do with the Spectrum. I guess this would only work on targets that use ram and not rom.

suborb, 2018-08-11:
It's not possible just to modify a single character. On several targets we use the ROM font by default. All the targets listed here: https://github.com/z88dk/z88dk/issues/732#issuecomment-387026960 support UDGs in the same way detailed above.

Fabrizio-Caruso, 2018-08-11:
@suborb on the Spectrum I do:
extern char udg[];
I link one of the font-sets
and then I modify the desired characters by writing (memcpy or POKE) at the right offsets starting from udg.

https://github.com/Fabrizio-Caruso/CROSS-CHASE/blob/master/src/cross_lib/display/init_graphics/z88dk/spectrum/spectrum_redefined_chars_init_graphics.c

https://github.com/Fabrizio-Caruso/CROSS-CHASE/blob/master/src/cross_lib/display/redefine_characters/udg_map.h

I guess I could apply this solution to other ram-based systems.

Fabrizio-Caruso, 2018-08-11:
@suborb the cleanest and most efficient solution is the one based on Paolo's tool because the characters are modified at compilation/linking time with zero code.

Fabrizio-Caruso, 2018-08-12:
I am trying to get UDGs on the vg5k target.
By adding the section and declaration above in the .asm file generated by Paolo's tool I can compile and link but I see no udgs.

In my code I do:
```
#include <graphics.h>
...
#include <sys/ioctl.h>
extern unsigned char udgs[];
void INIT_GRAPHICS(void)
{
	void *param = &udgs;
	console_ioctl(IOCTL_GENCON_SET_UDGS, &param);
}
```

I compile with:
```
vg5k_exp_16k:
	$(Z88DK_PATH)$(MYZ88DK) +vg5k \
	$(SCCZ80_OPTS) \
	-DSOUNDS -vn -DFULL_GAME -D__VG5K__ -DBETWEEN_LEVEL -DEND_SCREEN -DASM_DISPLAY \
	-lndos -create-app -o $(BUILD_PATH)/FULL_vg5k_exp_16k.prg \
	-DCONIO \
	$(SOURCE_PATH)/end_screen.c \
	$(FULL_FILES) \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	$(CROSS_LIB_PATH)/display/graphics_data/z88dk/xchase.asm
	...
```

Fabrizio-Caruso, 2018-08-12:
I am not getting udgs on the SVI target, either.

@suborb @pauloscustodio What am I doing wrong?

suborb, 2018-08-12:
It looks like you're defining the entire font, in which case you need to change the font, not the udgs. You can either do this with:

	console_ioctl(IOCTL_GENCON_SET_FONT32, &param);

Or do it on the command line: `-pragma-redirect:CRT_FONT=_udgs`

Fabrizio-Caruso, 2018-08-12:
@suborb  @pauloscustodio 
I had also tried IOCTL_GENCON_SET_FONT32 with no result. The '*' still looks the same.
What should I do to display the modified '*' character?

I am using conio... Is it possible to use conio and udgs?
Should I use the generic display instead?

What is the difference between IOCTL_GENCON_SET_FONT32  and IOCTL_GENCON_SET_UDGS?

I have also tried:
```
#include <graphics.h>
#include "../../../display_macros.h"
#include <sys/ioctl.h>
extern unsigned char udgs[];
void INIT_GRAPHICS(void)
{
	void *param = &udgs;
	console_ioctl(IOCTL_GENCON_SET_FONT32, &param);
}
```

Fabrizio-Caruso, 2018-08-12:
Adding  -pragma-redirect:CRT_FONT=_udgs does not help, either...
```
svi_328:
	$(Z88DK_PATH)$(MYZ88DK) +svi $(SCCZ80_OPTS) \
	-clib=ansi -pragma-define:ansicolumns=32 -pragma-redirect:CRT_FONT=_udgs -vn -lndos \
	-DSOUNDS \
	-DFULL_GAME -D__SVI__ -DBETWEEN_LEVEL -DEND_SCREEN \
	-DCONIO \
	-create-app \
	$(CROSS_LIB_PATH)/sound/z88dk/psg/psg_sounds.c  \
	$(CROSS_LIB_PATH)/display/graphics_data/z88dk/xchase.asm \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	$(SOURCE_PATH)/end_screen.c \
	$(FULL_FILES)
```

suborb, 2018-08-12:
VG5k doesn't support changing the font, so only UDGs are permitted.

```
#include <stdio.h>
#include <sys/ioctl.h>

extern unsigned char udgs[];

int main() {
        void  *param = &udgs;

        console_ioctl(IOCTL_GENCON_SET_UDGS, &param);
        for ( int i = 32; i < 128 ; i++ ) {
                printf("%c",i);
        }
        printf("\n\n");
        for ( int i = 128; i < 256; i++ ) {
                printf("%c",i);
        }
}
```

```
zcc +vg5k world.c xchase.asm -create-app -m -Cl-v -clib=default
```

[image]

Fabrizio-Caruso, 2018-08-12:
@suborb what does changing the font vs udg mean?
I do not understand the difference. I want some change the way some characters look.

What is the difference between  IOCTL_GENCON_SET_FONT32 and IOCTL_GENCON_SET_UDGS?

What I want to do is to change the way a bunch of characters look on multiple targets.

I am not able to display the modified '*' in my game that I compile with:
```
	$(Z88DK_PATH)$(MYZ88DK) +svi $(SCCZ80_OPTS) \
	-clib=ansi -pragma-define:ansicolumns=32 -pragma-redirect:CRT_FONT=_udgs -vn -lndos \
	-DSOUNDS \
	-DFULL_GAME -D__SVI__ -DBETWEEN_LEVEL -DEND_SCREEN \
	-DCONIO \
	-create-app \
	$(CROSS_LIB_PATH)/sound/z88dk/psg/psg_sounds.c  \
	$(CROSS_LIB_PATH)/display/graphics_data/z88dk/xchase.asm \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	$(SOURCE_PATH)/end_screen.c \
	$(FULL_FILES)
```

Fabrizio-Caruso, 2018-08-12:
If I run:
```
#include <graphics.h>
...
#include <sys/ioctl.h>
extern unsigned char udgs[];
void INIT_GRAPHICS(void)
{
	void *param = &udgs;
	console_ioctl(IOCTL_GENCON_SET_FONT32, &param);
	printf("%c",'*');
	while(1){};
}
```
on the SVI I get some corrupted characters on the start screen:
[image]

Fabrizio-Caruso, 2018-08-12:
If I run the very same code:
```
#include <graphics.h>
...
#include <sys/ioctl.h>
extern unsigned char udgs[];
void INIT_GRAPHICS(void)
{
	void *param = &udgs;
	console_ioctl(IOCTL_GENCON_SET_FONT32, &param);
	printf("%c",'*');
	while(1){};
}
```
under the VG5K, the * character is displayed unmodified:
[image]

suborb, 2018-08-12:
Re the vg5k, you'll be wanting this commit: 37c9af57a9e2590956f3c857f35e2ebc63cac633 - it wasn't correctly setting a flag when programmatically setting the font.

suborb, 2018-08-12:
When using the generic console:

* Fonts are character codes 32 - 127
* udgs are character codes 128

They are kept in separate banks and can be pointed to character sets separately.

Thus IOCTL_GENCON_SET_FONT32 sets character codes 32-127 and IOCTL_GENCON_SET_UDGS sets codes 128-x. Where x is either 255 or as many UDGs as are supported by the platform.

suborb, 2018-08-12:
Re SVI: Try doing a clear screen - for some reason the SVI isn't switched into graphics mode automatically on startup.

Fabrizio-Caruso, 2018-08-12:
Thanks! So, if I modify charactwrs with Paulo's tool, I am not modifying udgs and I need
IOCTL_GENCON_SET_FONT32.

Can I use conio/ansi/vt100 wirh IOCTL_GENCON_SET_FONT32 to display modified characters?

suborb, 2018-08-12:
`IOCTL_GENCON_SET_FONT32` only works with the generic console. There is a way to change the font for the vt100 driver, but it differs based on how many columns are configured.

The good news is that everything in conio.h does work with the generic console and I think most ports are using it by default - the exception being vg5k, hence the -clib=default above.

Fabrizio-Caruso, 2018-08-12:
@suborb OK I need to wait for the next snapshot to see the modified * character?

I have tried:
```
#include <graphics.h>
#include "../../../display_macros.h"
#include <sys/ioctl.h>
extern unsigned char udgs[];
void INIT_GRAPHICS(void)
{
	void *param = &udgs;
	console_ioctl(IOCTL_GENCON_SET_FONT32, &param);
	CLEAR_SCREEN();
	printf("%c",'*');
	while(1){};
}
```
compiled with the clib=default option:
```
vg5k_exp_16k:
	$(Z88DK_PATH)$(MYZ88DK) +vg5k -clib=default \
	$(SCCZ80_OPTS) \
	-DSOUNDS -vn -DFULL_GAME -D__VG5K__ -DBETWEEN_LEVEL -DEND_SCREEN -DASM_DISPLAY \
	-lndos -create-app -o $(BUILD_PATH)/FULL_vg5k_exp_16k.prg \
	-DCONIO \
	-DCONIO_VT52 \
	$(SOURCE_PATH)/end_screen.c \
	$(FULL_FILES) \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	$(CROSS_LIB_PATH)/display/graphics_data/z88dk/xchase.asm
```
and I still see the unmodified * character.

Fabrizio-Caruso, 2018-08-12:
On the SVI target I get the very same result with the very same code even if I clear the screen: the * character remains unmodified.

Remark: I am using conio and compiling with:
```
svi_328:
	$(Z88DK_PATH)$(MYZ88DK) +svi $(SCCZ80_OPTS) \
	-clib=ansi -pragma-define:ansicolumns=32 -pragma-redirect:CRT_FONT=_udgs -vn -lndos \
	-DSOUNDS \
	-DFULL_GAME -D__SVI__ -DBETWEEN_LEVEL -DEND_SCREEN \
	-DCONIO \
	-create-app \
	$(CROSS_LIB_PATH)/sound/z88dk/psg/psg_sounds.c  \
	$(CROSS_LIB_PATH)/display/graphics_data/z88dk/xchase.asm \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	$(SOURCE_PATH)/end_screen.c \
	$(FULL_FILES)
```

Are the options:
```
-clib=ansi -pragma-define:ansicolumns=32 -pragma-redirect:CRT_FONT=_udgs 
```
prevent the * character from being modified?

suborb, 2018-08-12:
Sorry @Fabrizio-Caruso I've just been through a bit of emulator pain with the SVI, I've finally got something working now.

    zcc +svi world.c xchase.asm -create-app -pragma-redirect:fputc_cons=fputc_cons_generic -o xchase

Gives this screenshot:

[image]

Fabrizio-Caruso, 2018-08-13:
@suborb  thanks
I don't understand what you wrote about conio supporting generic display.

Can I use the conio APIs (included in conio.h) and see the modified characters with the commands described above? Or should I rather avoid including conio.h?
If I include conio.h and I want to display the modified characters, do I have to pass a pragma to force 8x8 fonts?

Fabrizio-Caruso, 2018-08-13:
@suborb It does not seem to work with conio.h.

I can compile
```
svi_328:
	$(Z88DK_PATH)$(MYZ88DK) +svi -pragma-redirect:fputc_cons=fputc_cons_generic $(SCCZ80_OPTS) \
	 -vn -lndos \
	-DSOUNDS \
	-DFULL_GAME -D__SVI__ -DBETWEEN_LEVEL -DEND_SCREEN \
	-DCONIO \
	-create-app \
	$(CROSS_LIB_PATH)/sound/z88dk/psg/psg_sounds.c  \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	$(CROSS_LIB_PATH)/display/graphics_data/z88dk/xchase.asm \
	$(SOURCE_PATH)/end_screen.c \
	$(FULL_FILES)
```
and use the code:
```
#include <graphics.h>
#include "../../../display_macros.h"
#include <sys/ioctl.h>
extern unsigned char udgs[];
void INIT_GRAPHICS(void)
{
	void *param = &udgs;
	console_ioctl(IOCTL_GENCON_SET_FONT32, &param);
}
```
but I cannot display the modified *:

[image]

Fabrizio-Caruso, 2018-08-13:
@suborb I have managed to get the modified character by redefining gotoxy with VT52 control codes!
How do I get the right colors in VT52/generic mode?

suborb, 2018-08-13:
You shouldn't need to redefined gotoxy() - it will work with both ANSI (vt100) and gencon (vt52) terminals. The only time you might need to define it is if your target supports neither.

I'm working through adding the colour mapping (from the conio colours to native) - I've just added appropriate mapping for the vg5k and a couple of days ago I did a few of the MC6847 targets.

Fabrizio-Caruso, 2018-08-13:
@suborb you are right (as usual) about the gotoxy.

THANKS a lot for working on the color mapping! 
The vg5k still has wrong colors with today's snapshot (2018/08/13).

Generic user-defined graphics is verr cool for color tile/character-based games! 
(such as the ones my cross-devkit library supports).

suborb, 2018-08-13:
When I said just, I meant just! ff441733e448bafb408b5044999f96570a6ba5dc - just a few minutes before that post, should be sorted in the next nightly.

Between you and RobertK I think we're getting there in terms of making cross platform games easier, I realise it's still not as easy as it could be, but we'll get there. 

Though I am getting a bit fed up with all the different ways an MC6847 can be connected!

Fabrizio-Caruso, 2018-08-14:
@suborb THANKS a lot for your huge effort. I will soon check the colors.
It is a worthwhile effort to improve portability across different targets.
It is not optimal but it is quite good and improving.
In the future it would be better to have targets use the same options. It is sometimes a matter or trial and error for me but I cannot complain at all. Z88DK allows coding portable code in many different ways.

Fabrizio-Caruso, 2018-08-14:
@suborb, 
colors on the VG5K seems more or less of with yello being wrong (displayed as red).

On the vg5k 8x8 fonts are displayed in 8x10 characters with horizontal top and botton blank lines.
Graphics with tiles is usually done by combining multiple characters. This is made impossible with 8x8 fonts on the VG5K.

suborb, 2018-08-14:
I'll check yellow, but it should be right:

YELLOW = 14

And then the 15th row in this table has red and green bits set.

        defb    @00000000       ;BLACK -> GREEN
        defb    @00000100       ;BLUE -> BLUE
        defb    @00000010       ;GREEN -> GREEN
        defb    @00000110       ;CYAN -> CYAN
        defb    @00000001       ;RED -> RED
        defb    @00000101       ;MAGENTA -> MAGENTA
        defb    @00000001       ;BROWN -> RED
        defb    @00000111       ;LIGHTGRAY -> WHITE
        defb    @00000111       ;DARKGRAY -> WHITE
        defb    @00000100       ;LIGHTBLUE -> BLUE
        defb    @00000010       ;LIGHTGREEN -> GREEN
        defb    @00000110       ;LIGHTCYAN -> CYAN
        defb    @00000001       ;LIGHTRED -> RED
        defb    @00000101       ;LIGHTMAGENTA -> MAGENTA
        defb    @00000011       ;YELLOW -> YELLOW
        defb    @00000111       ;WHITE -> BUFF

It was a deliberate decision to pad out the 8x8 to 8x10 tiles. This way we get compatibility for fonts and code across all the platforms. An extra (non-portability) IOCTL can be added for 8x10 fonts but that will mean you'll have assets and code just for the VG5K.

Fabrizio-Caruso, 2018-08-14:
@suborb I set the color with textcolor(color).

Fabrizio-Caruso, 2018-08-14:
@suborb  Thanks! 
SVI still has wrong colors.
I don't know what you have fixed so far.
Which targets should I test?

suborb, 2018-08-14:
Okay, SVI, MSX, PV2000, Coleco etc etc now have the following colours when using gencon:

        defb    $1      ;BLACK -> BLACK
        defb    $4      ;BLUE -> DARK BLUE
        defb    $c      ;GREEN -> MEDIUM GREEN
        defb    $7      ;CYAN -> CYAN
        defb    $6      ;RED -> DARK RED
        defb    $d      ;MAGENTA -> MAGENTA
        defb    $8      ;BROWN -> MEDIUM RED
        defb    $b      ;LIGHTGRAY -> LIGHT YELLOW
        defb    $b      ;DARKGRAY -> GRAY
        defb    $5      ;LIGHTBLUE -> LIGHT BLUE
        defb    $3      ;LIGHTGREEN -> LIGHT GREEN
        defb    $7      ;LIGHTCYAN -> CYAN
        defb    $9      ;LIGHTRED -> LIGHT RED
        defb    $d      ;LIGHTMAGENTA -> MAGENTA
        defb    $a      ;YELLOW -> DARK YELLOW
        defb    $f      ;WHITE -> WHITE

[image]

Fabrizio-Caruso, 2018-08-14:
@suborb  Can I modify some characters with the MC1000? How?
I am getting no changes in the characters with the usual code:
```
#include <graphics.h>
#include "../../../display_macros.h"
#include <sys/ioctl.h>
extern unsigned char udgs[];
void INIT_GRAPHICS(void)
{
	void *param = &udgs;
	console_ioctl(IOCTL_GENCON_SET_FONT32, &param);
}
```
and compiling with:
```
mc1000_48k_udg:
	$(Z88DK_PATH)$(MYZ88DK) +mc1000 $(SCCZ80_OPTS) \
	-subtype=gaming -pragma-define:ansicolumns=32 \
	-clib=ansi \
	-vn  -lndos -create-app -Cz--audio \
...
```

suborb, 2018-08-14:
Another MC6847 machine! Though I've not spent as much time on this one as the others (it's on my list to support more graphics modes)

Right,

You need these options: -clib=default -pragma-redirect:fputc_cons=fputc_cons_generic

And this code:

    int mode = 1;
    console_ioctl(IOCTL_GENCON_SET_MODE, &mode);

And then it will work as you expect.

Fabrizio-Caruso, 2018-08-14:
@suborb could you please provide a complete list of the systems I should test?
I can test both
1. modified characters 
2. colors with generic display

I also need to know how the required options and extra code for each such system.

suborb, 2018-08-14:
The information regarding targets and required screen modes that support UDGs is here: https://github.com/z88dk/z88dk/issues/732#issuecomment-387026960

In all cases you need the generic console, this means:

1. If you're specifying -clib=ansi then don't - the VT100 terminal doesn't support UDGs
2. Add -pragma-redirect:fputc_cons=fputc_cons_generic (for some targets this is enabled by default, but it shouldn't harm to force it)

Fabrizio-Caruso, 2018-08-14:
@suborb, are colors fixed on all of them?

Is the MC1000 the only one that needs extra code (to set a specific screen mode or other code)?

suborb, 2018-08-14:
The comment: https://github.com/z88dk/z88dk/issues/732#issuecomment-387026960 indicates what screen modes are required. 

Most of the colours won't be fixed, it's a fairly easy job to do them, but it's a question of time/priorities.

Fabrizio-Caruso, 2018-08-15:
@suborb 
KO:  SC3000, SVI, MTX and probably others all have wrong WHITE color (shown as YELLOW)
OK: VG5K has correct colors now

suborb, 2018-08-15:
I can't reproduce this one either, the following:

```
#include <stdio.h>
#include <sys/ioctl.h>
#include <conio.h>

extern unsigned char udgs[];

int main() {
        void  *param = &udgs;
        console_ioctl(IOCTL_GENCON_SET_FONT32, &param);

        printf("%c",12);
        for ( int i = 32; i < 128 ; i++ ) {
                printf("%c",i);
        }
        printf("\n\n");
        for ( int i = 0 ; i < 16; i++ ) {
                textcolor(i);
                printf("Colour %d\n",i);
        }
        textcolor(WHITE);
        printf("This is in white\n");
        while ( 1 ) {
        }
}
```

Compiled with:

    zcc +sc3000 world.c xchase.asm -subtype=rom -create-app

Produced:

[image]

suborb, 2018-08-15:
`COLOR_WHITE` doesn't exist within z88dk source tree!

Fabrizio-Caruso, 2018-08-15:
@suborb  WHITE works but COLOR_WHITE produces the yellow color.
COLOR_WHITE is used by other devkits. I would prefer to use COLOR_WHITE for white because this makes my code work on CC65, CMOC and other compilers.

Fabrizio-Caruso, 2018-08-15:
I am surprised everything has worked perfectly so far.

@suborb I must have redefined somewhere else...

suborb, 2018-08-15:
That's great, but you must have a `#define COLOR_WHITE 7` somewhere in your code :)

suborb, 2018-08-15:
Comments passing in the night...

Maybe you just got lucky with your redefines - they happen to match the conio.h colours defined here: https://code-reference.com/c/conio.h/textcolor (which are the ones we use)

Fabrizio-Caruso, 2018-08-15:
@suborb  you are right (as usual). I close this question/issue.

closed 2018-08-15 by Fabrizio-Caruso