z88dk/z88dk #873: [SC3000][bug][keyboard] Keyboard input no longer working

Fabrizio-Caruso, 2018-08-15:
Keyboard input seems to be broken on the SC3000: 
1. my game waits for keyboard input forever.
2. if I disable the "wait for key" routine, my game starts and no key presses are detected

suborb, 2018-08-15:
I can't reproduce this, the following works as expected:

    #include <stdio.h>
    int main()
    {

        while ( 1 ) {
        int k;
                while ( (k = getk())== 0)
                        ;
                printf("%d %c\n",k,k);
        }
    }

Compile lines tested:

    zcc +sc3000 getk.c -create-app -subtype=rom -clib=ansi
    zcc +sc3000 getk.c -create-app -subtype=rom

Fabrizio-Caruso, 2018-08-15:
I will try to better isolate the issue in my code and provide more details.

Fabrizio-Caruso, 2018-08-15:
@suborb 
If I include the very same code in one of the first functions called at start-up I get a frozen screen and no response to input:
```
void INIT_GRAPHICS(void)
{
	#if defined(__SC3000__)
		while ( 1 ) {
		int k;
				while ( (k = getk())== 0)
						;
				printf("%d %c\n",k,k);
		}
	#endif
...
```

I am compiling the game with:
```
sc3000_32k_udg:
	$(Z88DK_PATH)$(MYZ88DK) +sc3000 \
	$(SCCZ80_OPTS) \
	-DFULL_GAME -DEND_SCREEN -DBETWEEN_LEVEL \
	-DCONIO \
	-vn -lndos -create-app -Cz--audio \
	-DANIMATE_PLAYER \
	-DUDG_GRAPHICS \
	-clib=default -pragma-redirect:fputc_cons=fputc_cons_generic \
	-o $(BUILD_PATH)/FULL_sc3000_32k_udg.prg \
```

suborb, 2018-08-15:
The big difference is that I’m using the rom subtype - I couldn’t find the appropriate basic - which is also compatible with the sg1000. 

If I remember correctly keyboard reading is a two stage process so there’s a possibility that the BASIC ROM is interfering. 

I don’t know the last time you had it working, but The major change on this target was at the end of may.

Fabrizio-Caruso, 2018-08-15:
End of May could be the last time I tested the SC3000. It was even tested on real hardware! :-)

I am using Sega_BASIC_Level_3_V1_(SC-3000) as BASIC ROM.

Fabrizio-Caruso, 2018-08-20:
The keyboard input now "works" in BASIC mode. It is the original rom routine. Not good for gaming but better than no input. 
I wonder why this routine interferes with the one you wrote for the rom mode.

suborb, 2018-08-20:
I’m guessin

closed 2018-08-20 by suborb

Fabrizio-Caruso, 2018-08-20:
@suborb whay should I them do if I want to use BASIC mode (that is the most useful mode with real hardware) and the best keyboard input routine?

Fabrizio-Caruso, 2018-08-20:
What may be failing in
https://github.com/Fabrizio-Caruso/CROSS-CHASE/blob/master/src/chase/main.c
is
```
...
				--lives;
				if(lives>0)
				{
					player._status = 1;
}
...
```

I am going to try a few things:
1. to compile without optimization 
2. change the code if something like "if(--lives>0)" and see if I get the same problem
3. maybe I will try to isolate the issue but it may be complicated

Fabrizio-Caruso, 2018-08-20:
@aralbrec  
PROBLEM isolated:
KO: 	--lives; if(lives>0) ...
OK: 	if(--lives>0)...

the KO case is somehow miscompiled.

CC65, ZSDCC, GCC all compile the code correctly

commit 0fbb8b80 (2018-08-19): Fix #873: BASIC build uses ROM. ROM build uses inkey for keyboard input