z88dk/z88dk #874: [Exidy Sorcer(SRR)][bug][keyboard] Program locked if key kept down

Fabrizio-Caruso, 2018-08-15:
A program is locked while any key is kept down.

I compile my game with:
```
srr_udg:
	$(Z88DK_PATH)$(MYZ88DK) +srr $(SCCZ80_OPTS) -pragma-redirect:fputc_cons=fputc_cons_generic \
	-D__SRR__ -vn \
	-lndos \
	-clib=default -pragma-redirect:fputc_cons=fputc_cons_generic \
	-create-app \
...
```

suborb, 2018-08-15:
How are you reading the keyboard? getk() or in_KeyPressed()

Fabrizio-Caruso, 2018-08-15:
I am using getk(). 
(This issue may be related to SC3000's issue, which has totally broken input now but it was working a month or so ago)

suborb, 2018-08-15:
Try getk_inkey() - it scans the keyboard directly rather than calling the rom

Sc3000 will be something different - are you using getk() on that as well?

Fabrizio-Caruso, 2018-08-15:
@suborb  yes, I am using getk() as the default for the Z88DK non-turn-based version of my game with just a few exceptions (Spectrum, trs-80, eaca, msx).

Fabrizio-Caruso, 2018-08-15:
getk_inkey() SOLVES the problem! THANKS A LOT!

Can I use getk_inkey() on other targets? On the Spectravideo?

closed 2018-08-15 by Fabrizio-Caruso