z88dk/z88dk #875: [KC][bug][broken][crash] KC85 crashes and glitches

Fabrizio-Caruso, 2018-08-15:
My game crashes at start-up on the emulated KC85/2 (and other models including KC85/5)

I get a crash soon after start-up if I compile my game with sprites:
```
kc_tiny:
	$(Z88DK_PATH)$(MYZ88DK) +kc -subtype=tap-subtype=tap \
	-D__KC__ \
	-DTINY_GAME -DLESS_TEXT -DNO_SLEEP \
	-create-app -o$(BUILD_PATH)/TINY_kc.bin \
	$(SOURCE_PATH)/display_macros.c \
	$(SOURCE_PATH)/enemy.c \
	$(SOURCE_PATH)/level.c $(SOURCE_PATH)/character.c $(SOURCE_PATH)/game_text.c  \
	$(SOURCE_PATH)/strategy.c $(SOURCE_PATH)/input_macros.c \
	$(SOURCE_PATH)/main.c
	rm $(BUILD_PATH)/TINY_kc.bin
```

On the other hand, KC85 freezes before starting the game if I try to compile the game with udg:
```
kc_udg:
	$(Z88DK_PATH)$(MYZ88DK) +kc -subtype=tap $(SCCZ80_OPTS) \
	-pragma-redirect:fputc_cons=fputc_cons_generic \
	-D__SRR__ -vn \
	-DFULL_GAME -DSOUNDS \
	-DEND_SCREEN -DBETWEEN_LEVEL -DNO_WAIT \
	-DCONIO_VT52 \
	-DCONIO \
	-lndos \
	-DANIMATE_PLAYER \
	-DUDG_GRAPHICS \
	-clib=default -pragma-redirect:fputc_cons=fputc_cons_generic \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	$(CROSS_LIB_PATH)/display/graphics_data/z88dk/xchase.asm \
	$(SOURCE_PATH)/end_screen.c \
	$(FULL_FILES) \
	-create-app -o$(BUILD_PATH)/FULL_kc.bin
```

I can load the udg version of the game with KC85/5 but it will produce just random images and moving glitches.

[image]

suborb, 2018-08-15:
I'm halfway through improving the kc target. There's a few problems with the native console output (not filling the whole screen is one of them!).

The gencon code only works on the 85/2 and 85/3 at the moment - the other models have a different screen layout so I'll need to write different code and figure out a way of detecting which machine is which.

I'm wondering if there's an issue with the size of your program where it ends up paging out the running code to write to the video display. I'll change the console section to end up lower in memory and with luck it won't be paged out.

Fabrizio-Caruso, 2018-08-26:
THANKS a lot! I am going to test it across all targets including HC900.

Fabrizio-Caruso, 2018-08-26:
The version that uses Z88DK monochrome sprites starts but still crashes.

The version that uses UDG graphics won't even start (on either KC 85/4 or KC 85/5) and I get a frozen screen. I get the same result on KC 85/2.

@suborb Are your changes already available in the latest snapshot?

Fabrizio-Caruso, 2018-08-26:
In today's version of my game I have 4 different versions (sprites, putc4x6 and 2 with udg) of my game and none of them works:
- kc_tiny_udg and kc_udg: frozen screen after loading (Starten)
- kc_tiny_sprites: it can start but it will crash soon after
- kc_tiny_putc4x6: won't compile and I cannot spot any useful error message

suborb, 2018-08-26:
I had it working last night. I’ll let you know the compile line and cross chase commit hash later on. As well as checking that I’ve conmitted everything!

Fabrizio-Caruso, 2018-08-26:
@suborb indeed you are right!
I managed to get the game to work on the KC 85/4 and /5 but not on the /2, /3, HC900.

Fabrizio-Caruso, 2018-08-27:
@suborb but my game come in 3 sizes!
Both the tiny and light versions seem to run fine under the KC 85/2

suborb, 2018-08-29:
Okay. I think I can close this one now?

Fabrizio-Caruso, 2018-08-29:
@suborb not really because it does crash when Z88DK monochromatic sprites are used. Doesn't it?
My game works if udg are used. Need to re-test it with sprites and putc4x6.

suborb, 2018-08-29:
Ah okay, sorry I'm losing track of issues!

The plotting code should be good now, but we can do a sanity check with sprites and 4x6.

Fabrizio-Caruso, 2018-08-29:
There are some glitches with sprites and putx4x6 and maybe also with udg.

It may be related to clg/clearscreen.

[image]

zx70, 2019-07-09:
I just tested dstar and it seems good. 
I thinking the screen attributes are not cleaned,  but putsprite looks ok.
Fabrizio,  how's your code on the last versions?

Hojoe42, 2022-10-01:
> There are some glitches with sprites and putx4x6 and maybe also with udg.
> 
> It may be related to clg/clearscreen.

I think https://github.com/z88dk/z88dk/issues/2067 fixes the bug with clg()

closed 2022-10-02 by feilipu