z88dk/z88dk #877: [mc1000][sound][bitbang & psg] missing explosion sounds in bitbang - psg produces beeps

Fabrizio-Caruso, 2018-08-17:
PSG is not working as expected for the MC1000 target:
all effects in my game sound horribly wrong.

Bit bang works ok except for noise effects that are reproduced as beeps.

zx70, 2018-08-17:
some (few) noise effect uses the rom data as random data for white noise, but the mc1000 does not have a visible ROM when in games mode.  What do you get if running in BASIC mode?   The vdp support is still missing iirc

Fabrizio-Caruso, 2018-08-18:
@zx70 same issue on same effect on the Jupiter Ace

Fabrizio-Caruso, 2018-09-06:
SC3000 is also badly affected by sound issues: bit-bang has extremely long and wrong sounds.

suborb, 2018-09-09:
Sound is very subjective :)

I suspect there's a constant wrong somewhere for the speed of the PSG device

However, could you:

a) Attach an example program that shows the fault (including command line)
b) Attach a recording of it working properly (say which port)
c) Attach a recording of it failing on the port that's wrong

And then these issues will be a lot easier to investigate!

Fabrizio-Caruso, 2018-09-09:
@suborb The following example shows some of the issues.

One of the worst issues are the first two sounds on the m5 (third sound is OK on the m5).

I compile the m5 case with:
```
	$(Z88DK_PATH)$(MYZ88DK) +m5 -create-app \
	-vn -lndos \
	$(SOURCE_PATH)/../experiments/sounds.c
```

The following code:
```
#include <sound.h>
#include <stdio.h>

int main(void)
{
	printf("Press any key to start");
	getchar();
	
	printf("bit_fx(1)	Fast increasing scale\n");
	// M5 and SC3000 have an issue with this
	bit_fx(1);
	getchar();
	
	printf("bit_fx(7)	Medium duration moving down, then up\n");
        // M5 and SC3000 have an issue with this	
	bit_fx(7);
	getchar();
	
	printf("bit_fx3(0)	Quick noisy high pitch buzz / break noise\n");
	// ACE, MC1000, SC3000 have an issue with this
	bit_fx3(0);
	getchar();
	
	return 0;
}
```

Fabrizio-Caruso, 2018-09-09:
I have attached a recording of the two very wrong sounds on the m5

[2018-09-09 23-10-01.zip](https://github.com/z88dk/z88dk/files/2364658/2018-09-09.23-10-01.zip)

Fabrizio-Caruso, 2018-09-10:
@zx70  @suborb 
I have attached the binary produced for the Sepcrum, which has the good sounds but I am not picky about the sounds. The sounds could be different as long as it vaguely resembles their description and do not last multiple seconds when they are described as short/quick.
[spectrum_ok.zip](https://github.com/z88dk/z88dk/files/2366456/spectrum_ok.zip)

zx70, 2018-09-13:
Thanks for the samples, they clarify several things.
This problem is not related to the white noise generation, it affects the effects based on bit_beep (square wave generation).   My first impression is that the impacted targets are those machines using the new SN76489 library:    https://github.com/z88dk/z88dk/blob/master/libsrc/psg/sn76489.inc

Fabrizio-Caruso, 2018-09-13:
Yes, indeed. We also have a noise issue on some targets, which is a different issue.

zx70, 2019-01-09:
I think I found one of the problems you experimented with the sound rountines: part of them used Self Modifying Code, which cannot be run in ROM.   I changed the section assignments to smc_clib where necessary, so that the segment initialization will do the proper job at startup.
Please check it after the next nighty build  ;)

zx70, 2024-06-21:
I'm on the sound stuff again.
I took the opportunity to brush up this old issue, I think the MC-1000 is not as bad as it was at the time.
bit-fx3(0) doesn't sound right on the MC-1000, but it is a special case, it is a prototype by Ensjo bitbanging the YM PSG soundchip which I still couldn't get running on other targets with the same chip.

Do we have more to fix?  Fabrizio, can you help with a fresh report?

zx70, 2024-06-21:
Nevermind, I noticed there is still something to be fixed case by case, e.g. we have variable delay in loops using a ROM region as a random source. it creates those long silences when a clean RAM area is accidentally used in place of the expected dirt.

Fabrizio-Caruso, 2024-06-21:
Hi! Sure I can rerun some tests. 
I must have told you before but I prefer to repeat myself: Since all of these sounds are CPU blocking sounds, the shorter they are the more usable they will be (not just in games).
So, if the few ones that are supposed to be very short (barely more than a beep/burp) happened not to be short on a given target, then they are no longer usable as portable effects and people have to add extra conditions to avoid them.
If you ever decide to add new sounds, please consider very short ones.

zx70, 2024-06-24:
I'm not adding new sounds at the moment.  I've shortened some of them considering that short samples can be repeated to get a longer soung or mixed to create personalized effects.

zx70, 2024-06-24:
Fine, I've run through the library again trying to make it hardware independent.
A feedback after the next nighty build would be very welcome  :)

zx70, 2024-06-24:
I should also point out that the 1bit library grew up in the meantime.
To play music with something different than a square wave you can try bit_synth(), it is possibly slow in a game loop because it requires FP maths, but it makes some hard stuff easies.

		synth_phase(0x1212);
		synth_play("B");
		synth_phase(0x1232);
		synth_play("B");
		synth_phase(0x3231);
		synth_play("B");

...then you have bit_noise(frequency,duration) which generates white noise and can be shortened as much as you like.
E.G.

		bit_noise(300,10);
		bit_noise(300,20);

Or you have patch_beep() getting whatever zero terminated byte stream:

		patch_beep(300,100,"Hello");
		patch_beep(200,200,"World");

zx70, 2024-06-24:
I forgot some new effects.
The syntax is different, e.g.

   bit_beepfx(BEEPFX_YEAH);

-------   Full sampler   ------- 

#include <sound.h>
#include <stdio.h>

void main()
{

getk();

printf("BEEPFX_SHOT_1     \n");
   bit_beepfx(bfx_0);
   while (!getk()) {};
printf("BEEPFX_SHOT_2     \n");
   bit_beepfx(bfx_1);
   while (!getk()) {};
printf("BEEPFX_JUMP_1     \n");
   bit_beepfx(bfx_2);
   while (!getk()) {};
printf("BEEPFX_JUMP_2     \n");
   bit_beepfx(bfx_3);
   while (!getk()) {};
printf("BEEPFX_PICK       \n");
   bit_beepfx(bfx_4);
   while (!getk()) {};
printf("BEEPFX_DROP_1     \n");
   bit_beepfx(bfx_5);
   while (!getk()) {};
printf("BEEPFX_DROP_2     \n");
   bit_beepfx(bfx_6);
   while (!getk()) {};
printf("BEEPFX_GRAB_1     \n");
   bit_beepfx(bfx_7);
   while (!getk()) {};
printf("BEEPFX_GRAB_2     \n");
   bit_beepfx(bfx_8);
   while (!getk()) {};
printf("BEEPFX_FAT_BEEP_1 \n");
   bit_beepfx(bfx_9);
   while (!getk()) {};
printf("BEEPFX_FAT_BEEP_2 \n");
   bit_beepfx(bfx_10);
   while (!getk()) {};
printf("BEEPFX_FAT_BEEP_3 \n");
   bit_beepfx(bfx_11);
   while (!getk()) {};
printf("BEEPFX_HARSH_BEEP_\n");
   bit_beepfx(bfx_12);
   while (!getk()) {};
printf("BEEPFX_HARSH_BEEP_\n");
   bit_beepfx(bfx_13);
   while (!getk()) {};
printf("BEEPFX_HARSH_BEEP_\n");
   bit_beepfx(bfx_14);
   while (!getk()) {};
printf("BEEPFX_HIT_1      \n");
   bit_beepfx(bfx_15);
   while (!getk()) {};
printf("BEEPFX_HIT_2      \n");
   bit_beepfx(bfx_16);
   while (!getk()) {};
printf("BEEPFX_HIT_3      \n");
   bit_beepfx(bfx_17);
   while (!getk()) {};
printf("BEEPFX_HIT_4      \n");
   bit_beepfx(bfx_18);
   while (!getk()) {};
printf("BEEPFX_JET_BURST  \n");
   bit_beepfx(bfx_19);
   while (!getk()) {};
printf("BEEPFX_BOOM_1     \n");
   bit_beepfx(bfx_20);
   while (!getk()) {};
printf("BEEPFX_BOOM_2     \n");
   bit_beepfx(bfx_21);
   while (!getk()) {};
printf("BEEPFX_BOOM_3     \n");
   bit_beepfx(bfx_22);
   while (!getk()) {};
printf("BEEPFX_BOOM_4     \n");
   bit_beepfx(bfx_23);
   while (!getk()) {};
printf("BEEPFX_BOOM_5     \n");
   bit_beepfx(bfx_24);
   while (!getk()) {};
printf("BEEPFX_BOOM_6     \n");
   bit_beepfx(bfx_25);
   while (!getk()) {};
printf("BEEPFX_BOOM_7     \n");
   bit_beepfx(bfx_26);
   while (!getk()) {};
printf("BEEPFX_BOOM_8     \n");
   bit_beepfx(bfx_27);
   while (!getk()) {};
printf("BEEPFX_ITEM_1     \n");
   bit_beepfx(bfx_28);
   while (!getk()) {};
printf("BEEPFX_ITEM_2     \n");
   bit_beepfx(bfx_29);
   while (!getk()) {};
printf("BEEPFX_ITEM_3     \n");
   bit_beepfx(bfx_30);
   while (!getk()) {};
printf("BEEPFX_ITEM_4     \n");
   bit_beepfx(bfx_31);
   while (!getk()) {};
printf("BEEPFX_ITEM_5     \n");
   bit_beepfx(bfx_32);
   while (!getk()) {};
printf("BEEPFX_ITEM_6     \n");
   bit_beepfx(bfx_33);
   while (!getk()) {};
printf("BEEPFX_SWITCH_1   \n");
   bit_beepfx(bfx_34);
   while (!getk()) {};
printf("BEEPFX_SWITCH_2   \n");
   bit_beepfx(bfx_35);
   while (!getk()) {};
printf("BEEPFX_POWER_OFF  \n");
   bit_beepfx(bfx_36);
   while (!getk()) {};
printf("BEEPFX_SCORE      \n");
   bit_beepfx(bfx_37);
   while (!getk()) {};
printf("BEEPFX_CLANG      \n");
   bit_beepfx(bfx_38);
   while (!getk()) {};
printf("BEEPFX_WATER_TAP  \n");
   bit_beepfx(bfx_39);
   while (!getk()) {};
printf("BEEPFX_SELECT_1   \n");
   bit_beepfx(bfx_40);
   while (!getk()) {};
printf("BEEPFX_SELECT_2   \n");
   bit_beepfx(bfx_41);
   while (!getk()) {};
printf("BEEPFX_SELECT_3   \n");
   bit_beepfx(bfx_42);
   while (!getk()) {};
printf("BEEPFX_SELECT_4   \n");
   bit_beepfx(bfx_43);
   while (!getk()) {};
printf("BEEPFX_SELECT_5   \n");
   bit_beepfx(bfx_44);
   while (!getk()) {};
printf("BEEPFX_SELECT_6   \n");
   bit_beepfx(bfx_45);
   while (!getk()) {};
printf("BEEPFX_SELECT_7   \n");
   bit_beepfx(bfx_46);
   while (!getk()) {};
printf("BEEPFX_ALARM_1    \n");
   bit_beepfx(bfx_47);
   while (!getk()) {};
printf("BEEPFX_ALARM_2    \n");
   bit_beepfx(bfx_48);
   while (!getk()) {};
printf("BEEPFX_ALARM_3    \n");
   bit_beepfx(bfx_49);
   while (!getk()) {};
printf("BEEPFX_EAT        \n");
   bit_beepfx(bfx_50);
   while (!getk()) {};
printf("BEEPFX_GULP       \n");
   bit_beepfx(bfx_51);
   while (!getk()) {};
printf("BEEPFX_ROBOBLIP   \n");
   bit_beepfx(bfx_52);
   while (!getk()) {};
printf("BEEPFX_NOPE       \n");
   bit_beepfx(bfx_53);
   while (!getk()) {};
printf("BEEPFX_UH_HUH     \n");
   bit_beepfx(bfx_54);
   while (!getk()) {};
printf("BEEPFX_OLD_COMPUTE\n");
   bit_beepfx(bfx_55);
   while (!getk()) {};
printf("BEEPFX_YEAH       \n");
   bit_beepfx(bfx_56);
   while (!getk()) {};
printf("BEEPFX_AWW        \n");
   bit_beepfx(bfx_57);                 		
}

zx70, 2024-06-25:
Sparse experiments with patch_beep().  If you had weird data or code areas you might try to "play" them and see what comes up (remember it must be zero terminated.

#include <sound.h>
#include <stdio.h>

unsigned char a2[100];

int i;

void main()
{
a2[99]=0;

for (i=0; i<99; i++)
	a2[i]=255+(isin(i/2)/255);
patch_beep(2300,100, a2);   

for (i=0; i<99; i++)
	a2[i]=0x22+i;
patch_beep(2300,100, a2);   

for (i=0; i<99; i++)
	a2[i]=0x22+(rand()/7);
patch_beep(2300,100, a2);

for (i=0; i<99; i++)
	a2[i]=0xAA/i;
patch_beep(2300,100, a2);   

for (i=0; i<99; i++)
	a2[i]=255+(isin(i/2)/255);
for (i=0; i<99; i++)
	patch_beep(10,i, a2);

for (i=0; i<99; i++)
	a2[i]=0x22+i;
for (i=0; i<99; i++)
	patch_beep(10,i, a2);

for (i=0; i<99; i++)
	a2[i]=0x22+(rand()/7);
for (i=0; i<99; i++)
	patch_beep(10,i, a2);

for (i=0; i<99; i++)
	a2[i]=0xAA/i;
for (i=0; i<99; i++)
	patch_beep(10,i, a2);
}