z88dk/z88dk #881: [sord m5][beep] Beep from keyboard routine

Fabrizio-Caruso, 2018-08-18:
It would be nice to remove the beep produced by the rom keyboard routine.

Remark: I can mitigate it a bit by writing the value 1 into
0x703A (frequency)
0x703B (duration).

I fear that there is no way to avoid this beep with the rom routine :-(

Fabrizio-Caruso, 2018-08-18:
Probable solution found:
```
POKE(0x701A,PEEK(0x701A)&0x7F);
```
seems to solve the issue.

Please consider this solution as an initialization of the keyboard routines.

closed 2018-08-18 by Fabrizio-Caruso

suborb, 2018-08-18:
Have you tried using getk_inkey()

Fabrizio-Caruso, 2018-08-18:
@zx70 I don't know which targets have getk_inkey()... I will try it on the Sord M5 but I found a solution to disable the click anyway (unset bit 7 of 0x701A).
I tried on other targets and it was not found. I would need getk_inkey() for the SVI because the rom routine for the Spectravideo is not usable for games (keeping a key press is detect with a huge delay).

suborb, 2018-08-18:
I’ve been doing them as problems arise with the firmware drivers. 

Given that there’s a problem with the svi I’ll tackle it next weekend.

Fabrizio-Caruso, 2018-08-18:
@suborb  thanks!

Worst case is SC 3000 now no longer working in Basic mode since mid-May
#873

suborb, 2018-08-18:
Yeah. As I mentioned I was focusing on the rom target. 

I didn’t delete the code so I’ll restore the basic mode key input and just use inkey for the rom mode.