z88dk/z88dk #885: z80asm: bug in application of all operators in defined constants in some circumstances

aralbrec, 2018-08-20:
Reported at https://www.z88dk.org/forum/viewtopic.php?id=10496

The bug only occurs when sccz80 is used as compiler.  The same source compiled with zsdcc produces correct symbol values.  The only material difference I can see in the generated asm is that sccz80 is creating two `module` directives.

test.c
```
void main(void)
{
__asm
ici: 
    DEFC toto = ici %8
__endasm;
}
```

zsdcc compile:
zcc +zx -vn -clib=sdcc_iy -m --list test.c -o test

map file shows:
```
ici                             = $8C45 ; addr, local, , test_zsdcc_c, code_compiler, test-zsdcc.c:245
toto                            = $0005 ; addr, local, , test_zsdcc_c, code_compiler, test-zsdcc.c:246
```

The translated asm:
```
1     0000              MODULE test_zsdcc_c
2     0000              LINE 0, "test-zsdcc.c"
0     0000              
1     0000              ;--------------------------------------------------------
2     0000              ; File Created by SDCC : free open source ANSI-C Compiler
3     0000              ; Version 3.6.9 #9958 (MSVC)
4     0000              ;--------------------------------------------------------
5     0000              ; Processed by Z88DK
6     0000              ;--------------------------------------------------------
7     0000              
8     0000              	EXTERN __divschar
9     0000              	EXTERN __divschar_callee
10    0000              	EXTERN __divsint
11    0000              	EXTERN __divsint_callee
12    0000              	EXTERN __divslong
13    0000              	EXTERN __divslong_callee
14    0000              	EXTERN __divslonglong
15    0000              	EXTERN __divslonglong_callee
16    0000              	EXTERN __divsuchar
17    0000              	EXTERN __divsuchar_callee
18    0000              	EXTERN __divuchar
19    0000              	EXTERN __divuchar_callee
20    0000              	EXTERN __divuint
21    0000              	EXTERN __divuint_callee
22    0000              	EXTERN __divulong
23    0000              	EXTERN __divulong_callee
24    0000              	EXTERN __divulonglong
25    0000              	EXTERN __divulonglong_callee
26    0000              	EXTERN __divuschar
27    0000              	EXTERN __divuschar_callee
28    0000              	EXTERN __modschar
29    0000              	EXTERN __modschar_callee
30    0000              	EXTERN __modsint
31    0000              	EXTERN __modsint_callee
32    0000              	EXTERN __modslong
33    0000              	EXTERN __modslong_callee
34    0000              	EXTERN __modslonglong
35    0000              	EXTERN __modslonglong_callee
36    0000              	EXTERN __modsuchar
37    0000              	EXTERN __modsuchar_callee
38    0000              	EXTERN __moduchar
39    0000              	EXTERN __moduchar_callee
40    0000              	EXTERN __moduint
41    0000              	EXTERN __moduint_callee
[... 487 lines cut ...]
21    0000              ici:
22    0000                  DEFC toto = ici %8
23    0000  C9          	ret
24    0001              
25    0001              
26    0001              
27    0001              ; --- Start of Static Variables ---
28    0001              
29    0001              	SECTION	bss_compiler
30    0000              	SECTION	code_compiler
31    0001              
32    0001              
33    0001              ; --- Start of Scope Defns ---
34    0001              
35    0001              	GLOBAL	_main
36    0001              
37    0001              
38    0001              ; --- End of Scope Defns ---
39    0001              
40    0001              
41    0001              ; --- End of Compilation ---
42    0001              
```

As stated the only material difference I can see is a second module directive in the asm generated by sccz80.

aralbrec, 2018-08-20:
The second module directive is not the issue.  I translated the .c file to .asm with this:

zcc +zx -vn -a -clib=new test.c

Then manually edited the resulting test.c.asm to remove the module directive.  Then compiled using this asm file:

zcc +zx -vn -clib=new -m --list test.c.asm -o test

The result is one module directive but the issue is still there.

pauloscustodio, 2018-08-20:
Thanks for reporting. Problem reproduced.

commit db0328b1 (2018-08-21): Fix #885: bug in calculating expressions with one symbol and other operands

When an expression was defined as a label and some computuation (e.g. label %8)
the computation was done at assembly time, when the label is not yet defined,
causing wrong results.

Moved the computation of these expressions to the link phase.
Moved also similar expressions used as JR arguments to the link phase.
Bumpped the object file version to accomodate the JR argument expression.

One side effect of this fix is that JR to different modules is no longer forbidden,
and the int range error if the jump is too far co [... cut]