z88dk/z88dk #892: classic zx: add esxdos functions from newlib

aralbrec, 2018-08-25:
There is some demand to add the esxdos functions to the classic library.  esxdos is a posix-like disk interface to sd cards and ide drives.  It uses a "rst 8; defb service" interface to carry out disk functions.

There is a esxdos interface in the newlib already in arch/zxn/esxdos and arch/zx/esxdos.  The version in arch/zxn/esxdos includes the functions from arch/zx/esxdos and it would be preferred to put both in.  esxdos itself is incompletely documented but the zxn esxdos api is completely documented and has a larger api, a lot of which should work in esxdos.  Anyway until the esxdos api is fully documented, this seems to be the best way to get at all the functionality.

I will do this myself but I will put it up in case someone else wants a crack it first.  The esx_ and esxdos_ defines can be found in https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/zxn/config_zxn_private.inc

suborb, 2018-08-25:
I imported the fcntl level of functions a while back so reading/writing files is already available. 

Which additional functions are most in demand?

Xalior, 2018-08-25:
For me, I needed tapfile manipulation.

timgilberts, 2018-08-25:
I was hoping to use things like the ANSI library but, mixed with with other ESXDOS API calls like esx_m_drvapi

suborb, 2018-08-25:
I think this is done now and will be available in the 20180826 build. When you do a classic +zx compile and link with -lesxdos then the following is true:

- POSIX functions such as open/close/read/write are passed through to ESXDOS
- stdio FILEs will use ESXDOS
- The newlib ESXDOS API is available in classic - `<arch/zx/esxdos.h>` defines the API as it does with newlib
- The ESX functions are also available to sdcc compiles

The time conversion functions haven't been added as of yet - I need to harmonise struct tm between newlib and classic - but hopefully they're not critical.

closed 2018-08-25 by suborb

aralbrec, 2018-08-26:
The header <arch/zx/esxdos.h> is not committed and not available in the nightlies.

Xalior, 2018-08-26:
I noticed that too, after a "make clean && ./build.sh" on my local system that file was copied into the classic include/arch/zx/ folder.

Looking at my local buildlogs, and the ones on https://travis-ci.org/z88dk/z88dk/jobs/420468251 I notice that the build ends:
`z80asm --cpu=z80-zxn  -xlib/sccz80/zxn @target/zxn/library/zxn_sccz80.lst
z80asm --cpu=z80-zxn  -xlib/sdcc_ix/zxn -D__SDCC -D__SDCC_IX @target/zxn/library/zxn_sdcc_ix.lst
z80asm --cpu=z80-zxn  --IXIY -xlib/sdcc_iy/zxn -D__SDCC -D__SDCC_IY @target/zxn/library/zxn_sdcc_iy.lst` at https://travis-ci.org/z88dk/z88dk/jobs/420468251#L4008 

Where as my local build ends:
`z80asm --cpu=z80-zxn  -xlib/sccz80/zxn @target/zxn/library/zxn_sccz80.lst
z80asm --cpu=z80-zxn  -xlib/sdcc_ix/zxn -D__SDCC -D__SDCC_IX @target/zxn/library/zxn_sdcc_ix.lst
z80asm --cpu=z80-zxn  --IXIY -xlib/sdcc_iy/zxn -D__SDCC -D__SDCC_IY @target/zxn/library/zxn_sdcc_iy.lst
cp -r sccz80/adt ../
cp -r sccz80/alloc ../
cp -r sccz80/arch/zx/esxdos.h ../arch/zx/esxdos.h
cp -r sccz80/arch/zxn/esxdos.h ../arch/zxn/esxdos.h`

Hope that helps.

-Dx

suborb, 2018-08-26:
Annoyingly (and my fault) there's 3 different ways of building:

- nightly
- manual
- travis

Nightly/travis weren't regenerating the newlib headers so never copied the esxdos.h into the classic include folder. The same has been true for adt and alloc headers for months :)