z88dk/z88dk #898: zsdcc: bug in static initializer for unions in structs

aralbrec, 2018-08-27:
I'll just copy this out of #894 : 

I'm amazed but it looks like zsdcc does named initilaization:

```
#include <stdint.h>

struct esx_drvapi
{
   union
   {
      uint16_t bc;
      struct
      {
         uint8_t driver;
         uint8_t function;
      }
      call;
   };

   uint16_t de;
   uint16_t hl;
};

//struct esx_drvapi rtc = { {0}, 0, 0 };   // sdcc incorrectly initializing c89 way too
struct esx_drvapi ruc = { .call.driver = 1, .call.function = 2, .de = 3, .hl = 4 };
```

zcc +zxn -a -clib=sdcc_iy zzz.c

but it's getting the union wrong and doing both members of the union in the data structure:

```
_ruc:
	DEFW +0x0000
	DEFB +0x01
	DEFB +0x02
	DEFW +0x0003
	DEFW +0x0004
```

So .bc and the .call struct are not sharing space. That's why the weird initilaizer was working - the struct layout is broken.

So I think the only way to do it right for both compilers is to define the struct in asm.

In c make the compiler aware of it:
```
extern struct esx_drvapi ruc;
```

In a separate asm file define ruc's contents. Notice a data section is used because this will be non-zero initialized.

asm
```
SECTION data_user

PUBLIC _ruc

_ruc:

   defb 0   ; .call.driver
   defb 0   ; .call.function
   defw 0   ; .de
   defw 0   ; .hl
```

Then add that asm file to the compile line.

aralbrec, 2018-08-27:
Thankfully it looks like the code generator gets it right so it's just the initialization broken in sdcc:

```
#include <stdint.h>

struct esx_drvapi
{
   union
   {
      uint16_t bc;
      struct
      {
         uint8_t driver;
         uint8_t function;
      }
      call;
   };

   uint16_t de;
   uint16_t hl;
};

//struct esx_drvapi rtc = { {0}, 0, 0 };
struct esx_drvapi ruc = { .call.driver = 1, .call.function = 2, .de = 3, .hl = 4 };

void main(void)
{
	ruc.bc = 10;
	ruc.de = 12;
	ruc.hl = 15;
	ruc.call.driver = 1;
	ruc.call.function = 2;
}
```

```
_main:
	ld	hl,0x000a
	ld	(_ruc), hl
	ld	l,0x0c
	ld	((_ruc + 0x0002)), hl
	ld	l,0x0f
	ld	((_ruc + 0x0004)), hl
	ld	hl,_ruc
	ld	(hl),0x01
	ld	hl, +(_ruc + 0x0001)
	ld	(hl),0x02
	ret
```

aralbrec, 2018-08-27:
Will have to check if this is addressed in later sdcc versions.

spth, 2018-09-10:
I just did some testing with current SDCC 3.7.2 #10539; there is indeed a problem when using designated initializers with anonymous unions; it seems it results in an error message, not a silent failure though. I filed https://sourceforge.net/p/sdcc/bugs/2810/

Philipp

suborb, 2020-07-06:
On r11715 this now reports:

```
init.c:21: warning 147: excess elements in scalar initializer after 'ruc'
init.c:22: error 47: indirections to different types assignment
Caught signal 11: SIGSEGV
```

feilipu, 2020-07-17:
```C
// sdcc -S bug2810.c 

struct s
{
   union
   {
      int a;
      int b;
   };

   int c;
   int d;
};

struct s s2 = { .a = 0, .c = 3, .d = 4 };
```

feilipu, 2021-05-22:
Still open as of r12402.

suborb, 2023-06-14:
r14110 now reports this:

```
named.c:21: warning 147: excess elements in scalar initializer after 'ruc'
named.c:22: error 47: indirections to different types assignment
Internal error: validateLink failed in SPEC_NOUN(type) @ SDCCsymt.c:3950: expected SPECIFIER, got UNKNOWN
```

suborb, 2023-06-14:
When I added named specifiers to sccz80 this case wasn't considered at all, so there we have:

```
named.c:21:32: fatal error: Unknown structure member call
Compilation aborted
```

feilipu, 2026-10-14:
Checked on zsdcc 4.6.0 r16639.

The anonymous union now overlaps. `sizeof(struct esx_drvapi)` is 6. `bc` and `call` are both at offset 0. `de` is at 2. `hl` is at 4. The old image stored `bc` and then `call`. That layout is gone. Assignments in `main` write the right offsets. The compiler no longer crashes.

The static initializers are still wrong.

```c
struct esx_drvapi ruc = { .call.driver = 1, .call.function = 2, .de = 3, .hl = 4 };
```

SDCC prints `warning 147: excess elements in struct initializer`. The image is:

```asm
.dw #0x0000
.dw #0x0001
.dw #0x0003
```

The correct image is:

```asm
.db #0x01
.db #0x02
.dw #0x0003
.dw #0x0004
```

`{ .bc = 0x0201, .de = 3, .hl = 4 }` emits `.dw #0x0201`, `.dw #0x0000`, `.dw #0x0003`. `{ .a = 0, .c = 3, .d = 4 }` emits `.dw #0x0000`, `.dw #0x0000`, `.dw #0x0003`. The value after the anonymous union moves into the wrong member. The last value is dropped.

`{ {0}, 0, 0 }` and `{ 0x0201, 3, 4 }` do not compile. SDCC reports error 47, then error 35 and error 2. `{ 0 }` compiles. That image is six zero bytes.

A named union is correct. `{ .u.call.driver = 1, .u.call.function = 2, .de = 3, .hl = 4 }` emits `01 02` then `.dw #0x0003` and `.dw #0x0004`. There is no warning.

This front end is SDCC. The z88dk patch does not change it. Upstream report [2810](https://sourceforge.net/p/sdcc/bugs/2810/) is still open. A note of 2026-08-02 says patches/508 fixes the designated bytes on r16747. That revision is after 4.6.0.