z88dk/z88dk #900: [pps][input] Keeping a key pressed slows down the computer significantly

Fabrizio-Caruso, 2018-08-27:
Keeping a key pressed slows down the computer significantly

Tested with my game Cross Chase

suborb, 2018-08-27:
Sorry there's nothing we can do about this, this is an emulator issue, the following demonstrates the issue:

```
#include <stdio.h>

int main() {
        unsigned char i = 0;
        while ( 1 ) {
                i++;
                fputc_cons((i %32)+ 'A');
#asm
        ld      c,$31   ;SCANKEY
        rst     $10
#endasm
        }
}
```

The assembler calls directly into the emulator (it's the same code used by getk()) into the file estex.cpp so there's something slow in there. Pressing a key when not scanning the keyboard has no ill effect.

suborb, 2018-08-27:
Ah, the problem is this line in sprint.cpp:

```
 if(D||E)
 {
  if(fdebug!=NULL) fprintf(fdebug,"\n>>> KBD #%2.2X #%2.2X (%c)\n",D,E,(E>32)?E:' ');
//  if(E&&!flag)
    delay(120);
 }
```

Where if there's a keypress then it pauses for a while!

closed 2018-08-27 by suborb

Fabrizio-Caruso, 2018-08-27:
Where is this code? The emulator?

suborb, 2018-08-27:
Yes, it's the emulator source code.

I've got a version that's 64 bit clean and compiles on a Mac here: https://github.com/suborb/sprint

Shaos used to have a copy on GitHub, but I presume he's moved it to somewhere on gitlab: https://gitlab.com/shaos

Fabrizio-Caruso, 2018-08-27:
Thanks @suborb for the investigation!