z88dk/z88dk #902: zsdcc: But in __smallc calling convention

suborb, 2018-08-27:
When zsdcc calls functions with a char argument it only pushes a single byte on the stack.

For compatibility with sccz80 it should be pushing a word.

This has been present for a while and has been worked around by promoting chars to ints in classic function prototypes.

However, that's not a great solution when classic pulls in newlib code.

spth, 2018-09-11:
Is this about char only? Also about unsigned char? signed char? How is the 8-bit value to be converted to the 16-bit value? Any value for the upper 8 bits allowed? Sign-extended? Zero-extended? Depends on the 8-bit type?

Philipp

suborb, 2018-09-11:
Good questions. It affects both signed and unsigned char. Any value is permitted for the upper 8 bits.

spth, 2018-09-13:
Assuming, that _Bool is 16-bit in __smallc calling convention, too, this is fixed in upstream SDCC as of today (so the fix will be in 3.8.0).

Philipp

suborb, 2018-09-13:
Ah brilliant. Thank you Philipp. Apologies for not responding as well.

suborb, 2019-01-27:
Closing this one now we're up to SDCC 3.8.5

closed 2019-01-27 by suborb