z88dk/z88dk #903: [Bandai Rx-78][question] How to test?

Fabrizio-Caruso, 2018-08-27:
What do I need to test the rom that Z88DK creates?

I am using Takeda specific emulator.
Inserting the rom gives a black screen.

Do I need additional roms or settings?

suborb, 2018-08-27:
I've got an ipl.rom, 8192 bytes: MD5 (ipl.rom) = f613b15c4b965013e4827d2ad79c7080

It is required because it detects whether a cartridge has been inserted and then jumps to it as well as handling the interrupt etc

Fabrizio-Caruso, 2018-08-27:
I need a similar rom for the Casio PV 2000

Fabrizio-Caruso, 2018-08-27:
and for the multi8...

Fabrizio-Caruso, 2018-08-27:
and maybe others... Is MAME also a viable emulator for these systems?

suborb, 2018-08-27:
I'll send you (via email) a zip file of my environment for these emulators that have the ROMs.

suborb, 2018-08-27:
I’ll close this one since you’ve got the ROMs now.

closed 2018-08-27 by suborb