z88dk/z88dk #907: [fp1100][bug][appmake] Building application code failed

Fabrizio-Caruso, 2018-08-28:
makeapp is failing to build:
```
/cygdrive/c/z88dk/bin/zcc.exe +fp1100 -O3 -pragma-redirect:fputc_cons=fputc_cons_generic \
-D__FP1000__ -vn \
-DCONIO_VT52 \
-DTINY_GAME -DSOUNDS \
-lndos \
-DCONIO -DNO_INIT_GRAPHICS \
cross_lib/display/display_macros.c cross_lib/display/init_images.c cross_lib/input/input_macros.c cross_lib/text/text_macros.c chase/move_player.c chase/ghost.c  chase/level.c chase/character.c chase/game_text.c chase/main.c chase/strategy.c \
-create-app
Building application code failed
make: *** [Makefile:2109: fp1100_tiny] Error 1
```

suborb, 2018-08-28:
My personal opinion will be to not bother with the FP-1100, it's a really terrible machine for games.

Actually, it's just a really terrible machine.

I'll investigate the failure above though - it might be just a windows thing since world.c works ok on a a Mac.

Fabrizio-Caruso, 2018-08-28:
LOL! Having my game run also on a terrible machines is part of the goal of my project.
Second goal of my project is to test all targets in Z88DK (as well as CC65 and CMOC and other 8-bit devkits).
You are not discouraging me, at all! 
Speaking of bad machines for games, I also wish the Texas calculators had a working input...
Thanks a lot for your support!

suborb, 2018-08-28:
You can't say that you've not been warned about it!

* Yes it's slow
* Yes the keyboard beeps for every keypress (even if you're not scanning the keyboard)
* I'll sort out the colour

I've never done any work on the TI calculators, I think @zx70 is your best bet for those.

Fabrizio-Caruso, 2018-08-28:
Thanks! 
I don't mind if it sucks badly as long as one can compile something runnable with some usability.
My game can run in turn-based mode with just plain text.
I am trying to cover all targets to report problems. 
I decided that I am going to become the official Z88DK tester (aka most hated client).

Fabrizio-Caruso, 2018-08-28:
I just managed to make a playable turn-based version for the G850 (which only displays 6 rows) by using the same game code. My game is very adaptable in terms of speed, memory and display size.
Sorry for being OT.

closed 2018-08-28 by suborb

zx70, 2019-02-19:
Fabrizio, probably you know it already, but the G850 should have both a 6 rows and an 8 rows mode.
In 8 rows mode it should now be possible to provide a custom font, but the format is not the  standard one.

commit 3b23cc52 (2018-08-28): Fix #907: Allocate a block of the right size...