z88dk/z88dk #908: [camputers lynx][create-app][tap] Jynx complains about the tap format

Fabrizio-Caruso, 2018-08-28:
I am compiling my game with:
```
	$(Z88DK_PATH)$(MYZ88DK) +lynx $(SCCZ80_OPTS) \
	-DCAMPUTERS_LYNX \
	-vn \
	-DCONIO_VT52 \
	-DNO_COLOR \
	-DFULL_GAME -DSOUNDS \
	-DEND_SCREEN -DBETWEEN_LEVEL \
	-lndos \
	-DCONIO -DNO_INIT_GRAPHICS \
	$(SOURCE_PATH)/end_screen.c \
	$(FULL_FILES) \
	-create-app -o $(BUILD_PATH)/FULL_camputers_lynx.bin
```
which will produce a .tap file.

Jynx cannot load the .tap files:
"Failed to parse .tap file"

The old Pale emulator does not run on modern Windows.

suborb, 2018-09-02:
The appmake issue was fixed in c041a6816c7c39e55d2538c7c924dd35e9301437 - the resulting .TAP file works in both Jynx and Pale (which works under wine).

In general, I think the Lynx could do with a bit of love. The following: http://www.russelldavis.org/CamputersLynx/files/Newsletters/Lynx%20User/Lynx%20User1.pdf has the information we need to:

1. Do a direct screen driver (use 8x8 fonts instead of the native 6x10)
2. Add graphics supports
3. Add inkey() keyboard support
4. One bit sound

suborb, 2018-09-02:
Bear with me for a few days (and don't raise any issues yet!) I'm getting this machine working in a more standard way, so a familiar screen shot of what's on the way:

[image]

suborb, 2018-09-03:
Hopefully what's in is good enough for Cross Chase.

I'm slightly dubious about the size of the screen - it looks bigger than the one in Mame but that can be adjusted easily enough.

closed 2018-09-03 by suborb

Fabrizio-Caruso, 2018-09-04:
@suborb only for the TINY version :-( 
The TINY(4.8k)  version loads and runs (with bad graphics as expected).
The FULL version (13k) starts loading and then freezes.
There must be a memory-related issue.