z88dk/z88dk #910: [newbrain][question] How to use the files created by appmake?

Fabrizio-Caruso, 2018-08-28:
I can compile for the newbrain target but the doc does provide any details on how to load it.

As suggested by the doc I am using the emulator by Chris Despinidis from
http://www.mything.nl/emulator_UK.htm
Which version works? I used the very latest.

Which files should I use? the _dir.txt? .bin? .data, .bas?

If I start the emulator, there is no way to access any external files.
I can put the created .bas, .data and _dir.txt into the basic folder insider the emulator's folder.
I can see the folder if I select: Storage -> Tape
If I click on the new folder I get an access violation and the emulator crashes.

zx70, 2018-09-12:
I think you can provide raze.dll if it is missing, a dll can stay in the same folder of the EXE file.

I just succeded with version 1.5:
The entries in 'my' _dir.txt  file are:
a.bas
a.dat

... I created a subfolder under "basic" named z88dk, no errors while selecting it from within the emulator.

Fabrizio-Caruso, 2018-09-12:
Been googling for the raze.dll file with no luck...
I can install a more recent version of the same emulator that does not require raze.dll.

I don't understand Z88DK documentation about the files. 
Which files do I really need to run a compiled program in the emulator?
Do I need the _dir.txt or it is just a piece of documentation about the generated files?
The emulator just let me select a .bas file but it cannot load it.
This is what I do:

POKE19,1
Tape->Select File for Loading -> I select the .bas : the emulator displays "load" in the console
I press enter : the emulator displays "a.bas"
I type run : The emulator displays "Not found.bas"

[image]

zx70, 2018-09-13:
You should have 2 files, a.bas and a.dat.
The virtual tape directory file (_dir.txt in the emulator versions I have) should include both in sequence.
I think you are close, don't give up  :)

Fabrizio-Caruso, 2018-09-13:
@zx70 What do I do with the _dir.txt. Is it of any use? Can I delete it?
I only place a.bas and a.dat in a directory?

I have managed to get the raze.dll and use version 1.15 but after starring the load process I get an infinite loop of
a.bas
a.bin
a.bas
...

Fabrizio-Caruso, 2018-09-13:
@zx70, I maged to load a "hello world" example.
My game will immediately freeze the emulator. I need to investigate this...