z88dk/z88dk #917: [z88][question/feature request] Z88 target compilation

Fabrizio-Caruso, 2018-08-29:
I could be nice to be able to compile a standard C program (with functions + main()) for the Z88.

Is this already possible? Is the default compilation command producing something loadable?
How do I load it within BBC BASIC?

suborb, 2018-08-29:
I had two ways of doing this on the real hardware:

- Initially I used the Import/Export popup. I was using an Amiga at the time so I used AToZ. But I think EasyLink worked on PCs
- Later on I switched to using the 3rd party XModem popup - it was a lot quicker

Once it was on the device, it was just a CH. "filename"

I don't really know how to do it using an emulator - I'm guessing if you create a file EPROM then you should be able to transfer it across to :RAM.0 or :RAM.1 using one of the built in apps. And then CH. "filename"

These days with emulators it's much easier to create an application. There's examples in examples/z88/app - dstar.c or rpn.c are the ones to look at.

There's a small amount of decoration that's needed to handle quitting the application on demand and the menu entry - this can be easily contained within a single file.

suborb, 2018-09-18:
Try slot 2. It will end up at the bottom of the menu. 

If you can’t get it work I’ll take a look later on.

suborb, 2018-09-18:
I knew there was a reason I'm keeping 9 years of nightly builds!

A bit of a status report...

It looks like something broke between changes cb0e01c and 0b501cc 

The map file changed during that time - swapping out z80asm fixes/breaks depending on which way you do it.

I'm continuing to investigate

suborb, 2018-09-18:
Should be sorted in the next nightly. Apologies for the inconvenience!

closed 2018-09-18 by suborb

Fabrizio-Caruso, 2018-09-19:
@suborb I still get a 16k rom image, which produces an error about its size (tested on ZEsarUX).

suborb, 2018-09-19:
I believe 16k images are valid and they do load correctly in OZvm.

Fabrizio-Caruso, 2018-09-19:
Sort of. The image can be loaded as eprom and I see the application but I get an error message about no available memory.

I am compiling the following "Hello World" example:
```
#if defined(__Z88__)
	#include <dor.h>
	#pragma -no-expandz88
	#define APP_INFO "Made by z88dk"
	#define APP_KEY  'U'
	#define APP_NAME "Useless"
	#define APP_TYPE AT_Good
	#include <application.h>
#endif

#include <stdio.h>
int main(void) {
    printf("Hello world! \n");
    while(1){};
    return 0;
}
```

[image]

suborb, 2018-09-19:
```
% zcc +z88 hello.c -subtype=app -create-app -m
Application will use 1 pages of bad memory
Safe data is 120
```

A Good application can't use bad memory.

Looking at the generated .map file shows:

```
__printf_format_table           = $2001 ; addr, public, , __printf_format_tabl
e, data_clib, __printf_format_table.asm:25
__printf_format_table64         = $2023 ; addr, public, , __printf_format_tabl
e64, data_clib, __printf_format_table64.asm:16
```

So, the printf table is in the data section, so needs to be copied to RAM. I'll shift them to rodata, but if you add `#pragma printf ""` then there's no format table and everything works:

[image]

commit ccfa7ee4 (2018-09-18): Fix #917: Make sure the parameter is the one we actually want.

Check for whitespace after the parameter name.

Change 0dd53b73db627d11dee4ba148ddcd06ca335cd21 introduced the new
map file format. In turn this changed the order of the symbols within
the mapfile.

The z88 application DOR has a lot of symbols with similar names, so
when search for "in_dor" we ended up matching the first one, which
actually didn't correspond to the actual DOR. Thus we ended up
corruption of the DOR chain.