z88dk/z88dk #918: [eg2k/trs80][input] Broken getk_inkey and Bad getk

Fabrizio-Caruso, 2018-08-29:
@suborb
The EG2K target (and the parent TRS80) suffer from bad input (keys kept pressed not detected)

I compile my game with:
```
eg2k_udg:
	$(Z88DK_PATH)$(MYZ88DK) +trs80 \
	-D__EG2K__ \
	-subtype=eg2000disk \
	$(SCCZ80_OPTS) \
	-D__EG2K__ \
	-DFULL_GAME  \
	-DEND_SCREEN -DBETWEEN_LEVEL \
	-DCONIO_VT52 \
	-DCONIO \
	-lndos \
	-DANIMATE_PLAYER \
	-DUDG_GRAPHICS \
	-DLESS_TEXT \
	-clib=default -pragma-redirect:fputc_cons=fputc_cons_generic \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	$(CROSS_LIB_PATH)/display/graphics_data/z88dk/xchase.asm \
	chase/end_screen.c \
	$(FULL_FILES) \
	-create-app -o$(BUILD_PATH)/FULL_eg2k_udg.bin
```
I get simlar but worse results with
	-pragma-redirect:getk=getk_inkey 

On the TRS80 (no EG2K) I have fixed the problem by reading straight from the port at 0x3802.

I forgot the port address for the EG2K... Do you have the port address?

suborb, 2018-08-29:
The eg2k matrix is at 0xf800

This is what getk_inkey() scans

It scans all 8 ports to detect what key has been pressed. If you know the key you are expecting then in_KeyPressed() may well prove to be more efficient.

Fabrizio-Caruso, 2018-08-29:
getk_inkey() may be scanning at the wrong times. I am trying to guess...
My simple very code works in my code (just peek at the matrix and check the value)

suborb, 2018-08-29:
Dumb question time....

getk_inkey() will return lower case by default (since it handles the shift and ctrl keys) - it does that on all platforms. getk() looks like it returns uppercase by default on the trs80.

So, erm, when using getk_inkey() are you holding down shift?

Fabrizio-Caruso, 2018-08-30:
You are right! and I am an id**t...
I was expecting capital letters!
Sorry for wasting your time on this!

closed 2018-08-30 by Fabrizio-Caruso