z88dk/z88dk #919: [tests][feature][collaboration] Status table

Fabrizio-Caruso, 2018-08-29:
Due to the large number of targets and sub-targets I am testing 
I may keep my test results on a table (maybe excel or something else).
Opening issues is good for tracking issues but not good to get a full picture.
If you want I could share this information with you guys and try to keep it up to date.
Ideally, such a table could be in the Wiki and I (or anyone else) could keep update it from time to time.

The table main have just a few columns:
target | options used | date of most recent test | compiles? (yes/no/warnings) | bugs | non-optimal |  
with primary/unique key = target | options used

suborb, 2018-08-29:
A google sheet is usually pretty handy for this sort of thing

Fabrizio-Caruso, 2018-08-30:
Ok!
I am currently working on
https://docs.google.com/spreadsheets/d/1qo2skBUtCUCAac3knEE2x-bUBdvxEA_6qvElacJ3xNY/edit#gid=0

I will add the remaining targets but for some I won't be able to run my tests because they cannot use the same code (different apis, special decoration, etc...).

closed 2018-09-03 by suborb

Fabrizio-Caruso, 2018-09-20:
@suborb @zx70 @aralbrec 

As of Sept 21st, these are the results of my tests:

Total number of targets: 72 (all targets except those that require Makefile, e.g., Pacman, Rabbit, etc.)

7% (5): still untested/hard to test (ABC800, REX, SOS, TI8X, TI86)
3% (2): look badly broken, i.e., not even "hello world" works (C7420, X07)
1% (1): minimally working with serious issues (MZ2500)
89% (64): working, i.e., at least able to run Cross-Chase in at least one configuration