z88dk/z88dk #921: [gal] clrscr not working

Fabrizio-Caruso, 2018-08-30:
clrscr does not clear the screen.

I compile with:
```
gal_22k:
	$(Z88DK_PATH)$(MYZ88DK) +gal \
	-pragma-need=ansiterminal \
	-D__GAL__ -DFULL_GAME -DEND_SCREEN -DBETWEEN_LEVEL \
	-DCONIO -DNO_INIT_GRAPHICS \
	-vn -lndos -create-app -o  $(BUILD_PATH)/FULL_galaksija.prg \
	$(SOURCE_PATH)/end_screen.c \
	$(FULL_FILES)
	rm $(BUILD_PATH)/FULL_galaksija.prg	
	rm $(BUILD_PATH)/FULL_galaksija.wav
```

suborb, 2018-09-10:
This looks like it works for my tests.

However, I've just implemented generic console for +gal which definitely has a working cls routine.

Fabrizio-Caruso, 2018-09-10:
In my game I have had to implement my own clear_screen that simple feels the screen with the space character.

suborb, 2018-09-10:
You really should be compiling with `-clib=ansi` rather than `-pragma-need=ansiterminal` - the clib option also sets #define that switches to the vt100 commands - this isn't necessarily an issue on the Galakjia since it doesn't have colour.

Fabrizio-Caruso, 2018-09-10:
I confirm the issue with even with clib=ansi.
I have included the malfuncting binary compiled with:
```
gal_6k:
	$(Z88DK_PATH)$(MYZ88DK) +gal \
	-compiler=sdcc \
	$(ZSDCC_OPTS) \
	-clib=ansi \
	--reserve-regs-iy \
	-pragma-include:$(CFG_PATH)/z88dk/zpragma.inc \
...
	-v -lndos -create-app -Cz--audio -o  $(BUILD_PATH)/TINY_galaksija_6k.prg \
	$(TINY_FILES)
```

[TINY_galaksija_6k.zip](https://github.com/z88dk/z88dk/files/2368569/TINY_galaksija_6k.zip)

suborb, 2018-09-11:
Sorry my test just had an appearance of clearing since it was overprinting what was on screen. 

All sorted now.