z88dk/z88dk #922: [z9001][input] Significant delayed dection when keys are kept down

Fabrizio-Caruso, 2018-08-30:
There is a significant delayed dection when keys are kept down

I compile with:
```
	$(Z88DK_PATH)$(MYZ88DK) +z9001 $(SCCZ80_OPTS) -clib=ansi \
	-D__Z9001__ -vn -DFULL_GAME  -DEND_SCREEN -DBETWEEN_LEVEL \
	-DCONIO -DNO_INIT_GRAPHICS \
	-lndos -create-app -o $(BUILD_PATH)/FULL_z9001.z80 \
	$(SOURCE_PATH)/end_screen.c \
	$(FULL_FILES)
```

suborb, 2018-08-30:
What's the behaviour like from the ROM monitor program? - similar delay in terms of key repeat?

Fabrizio-Caruso, 2018-08-30:
In-game it seems to stutter sometimes and be worse.
I do not see any such stuttering at the start-up of the emulated machine (JKCEMU).
I do see some delayed dection when I start the machine.

suborb, 2018-08-30:
On the machine selection go to the Sonstiges tab, for some of the machines there's some options to speed up the keyboard input (Tastaturbuffer) I'm not sure whether they are present for the Z9001.

Might be worth playing around with those if they're present - it's obviously a problem with some of those machines if the option is present in the emulator :)

Fabrizio-Caruso, 2018-08-30:
It would not explaining the in-game stuttering, i.e., once every 2 or 3 tests, with new key kept pressed:
A....A....AAAAAAAAAAAAAAAAA
where .... means no key detected
At start-up I get:
A..AAAAAAAAAAAAAAAAAAA
where .. is a shorter delay

It is not a vital issue. We can live very well with it.

Fabrizio-Caruso, 2018-08-30:
Other targets have worse keyboard detection.
I close this "issue". You may consider it a "wish-to-have".

closed 2018-08-30 by Fabrizio-Caruso