z88dk/z88dk #923: [microbee][input] Keyboard detection

Fabrizio-Caruso, 2018-08-30:
The microbee target seems to have non-optimal keyboard detection (delayed and "too buffered" keyboard detection).

I am not sure this issue is a real one because I am using the on-line emuator
http://nanowasp.org/
that may not be very accurate.

Fabrizio-Caruso, 2023-07-12:
@suborb the input behavior shown under Mame/Mess is worse (many key-presses are lost even if you hammer the keyboard).

suborb, 2023-07-12:
See: https://github.com/mamedev/mame/blob/master/src/mame/ausnz/mbee.cpp

```
    Old CRTC-based keyboard:
    - Paste drops many characters.
    - Typing can drop the occasional character.
    - mbee -bios 1 is unusable due to keyboard issues.
```  

So there's an emulator issue.

Fabrizio-Caruso, 2023-07-12:
Thanks OK. So the problem we have is just that we have no auto-repeat at all. Keeping a key down only triggers one key down detection, no matter how long you keep it down. All other observed issues are emulator-related (either wrong setting in NanoWasp or broken driver in Mame/Mess). I see the no auto-repeat issue on all 3 emulators.

suborb, 2023-07-12:
Yes, it looks that the hardware just doesn't support key repeat (I've just tried out CP/M mode in ubee to eliminate it not being supported by BASIC)

Fabrizio-Caruso, 2023-07-14:
@suborb according to the experts, the hardware does support it. The best way to do it may be different in two different series, though.
I asked in their forum and I got some technical explaination:
https://microbee-mspp.org.au/forum/viewtopic.php?f=56&t=3017

[image]

Would it be possible to implement a new getk that works with auto-repeat so that it can be used in games?

suborb, 2023-07-14:
If there's some nice generic example code that shows how to do it, then yes, a routine can be added.

Fabrizio-Caruso, 2023-07-15:
More information on how to detect key presses:
[image]

Fabrizio-Caruso, 2023-07-15:
@suborb, is the horrific snippet of code you posted in #2360 the one you have implemented in the library? Somehow it looks similar to what the expert in the MSPP site describes.

suborb, 2023-07-15:
I've just checked - it's in the library and is used by the joystick() routine.

Fabrizio-Caruso, 2023-07-15:
@suborb, so the expert says that that code does not have auto-repeat and that we should do something else:
[image]

Fabrizio-Caruso, 2023-07-15:
So it says the right code is in the CP/M BIOS. I have been asking him to give us a snippet of code, otherwise we will have to figure out where the exact code is...

suborb, 2023-07-15:
For reference I think I found something in this: `/Microbee/Software/Disks_original/CPM3/cpm3_1.1.0/microbee_cpm3-1.1.0_source.tar.gz, Jul 6, 2011` within, I think the retro repo.

Fabrizio-Caruso, 2023-07-15:
He is suggesting the PJB's CP/M 2 BIOS code but I suppose CP/M 3 must have the good code, as well.

Fabrizio-Caruso, 2023-07-20:
More information from the expert (who has looked at the current `getk`)
[image]