z88dk/z88dk #924: [x1][bug][memory/overflow?] X1 used to work and it is now broken: Crash at start-up

Fabrizio-Caruso, 2018-08-30:
X1 used to work. 
Now it seems it is broken: Crash at start-up if I use the FULL version of my game.

I compile it with:
```
	$(Z88DK_PATH)$(MYZ88DK) +x1 \
	-D__X1__ \
	-DFULL_GAME \
	-DBETWEEN_LEVEL -DEND_SCREEN \
	-DCONIO -DNO_INIT_GRAPHICS \
	-create-app -o $(BUILD_PATH)/FULL_x1.bin -vn -lndos \
	$(SOURCE_PATH)/end_screen.c \
	$(FULL_FILES)
```

BUT it can run the TINY version.

Tested with Takeda's emulator (X1Millenium emulator is not accurate)

Fabrizio-Caruso, 2018-08-31:
Could I place the code elsewhere with zorg and skip the nasty video ram? Would zorg=0x8000 do?

suborb, 2018-08-31:
After examination it looks like a copy/paste error when the port was created. 

I've removed it and now it runs (but super fast!)

closed 2018-08-31 by suborb

Fabrizio-Caruso, 2018-08-31:
Thanks a lot! Great! I will tune the speed!

Fabrizio-Caruso, 2018-09-04:
@suborb Great work! I confirm that the full version of my game loads and runs nicely! Thanks a lot!

commit 531dd005 (2018-09-02): Fix #924: The store to $3000 is just a copy/paste problem