z88dk/z88dk #925: [nascom][input] Input is not working

Fabrizio-Caruso, 2018-08-30:
Input is barely working:
1. the keyboard is not resposive: multiple key-presses not detected;
2. keys kept down not detected more than once

Tested with virtualnascom emulator.

Compiled with:
```
	$(Z88DK_PATH)$(MYZ88DK) +nascom $(SCCZ80_OPTS) -clib=ansi -vn -lndos \
	-D__NASCOM__ \
	-DSOUNDS \
	-DFULL_GAME \
	-DEND_SCREEN -DBETWEEN_LEVEL \
	-DCONIO -DNO_INIT_GRAPHICS \
	-lndos -create-app -o $(BUILD_PATH)/FULL_nascom_32k.prg \
	$(SOURCE_PATH)/end_screen.c \
	$(FULL_FILES)
```

suborb, 2018-08-30:
I think this is #590 ?

Fabrizio-Caruso, 2018-08-31:
@suborb indeed! Sorry, I though that issue was sold and this is a regression. It was never fixed.
I am closing this.

closed 2018-08-31 by Fabrizio-Caruso