z88dk/z88dk #933: [ti8x][bug/question] Invalid argument when loaded (Wabbiemu)

Fabrizio-Caruso, 2018-09-01:
If I am right TI83 Plus should be able to execute external programs without any shell.

This seems not to work with the TI8X target.
In Wabbiemu I get "Invalid argument" if I try to run the .8xp binary.

I compile it with:
```
	$(Z88DK_PATH)$(MYZ88DK) +ti8x \
	-subtype=asm \
	$(SCCZ80_OPTS) -D__TI8X__ \
	-DZ88DK_JOYSTICK \
	-DFORCE_XSIZE=32 \
	-clib=ansi -pragma-define:ansicolumns=32 \
	-vn \
	-DFULL_GAME  \
	-lndos \
	-DCONIO -DNO_INIT_GRAPHICS \
	-create-app  \
	$(FULL_FILES)
	mv $(SOURCE_PATH)/../a.8xp $(BUILD_PATH)/FULL_ti8x.8xp	
```

I have tried to build an Ion-binary but I have not been able to run it.

Fabrizio-Caruso, 2018-09-02:
@zx70 should the -subtype=asm option produce a binary that can be loaded without a Shell?
Can TI83+ models load asm files directly?

zx70, 2018-09-02:
https://github.com/z88dk/z88dk/wiki/Platform---TI-Calculators
for the ti83, try with -subtype=9 and for the ti83plus with -startup=10, but I've never been able to pass a small enough program. 
the way to run them should be via the asm(...   directive

zx70, 2018-09-02:
sorry,   I correct myself,  the right syntax is -subtype=asm

zx70, 2018-09-02:
the ti83 plus does not have a native assembler iirc, in any case the asm( directory ecpects a binary program

Fabrizio-Caruso, 2018-09-03:
Under Wabbitemu, the emulated TI83 Plus shows "ASM("
How do I transfter the binary?
Any attempt fails with an "Invalid argument error".

Fabrizio-Caruso, 2018-09-14:
Thanks @zx70 for explaining the load procedure! :-)
An important missing missing detail was that the name of the binary should not be the default "a".

closed 2018-09-14 by Fabrizio-Caruso