z88dk/z88dk #934: [x07][bug] Cannot load binary - Load stuck at second part

Fabrizio-Caruso, 2018-09-01:
It seems that the produce .cas files cannot be loaded with a recent version of Takeda's emulator
(I know this is reported in the doc but it could well be caused by improved emulation accuracy).

The first .cas (the loader) is loaded with no problem: it finds the file and returns to the prompt.
The emulator gets stuck with the second part after typing "RUN" and selecting the file to load.

I compile with:
```
	$(Z88DK_PATH)$(MYZ88DK) +x07 $(SCCZ80_OPTS) \
	-Cz--audio \
	-DFORCE_BOMBS_NUMBER=2 \
	-DFORCE_GHOSTS_NUMBER=6 \
	-DLESS_TEXT \
	-DNO_BLINKING \
	-DNO_CONTROL_INSTRUCTIONS \
	-DNO_RANDOM_LEVEL \
	-DNO_SET_SCREEN_COLORS \
	-D__X07__ \
	-vn \
	-DCONIO_VT52 \
	-DNO_COLOR \
	-DTINY_GAME \
	-lndos \
	-DCONIO -DNO_INIT_GRAPHICS \
	$(TINY_FILES) \
	-create-app 
```

zx70, 2018-09-01:
could you try with an older takeda's version?    please use short test programs in your first attempts and remember to detach the first tape file before opening the second one:

https://www.z88dk.org/wiki/doku.php?id=platform:x07

Fabrizio-Caruso, 2018-09-01:
@zx70, my binary is barely 4k (super-tiny version). I will try with something even smaller.

I know the doc mentions an old Takeda emulator. Could you please send me the old emulator?
Even if it worked with the old emulator, it also fails with a recent Mame version exactly as it fails with the Take emulator. I suspect two recent emulators behaving the exact same way, may be more accurate than an old one. Of course this is not a proof of a real problem.

zx70, 2018-09-01:
let me underline that it is highly improbable that the problem is consequence of the  increased emulator accuracy:  the first block is a byte loader written in basic which reads any single byte from the tape file (the rom supports buffered input, the tape file is split into blocks and the  tape motor is blocked when necessary by the rom),  the possible improvement could be to provide the wav equivalent but this, if working,  would be simply a workaround for one of the many existing emulator limits

Fabrizio-Caruso, 2018-09-01:
Bothe Mame 0.196 and the latest Takeda load the first block correctly (even the .wav) showing something like "found ...".
The result is a frozen screen at the end.
In order to exclude an issue with my code, I am going to test with something much smaller like a "hello world" test.

Fabrizio-Caruso, 2018-09-01:
Hello World test failed on both Mame 0.196 and recent Takeda.
(I have also tried to eject cassette after first part is loaded).

Both can load the first part.
Both are frozen after loading the second part.

I compiled:
```
#include <stdio.h>
int main()
{
	printf("Hello world\n");
	return 0;
}
```
with 
```
	$(Z88DK_PATH)$(MYZ88DK) +x07 $(SCCZ80_OPTS) \
	-Cz--audio \
	$(SOURCE_PATH)/../experiments/hello.c \
	-create-app 
```

zx70, 2018-09-11:
Ok, I looked what I got for the X07 'till now.   First of all let me say that the example code you pasted is failing also on the old Takeda emulator's version  :(      The WAV format was an attempt to extend the compatibility to mame but for some reason it doesn't work up to the end.

As written on the wiki:
NOTE: This target port is highly experimental, any kind of feedback is welcome, please post your comment in the z88dk.org forum.

The fact is that also the library is just stubbed and probably not complete enough to permit real programs to run,  I'll paste a working code sample soon.

Fabrizio-Caruso, 2025-10-18:
@zx70 Inufuto has published his own massively multi-target games that he compiles with his universal C-like compiler/Assembler.

Maybe this could help to unblock/complete the development for the x07:
https://github.com/inufuto/Cate_examples/tree/main/x07

@suborb For the other targets and not yet supported targets:
https://github.com/inufuto/Cate_examples/

Fabrizio-Caruso, 2025-12-11:
@suborb just to inform you that a new coding jam on this machine has been launched by "[Retro Programmers United for Obscure Systems](https://www.facebook.com/groups/rpufos/)". The actual coordination of the devs will be done in the group's discord channel
So a few people may use Z88DK on this target or just code for this machine in Assembly or BASIC. So more technical information on this machine may soon be available. I hope this can help you improve the support for this target.

feilipu, 2026-11-08:
Fixed in d5aad86c67 (`(x07) Get something loading`).

The two-block CAS loader is gone. `x07_exec` embeds the binary in a BASIC REM, and `x07_crt0.asm` decompresses it.

closed 2026-11-08 by feilipu