z88dk/z88dk #937: z80asm discussion: change --cpu option

pauloscustodio, 2018-09-02:
I think it would make the toolchain easier to use if we use the same options for similar purpose in different tools.

In this context I'm considering renaming z80asm's  "--cpu=xxx" to "-mXXX", as is used in the other tools.
We also have "-m" for map file, but I think the map file should be generated always as a complement to the binary file.

I volunteer to replace the usage in the z88dk library.

Any objections?

zx70, 2018-09-02:
I'm not sure about the map option,  I think it is the trickiest part..  isn't it enabled also by zcc, etc ?

suborb, 2018-09-02:
Certainly zcc parses separately for "-m", "-mz80" etc, so there's no conflict between -m being used for separate features.

Generally yes, would be good to have the same option across the tools.

aralbrec, 2018-09-02:
The map file is currently always generated automatically in the tmp directory for appmake.

zcc's -m option copies it to the user directory.  I'm not sure if we always want that to happen as zcc is set up now only to copy intermediate files the user request.

But zcc already distinguishes between "-m" and "-mxxxx" so a simple change at [line 497 of zcc.c](https://github.com/z88dk/z88dk/blob/master/src/zcc/zcc.c#L497) will change zcc for z80asm.

The library builds (newlib and classic) will have some --cpu options in them.

zx70, 2018-09-11:
so.. I understand that the overall answer is "no selevant objections", right ?

pauloscustodio, 2018-09-11:
Right. I'll do it when I have a spare time.

pauloscustodio, 2018-10-25:
Add -mCPU option as synonym to --cpu CPU; accept a single -m to generate a map file, as before.
Replace --ti83plus by -mti83 and -mti83plus.
Change --use-lib to --link-lib (long version of -i).
Add --out-dir as synonym to -O (capital oh).
Change --date-stamp to --update (long version of -d).
Remove message with total number of errors at the end.

commit 44fe7aa7 (2018-10-25): Fix #937: add -mCPU as synonym to --cpu=CPU

Add -mCPU option as synonym to --cpu CPU; accept a single -m to generate a map file, as before.
Replace --ti83plus by -mti83 and -mti83plus.
Change --use-lib to --link-lib (long version of -i).
Add --out-dir as synonym to -O (capital oh).
Change --date-stamp to --update (long version of -d).
Remove message with total number of errors at the end.
closed 2018-10-25 by pauloscustodio