z88dk/z88dk #943: [einstein][appake] No Sector 1 error

Fabrizio-Caruso, 2018-09-04:
When building my game for the einstein I get a malformed disk: which produces a "NO Sector 1:" error

I have compiled my game with:
```
	$(Z88DK_PATH)$(MYZ88DK) +cpm -subtype=einstein -leinstein \
	$(SCCZ80_OPTS) \
	-pragma-define:ansicolumns=32 \
	...
	-clib=ansi \
	-create-app -o$(BUILD_PATH)/FULL_einstein.bin \
	$(SOURCE_PATH)/end_screen.c \
	$(FULL_FILES)
```

@suborb 
[image]

suborb, 2018-09-04:
I think it's a program size thing - looking at the directory entry for adv_a (which uses up the same disc space) it looks like the RC field is being calculated incorrectly and this is the cause of the error.

suborb, 2018-09-04:
Hopefully sorted now, I can now load adva from disc so cross chase should be ok.

closed 2018-09-04 by suborb

Fabrizio-Caruso, 2018-09-05:
Things are worse now.
I can load APP.COM but it crashes with a corrupted screen.
[image]

Fabrizio-Caruso, 2018-09-05:
This is memory-related because if I compile with sdcc and optimizations, I get a working game.

suborb, 2018-09-09:
I've fixed it again (RC field again!) - I think it was only loading 8k of the program so it was potluck as to whether things appeared to work.

commit 8a8c807e (2018-09-04): Fix #943. There are 8 extents in a directory entry. Not 16. Doh
commit 44fa3704 (2018-09-09): Fix #943: And the RC field was still being calculated incorrectly