z88dk/z88dk #944: [mtx][appmake][bug] appmake produces malfunctioning .wav files

Fabrizio-Caruso, 2018-09-04:
The .wav files produced by MTX are no recognize by any Mame/Mess version.

I compile a hello world example with
```
	$(Z88DK_PATH)$(MYZ88DK) +mtx \
	-vn -lndos \
	$(SOURCE_PATH)/../experiments/hello.c \
	-create-app	
``` 
(I tried the startup=2 options with the same result)

I get a .WAV file which I try to load with multiple MAME/MESS versions.

I do:
``` 
POKE 64122,0
NEW
LOAD""
``` 
(tried without POKE and/or LOAD"A")

Nothing is detected while loading. It gets to the end of the tape and does nothing.

Remark: The MTX file works under an old Mess version. So I assume that this is an appmake-only issue.

zx70, 2018-09-12:
I tend to blame the MAME emulator, you can find plenty of audio files here, which are not loadable with it:
http://www.mtxworld.dk/download.php

moreover the sample "sounds" too slow while being played by MAME

Fabrizio-Caruso, 2018-09-12:
@zx70  Is there a recent version of an alternative emulator that can load Z88DK wav files?

zx70, 2018-09-12:
to be honest I don't remember,  I could have tested the validity of the WAV files with a conversion tool or in another way because I remember a lot of time spent around MESS for this kind of problem

Fabrizio-Caruso, 2018-09-17:
@zx70  The wav is OK and Mame sucks at emulating MTX tapes.
This has been proved by Martin Allcorn at 
https://www.facebook.com/groups/2443017753/permalink/10156838110697754/?comment_id=10156851432667754&reply_comment_id=10156851457987754&notif_id=1537187306380346&notif_t=group_comment
[image]

closed 2018-09-17 by Fabrizio-Caruso

zx70, 2018-09-18:
wonderful, so I guessed the right frequencies