z88dk/z88dk #947: [ti86] hundreds of out of range warnings

Fabrizio-Caruso, 2018-09-05:
When compiling for the ti86 I am getting hundreds of lines with "out of range" errors.

I compile:
```
/cygdrive/c/z88dk/bin/zcc.exe +ti86 \
-subtype=asm \
-O3 -D__TI86__ \
-DZ88DK_JOYSTICK \
-DFORCE_XSIZE=32 \
-clib=ansi -pragma-define:ansicolumns=32 \
-vn \
-DFULL_GAME  \
-lndos \
-DCONIO -DNO_INIT_GRAPHICS \
-create-app  \
cross_lib/display/display_macros.c cross_lib/display/init_images.c cross_lib/input/input_macros.c cross_lib/text/text_macros.c chase/move_player.c chase/ghost.c  chase/level.c chase/character.c chase/game_text.c chase/main.c chase/strategy.c chase/item.c chase/skull.c chase/bullet.c   chase/horizontal_missile.c chase/rocket.c
```
I get:
```
Warning at file 'c:\Z88DK\lib\config\..\..\\lib\target\ti86\classic\ti86_crt0.asm' line 215: integer '70237' out of range
Warning at file 'c:\Z88DK\lib\config\..\..\\lib\target\ti86\classic\ti86_crt0.asm' line 256: integer '70240' out of range
Warning at file 'c:\Z88DK\lib\config\..\..\\lib\target\ti86\classic\ti86_crt0.asm' line 257: integer '70244' out of range
Warning at file 'c:\Z88DK\lib\config\..\..\\lib\target\ti86\classic\ti86_crt0.asm' line 258: integer '70242' out of range
Warning at file 'c:\Z88DK\lib\config\..\..\\lib\target\ti86\classic\ti86_crt0.asm' line 274: integer '70240' out of range
Warning at file 'c:\Z88DK\lib\config\..\..\\lib\target\ti86\classic\ti86_crt0.asm' line 275: integer '70244' out of range
Warning at file 'c:\Z88DK\lib\config\..\..\\lib\target\ti86\classic\ti86_crt0.asm' line 276: integer '70242' out of range
Warning at file 'c:/Z88DK/lib/../lib/crt/classic/crt_section.asm' line 17: integer '70133' out of range
Warning at file 'c:/Z88DK/lib/../lib/crt/classic/crt_section.asm' line 18: integer '70134' out of range
Warning at file 'c:/Z88DK/lib/../lib/crt/classic/crt_section.asm' line 28: integer '70239' out of range
Warning at file 'c:/Z88DK/lib/../lib/crt/classic/crt_section.asm' line 31: integer '70137' out of range
Warning at file 'c:/Z88DK/lib/../lib/crt/classic/crt_section.asm' line 33: integer '70147' out of range
Warning at file 'c:/Z88DK/lib/../lib/crt/classic/crt_section.asm' line 35: integer '70157' out of range
Warning at file 'c:\Z88DK\lib\config\..\..\\lib\target\ti86\classic\ti86_crt0.asm' line 306: integer '70235' out of range
Warning at file 'cross_lib/display/display_macros.c' line 39: integer '69651' out of range
Warning at file 'cross_lib/display/display_macros.c' line 42: integer '66588' out of range
Warning at file 'cross_lib/display/display_macros.c' line 64: integer '67932' out of range
Warning at file 'cross_lib/display/display_macros.c' line 65: integer '69621' out of range
Warning at file 'cross_lib/display/display_macros.c' line 68: integer '68062' out of range
Warning at file 'cross_lib/display/display_macros.c' line 75: integer '68057' out of range
Warning at file 'cross_lib/display/display_macros.c' line 78: integer '66588' out of range
Warning at file 'cross_lib/display/display_macros.c' line 110: integer '68062' out of range
Warning at file 'cross_lib/display/display_macros.c' line 136: integer '67932' out of range
Warning at file 'cross_lib/display/display_macros.c' line 183: integer '70288' out of range
Warning at file 'cross_lib/display/display_macros.c' line 218: integer '68062' out of range
Warning at file 'cross_lib/display/display_macros.c' line 304: integer '70264' out of range
Warning at file 'cross_lib/display/display_macros.c' line 362: integer '70266' out of range
...
(and many more)
...
```

suborb, 2018-09-05:
See #776

closed 2018-09-05 by suborb

Fabrizio-Caruso, 2018-09-05:
@suborb sorry, you are right (as usual). I should have remembered.