z88dk/z88dk #951: [sdcc][__end_asm] Inline Assembly error only when compiling with sdcc

Fabrizio-Caruso, 2018-09-07:
How can I write inline code that should compile with both sdcc and sccz80?
#asm fails on sdcc
#asm( fails
__asm / __end_asm fail as follows:

```
/cygdrive/c/z88dk/bin/zcc.exe +cpm -startup=2 -lm -ltiki100 \
-compiler=sdcc \
-D__TIKI100__ \
-DNO_PRINT \
-DFULL_GAME \
-DNO_COLOR \
-DALT_MOVE \
-DNO_RANDOM_LEVEL \
-DNO_SLEEP \
-DZ88DK_SPRITES \
-DLESS_TEXT \
-DNO_HINTS \
-DNO_SET_SCREEN_COLORS \
-DANIMATE_PLAYER \
-create-app \
-DNO_INIT_GRAPHICS \
cross_lib/display/display_macros.c cross_lib/display/init_images.c cross_lib/input/input_macros.c cross_lib/text/text_macros.c chase/move_player.c chase/ghost.c  chase/level.c chase/character.c chase/game_text.c chase/main.c chase/strategy.c chase/item.c chase/skull.c chase/bullet.c   chase/horizontal_missile.c chase/rocket.c
cross_lib/display/graphics_mode/../graphics_data/z88dk/z88dk_sprites_definitions.h:969: syntax error: token -> '__endasm' ; column 8
```

suborb, 2018-09-07:
The following will work on both compilers:

```
__asm
__endasm;
```

Note the semi-colon at the end of `__endasm;`

Fabrizio-Caruso, 2018-09-07:
@suborb it fails with the semi-colon, as well:
```
/cygdrive/c/z88dk/bin/zcc.exe +cpm -startup=2 -lm -ltiki100 \
-compiler=sdcc \
-D__TIKI100__ \
-DNO_PRINT \
-DFULL_GAME \
-DNO_COLOR \
-DALT_MOVE \
-DNO_RANDOM_LEVEL \
-DNO_SLEEP \
-DZ88DK_SPRITES \
-DLESS_TEXT \
-DNO_HINTS \
-DNO_SET_SCREEN_COLORS \
-DANIMATE_PLAYER \
-create-app \
-DNO_INIT_GRAPHICS \
cross_lib/display/display_macros.c cross_lib/display/init_images.c cross_lib/input/input_macros.c cross_lib/text/text_macros.c chase/move_player.c chase/ghost.c  chase/level.c chase/character.c chase/game_text.c chase/main.c chase/strategy.c chase/item.c chase/skull.c chase/bullet.c   chase/horizontal_missile.c chase/rocket.c
cross_lib/display/graphics_mode/../graphics_data/z88dk/z88dk_sprites_definitions.h:969: syntax error: token -> '__endasm' ; column 8
make: *** [Makefile:2013: tiki100] Error 1
```

Fabrizio-Caruso, 2018-09-07:
The following code compiles only with SCCZ80:
```
	__asm
	._sprites

	defb    8,8
	defb %"--------"
	defb %"--------"
	defb %"--------"
	defb %"--------"
	defb %"--------"
	defb %"--------"
	defb %"--------"
	defb %"--------"

	defb    8,8
	defb    @00011000
	defb    @00100100
	defb    @00000000
	defb    @01100110
	defb    @10011001
	defb    @00000000
	defb    @00100100
	defb    @01100110

	defb 8,8
	defb %"--####--"
	defb %"-#----#-"
	defb %"#--##--#"
	defb %"#-#----#"
	defb %"#-#----#"
	defb %"#--##--#"
	defb %"-#----#-"
	defb %"--####--"

	__endasm;
```

suborb, 2018-09-07:
Ah, now I understand your problem (I think).

sccz80 supports defining assembler outside of the scope of functions. sdcc only supports inline assembler - i.e. within functions.

Your best bet here is to put the font assembler into a separate .asm file as you did for the font.

aralbrec, 2018-09-07:
Yes that's the problem here.

An issue that can pop up is that the zsdcc-produced assembler is in asz80 form and must pass through a translation step to standard zilog mnemonics.  This is done by copt, a regular expression engine, and as it doesn't understand z80 syntax it can mistakenly change something.  It won't lead to incorrect code generation but rather to syntax errors.  As an example, there was recent trouble with $ symbols in quoted strings and the problem still exists in single quotes but this is not something the compiler produces.  Another recent problem that was fixed was that copt couldn't handle assembly comments; this is again something the compiler does not produce but people do.   Another problem is that copt cannot handle non-ascii characters (like in comments).

It's always preferable to have asm in separate asm files rather than inlined for several reasons, this being one attached to zsdcc.

closed 2018-09-08 by suborb

Fabrizio-Caruso, 2019-09-25:
This issue is still valid as of today :-(

Fabrizio-Caruso, 2019-09-25:
@aralbrec How can I transform my included file into a separated asm file?
With SCCZ80 I use:
https://github.com/Fabrizio-Caruso/CROSS-CHASE/blob/master/src/cross_lib/display/graphics_data/z88dk/z88dk_sprites_definitions.h
I have created a .asm file with exactly the content in the __asm block:
```
.sprites
defb ...
...
__end_asm;
```
but it won't compile and produce errors
```
cross_lib/display/display_macros.c:190: error 42: pointer + pointer invalid
```

feilipu, 2019-09-25:
You might want to try using the other traditional syntax for SCCZ80?
It might be more robust.

And your `__asm` and `__end_asm` tags need to be in the leftmost column.

There is a discussion in this issue #1268, which might be useful. SDCC does a random thing to make its inline assembly syntax work, and SCCZ80 is not going to copy it.

Fabrizio-Caruso, 2019-09-26:
@feilipu  moving __asm and __end_asm to column 0 does not fix the problem:
```
cross_lib/display/graphics_data/z88dk/z88dk_sprites_definitions.h:969: syntax error: token -> '__endasm' ; column 8
make: *** [Makefile:1433: zx81_16k_wrx_zsdcc] Error 1
```
I want to get this to work with ZSDCC with either included .h file or .asm.
So far, none works.

feilipu, 2019-09-26:
You can try two different options.
1. (preferable) make a `.asm` file containing the bytes you want to define, and add it to a `.lst` file containing all your c and asm sources.
2. (less preferable) use defines to select the inline assembly syntax dialects for [sccz80](https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/proto/arch/yaz180.h#L122) and [sdcc](https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/proto/arch/yaz180.h#L159).  There is also the [`im2.h`](https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/proto/im2.h) example to look at too, where I got inspiration for using the respective dialects.

suborb, 2019-09-26:
I think the issue here is that there's an assembler block outside of a function scope. SDCC doesn't like this, so 1. is the way to go.

However, you could just wrap the asm block in a void __naked function. As long as it has non-static scope SDCC shouldn't dead-code eliminate it.

suborb, 2019-09-26:
> There is also the [`im2.h`](https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/proto/im2.h) example to look at too, where I got inspiration for using the respective dialects.

I think SDCC now has a `__asm__()` function style invocation that works in the same way as sccz80 `asm()`?