z88dk/z88dk #960: [PX8][sprites] Extremely slow sprites on PX8 (but ok on PX4)

Fabrizio-Caruso, 2018-09-10:
Sprite graphics is sluggish on the PX8 (about one second to draw a small 8x8 sprite).
Even drawing "Hello World" with sprites (one 8x8 sprite per character) would take several seconds.

I compile with:
```
	$(Z88DK_PATH)$(MYZ88DK) +cpm -subtype=px32k \
	-D__PX8__ \
	-DFULL_GAME \
...
	-subtype=px32k  -Cz--32k \
	-create-app \
	-vn -lndos \
	$(CROSS_LIB_PATH)/display/alt_print/memory_mapped_alt_print.c \
	$(FULL_FILES)
```

zx70, 2018-09-11:
It is expected, the PX8 as well as very few other targets (eg the Otrona Attachè) have a very basic sprite support.   Direct graphics memory access is not possible on the PX8, a secondary CPU must be involved and being tricky I left it behind;  the manuals provide an example on how to load/save the graphics memory which may be helpful on developing a full library.

Fabrizio-Caruso, 2018-09-11:
You are right.
Indeed Otrona Attache has even slower sprites...

zx70, 2018-09-11:
the Attachè direct video memory access is tricky for other reasons ( memory organisation split into nibbles), same solution used as temporary workaround.

suborb, 2018-09-14:
Another drawback of the Attache is that there's no way (using the firmware) to peek the screen.

The Mame source suggests that an in() should be able to read the VRAM - the manual has details on how to compute the latch value. However until it's confirmed that it works on real hardware I'm slightly reluctant to invest time in implementing it.

zx70, 2018-09-15:
I have something, I'll check what I got, but I suspended the work just because it was tricky and I had no users

zx70, 2018-10-08:
I've just commited code for a better graphics support on the Otrona Attachè, I'm now going try to kick the code the mono sprites (now already slightly faster) and make them 'run' fast enough to play microman.

zx70, 2018-10-08:
> However until it's confirmed that it works on real hardware I'm slightly reluctant to invest time in implementing it.

I borrowed the BIOS code trying to twist the IN/OUT core components at minimum, it should be valid also on the real hardware.    What could upset you is the three level nest of SMC, I think it is a  Guinnes Record !

https://github.com/z88dk/z88dk/blob/master/libsrc/graphics/attache/plot_setup.asm
https://github.com/z88dk/z88dk/blob/master/libsrc/graphics/attache/w_xorpixl.asm

zx70, 2019-02-03:
Fabrizio, the new putsprite function for the PX8 is now ready.   It is not very fast but much faster than before.

zx70, 2019-02-04:
I'm trying to use a powerful function available on the PX8 to make the sprites work much faster.
Sadly I'm also dealing with an emulator problem preventing the "XOR" operation from working correctly on the display functions, I sent Takeda an email with a couple of examples which hopefully will help.

zx70, 2019-02-15:
Reasonably good. I'm closing the issue

closed 2019-02-15 by zx70