z88dk/z88dk #962: [getch/kbhit][sdcc][enterprise/mz2500] When compiling with sdcc, getch and kbhit are not found

Fabrizio-Caruso, 2018-09-11:
When compiling with sdcc, getch and kbhit are not found.

Example:
enterprise_buffered:
```
	$(Z88DK_PATH)$(MYZ88DK) +enterprise $(SCCZ80_OPTS) \
	-compiler=sdcc \
	-vn \
...
	$(CROSS_LIB_PATH)/display/alt_print/memory_mapped_alt_print.c \
	$(FULL_FILES) \
	-create-app
```
Removing "-compiler=sdcc" allows the compilation.

I have tested this issue with enterprise, mz2500 but it may be present in many other targets.

suborb, 2018-09-11:
I can only reproduce this on those two targets if conio.h isn't included.

suborb, 2018-09-15:
@Fabrizio-Caruso was that the cause of the problem?

Fabrizio-Caruso, 2018-09-16:
I was doing something stupid here but I was surprised that I was getting a different error.
SDCC outputs errors about unused functions.
SCCZ80 is silent about unused functions.
So this is not a bug but a problem in my Makefile (now fixed).

closed 2018-09-16 by Fabrizio-Caruso