z88dk/z88dk #963: [Sharp OZ] Crash at start up (Exception... ACCESS VIOLATION...)

Fabrizio-Caruso, 2018-09-11:
I am testing the experimental Sharp OZ target. I could not find any emulator for the OZ-700.
I am using oz750 from Mame. Is it the right one?

I successfully compile and use makewzd to convert the binary:
```
	$(Z88DK_PATH)$(MYZ88DK) +oz $(SCCZ80_OPTS) \
	-vn -lndos \
...
	$(TINY_FILES)
	$(TOOLS_PATH)/z88dk/oz/makewzd.exe a
```
and I get:
```
C:\Emulation\TEST\0.196>mame64.exe oz750 -uimodekey DEL -window
-----------------------------------------------------
Exception at EIP=000000000190b865 (oz750_state::quickload_loadoz750(device_image_interface&, char const*, int)+0x0135): ACCESS VIOLATION
While attempting to read memory at 0000000000000050
-----------------------------------------------------
RAX=0000000000000000 RBX=000000000d2800a0 RCX=000000001d963260 RDX=0000000003298518
RSI=0000000010440790 RDI=0000000000000025 RBP=000000000d1a8090 RSP=000000000d1a8010
 R8=0000000000000000  R9=0000000000000007 R10=000000000d2800a0 R11=000000000d1a8290
R12=a19ec78b1c4ddcea R13=000000000d1a8280 R14=000000001044eb60 R15=000000000000001c
-----------------------------------------------------
Stack crawl:
  000000000d1a8340: 000000000190b865 (oz750_state::quickload_loadoz750(device_image_interface&, char const*, int)+0x0135)
  000000000d1a8420: 0000000002cec373 (snapshot_image_device::process_snapshot_or_quickload(void*, int)+0x0073)
  000000000d1a84d0: 000000000371efd8 (device_scheduler::timeslice()+0x0668)
  000000000d1a8570: 00000000036db858 (running_machine::run(bool)+0x0118)
  000000000d1af210: 0000000001f338aa (mame_machine_manager::execute()+0x020a)
  000000000d1af4c0: 0000000001fa1a84 (cli_frontend::start_execution(mame_machine_manager*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)+0x03f4)
  000000000d1af630: 0000000001fa1f45 (cli_frontend::execute(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&)+0x0045)
  000000000d1af690: 0000000001f3185a (emulator_info::start_frontend(emu_options&, osd_interface&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&)+0x002a)
  000000000d1afe50: 00000000073bf43d (main+0x017d)
  000000000d1aff20: 00000000004013c7 (__tmainCRTStartup+0x0247)
  000000000d1aff50: 00000000004014fb (mainCRTStartup+0x001b)
  000000000d1aff80: 00007ffe11d23034 (BaseThreadInitThunk+0x0014)
  000000000d1affd0: 00007ffe145d1431 (RtlUserThreadStart+0x0021)
```

suborb, 2018-09-11:
That one needs to be raised against MAME - the command line doesn’t reference the file generated by z88dk

Fabrizio-Caruso, 2018-09-11:
@suborb I have attached the file throught the quickload option after starting Mame.

Fabrizio-Caruso, 2018-09-11:
There is no -cart or -cart1 option. How do you run the binary otherwise?

Fabrizio-Caruso, 2018-09-11:
Any better emulator that can be run on Windows 64 bit or Linux 32/64 bit?

Fabrizio-Caruso, 2018-09-11:
Maybe I got it! The binary is 8700 bytes! Too big maybe for this target.

Fabrizio-Caruso, 2018-09-11:
No, :-( It fails with a smaller than 8k file (7000). So the issue is not related to the size.

suborb, 2018-09-11:
Does a file not created by z88dk load correctly?

Fabrizio-Caruso, 2018-09-11:
You mean the .wzd file? Is makewzd doing something wrong?

I create the .bin with Z88DK. I follow Z88DK and run makewzd (included in Z88DK) and I try to load the result into the emulated machine.

suborb, 2018-09-11:
That's what I'm wondering. It's mame crashing here whilst trying to read the file. 

There's obviously something in the file it objects to, so if a 3rd party one works then we might be able to figure out what's going wrong.

Fabrizio-Caruso, 2018-09-11:
Ok! I will try to test Mame with other files (known to be OK)

zx70, 2018-09-11:
https://web.archive.org/web/20030205125236/http://www.ozdev.com:80/

zx70, 2018-09-11:
an emulator called voz existed for windows

zx70, 2018-09-12:
I did some experiment with MAME and I agree with Dom about the problem being on the emulator.
The strange thing is that the barebone code looks accurate, even if the REX6000/Dataslim and the OZ Wizard support are collapsed in a single driver file the Quickload components are target specific.   At a first glance the expected format for the Quickload functionality on the oz700 is WZD, but the emulator crashes when I try it.

Fabrizio-Caruso, 2018-09-12:
I am not able to start a simple "hello world" example (.wzd attached here) with the vOZ emulator 0.2:
```
...
int main(void) {
        printf("Hello world\n");
        while(1) {};
        return 0;
}
```
with
```
	$(Z88DK_PATH)$(MYZ88DK) +oz $(SCCZ80_OPTS) \
	$(SOURCE_PATH)/../experiments/hello.c 
	$(TOOLS_PATH)/z88dk/oz/makewzd.exe a
```

I start the emulator.
Drag the rom into it and then the .wzd file.
Nothing happens. I enter a menu with F11 but I cannot do anything useful.
What am I missing?

[oz_hello.zip](https://github.com/z88dk/z88dk/files/2376871/oz_hello.zip)

zx70, 2018-09-18:
Here's what I've done.
- Launch vOZ
- Drag oz750.rom on the emulator's windows
- choose the memory size (256K in my case)
- press ENTER (CTRL key) to exit the display settings
- press ENTER (CTRL key) again
- (optional) Press F6 for brighter display
- Drag oz_hello.wzd on the emulator's windows
- Choose the program 'slot' from the emulator's dialog window
- Press F5 to select the custom program

[image]

Fabrizio-Caruso, 2018-09-19:
After dragging oz750.rom I can select 256 and then I simply get a black screen and no key responds.
[image]

Could you please tell which exact emulator versions you are using?

Fabrizio-Caruso, 2018-09-19:
@zx70 it seems to work in plain stdlib.
If I try to load some code compiled with sprites (size 8k) I get a frozen "a" at the top-left of the screen.

Fabrizio-Caruso, 2018-09-29:
@zx70 I confirm that sprites are broken on this target.

I have attached the binary the causes the issue.

The binary is compiled with:
```
	
oz_tiny:
	$(Z88DK_PATH)$(MYZ88DK) +oz \
	$(SCCZ80_OPTS) \
[TINY_oz.zip](https://github.com/z88dk/z88dk/files/2430847/TINY_oz.zip)
[TINY_oz.zip](https://github.com/z88dk/z88dk/files/2430848/TINY_oz.zip)

	-vn -lndos \
...
	$(TINY_FILES) \
	$(CROSS_LIB_PATH)/display/alt_print/memory_mapped_alt_print.c
	$(TOOLS_PATH)/z88dk/oz/makewzd.exe a
	rm a.bin
	mv a.wzd $(BUILD_PATH)/TINY_oz.wzd
```