z88dk/z88dk #965: [laser500] It works! Only minor issue: wrong colors

Fabrizio-Caruso, 2018-09-12:
I have tested the binary with https://github.com/nippur72/laser500emu.git

The game works including the keyboard.

[FULL_vz500.zip](https://github.com/z88dk/z88dk/files/2374103/FULL_vz500.zip)

The only remaining problem is the color mapping.

[image]

suborb, 2018-09-12:
The mapping table is:

```
        defb    $0      ;BLACK -> BLACK
        defb    $1      ;BLUE -> BLUE
        defb    $2      ;GREEN -> GREEN
        defb    $3      ;CYAN -> CYAN
        defb    $4      ;RED -> RED
        defb    $5      ;MAGENTA -> MAGENTA
        defb    $e      ;BROWN -> BRIGHT YELLOW
        defb    $7      ;LIGHTGRAY -> BRIGHT GREY
        defb    $8      ;DARKGRAY -> DARK GREY
        defb    $9      ;LIGHTBLUE -> BRIGHT BLUE
        defb    $a      ;LIGHTGREEN -> BRIGHT GREEN
        defb    $b      ;LIGHTCYAN -> BRIGHT CYAN
        defb    $c      ;LIGHTRED -> BRIGHT RED
        defb    $d      ;LIGHTMAGENTA -> BRIGHT MAGENTA
        defb    $6      ;YELLOW -> YELLOW
        defb    $f      ;WHITE -> WHITE
```

Based on these same colours:

```
static const rgb_t vt_colors[] =
{
	rgb_t::black(),
	rgb_t(0x00, 0x00, 0x7f),  /* blue */
	rgb_t(0x00, 0x7f, 0x00),  /* green */
	rgb_t(0x00, 0x7f, 0x7f),  /* cyan */
	rgb_t(0x7f, 0x00, 0x00),  /* red */
	rgb_t(0x7f, 0x00, 0x7f),  /* magenta */
	rgb_t(0x7f, 0x7f, 0x00),  /* yellow */
	rgb_t(0xa0, 0xa0, 0xa0),  /* bright grey */
	rgb_t(0x7f, 0x7f, 0x7f),  /* dark grey */
	rgb_t(0x00, 0x00, 0xff),  /* bright blue */
	rgb_t(0x00, 0xff, 0x00),  /* bright green */
	rgb_t(0x00, 0xff, 0xff),  /* bright cyan */
	rgb_t(0xff, 0x00, 0x00),  /* bright red */
	rgb_t(0xff, 0x00, 0xff),  /* bright magenta */
	rgb_t(0xff, 0xff, 0x00),  /* bright yellow */
	rgb_t::white()
};
```

If there's problems can you suggest an alternative?

Fabrizio-Caruso, 2018-09-12:
@suborb @nippur72 
What is wrong is (as shown on @nippur72 's emulator):
- red -> shown as yellow
- yellow -> shown as brown

Fabrizio-Caruso, 2018-09-12:
@suborb @nippur72 
Sorry guys. It is my stupidity. I need to use RED and not redefine COLOR_RED with some constant value.

closed 2018-09-12 by Fabrizio-Caruso