z88dk/z88dk #966: Build failure on OpenBSD: 'uthash.h' file not found

vext01, 2018-09-12:
On today's master:

```
cc -g -O2 -MMD -Wall -I../../ext/uthash/src/ -g -pedantic   -c -o callfunc.o callfunc.c
In file included from callfunc.c:9:
In file included from ./ccdefs.h:19:
./define.h:10:11: fatal error: 'uthash.h' file not found
 #include "uthash.h"
          ^~~~~~~~~~
```

Any ideas?

suborb, 2018-09-12:
Have you pulled down the submodules?

vext01, 2018-09-12:
gah! I bet that's it. I was following the instructions on the wiki, which still reference svn!

suborb, 2018-09-12:
Which page is that? I'll correct/zap it as necessary.

vext01, 2018-09-12:
https://www.z88dk.org/wiki/doku.php?id=temp:front

but maybe it's correct, because it's talking about sdcc?

Oh, btw, also noticed a hard-coded 'make' somewhere:
```
$gmake clean
...
gmake[1]: Entering directory '/home/edd/source/z88dk/src/z80asm'
make -C dev/z80asm_lib clean
*** Parse error in /home/edd/source/z88dk/src/z80asm/dev/z80asm_lib: Need an operator in 'MAKE_LIB' (Makefile:19)
*** Parse error: Need an operator in 'endef' (Makefile:30)
*** Parse error: Need an operator in '' (Makefile:35)
*** Parse error: Need an operator in '' (Makefile:36)
*** Parse error: Need an operator in '' (Makefile:37)
*** Parse error: Need an operator in '' (Makefile:38)
*** Parse error: Need an operator in '' (Makefile:39)
*** Parse error: Need an operator in '' (Makefile:40)
*** Parse error: Need an operator in '' (Makefile:41)
*** Parse error: Need an operator in '' (Makefile:42)
*** Parse error: Need an operator in '' (Makefile:43)
*** Parse error: Need an operator in '' (Makefile:44)
```

Notice how BSD make got invoked?

Cheers

vext01, 2018-09-12:
It'd be useful to have build instructions in the top-level README including the mention of a recursive clone.

Cheers :)

suborb, 2018-09-12:
Info is here: https://github.com/z88dk/z88dk#installation - in the top level README

I'll sort out those makes this evening

vext01, 2018-09-12:
Here's another BSD make invocation in the build target:
```
cc -MMD -I. -Ilib -It -g -Wall -ftabstop=4 -std=gnu11 -O3 -I../common -I../../ext/optparse -I../../ext/regex -I../../ext/uthash/src  -g -O2 -c -o ../../ext/regex/regfree.o ../../ext/regex/regfree.c
cc -MMD -I. -Ilib -It -g -Wall -ftabstop=4 -std=gnu11 -O3 -I../common -I../../ext/optparse -I../../ext/regex -I../../ext/uthash/src  -g -O2 -o z80asm listfile.o expr.o modlink.o error_func.o module.o codearea.o symtab.o z80pass.o parse.o main.o scan.o options.o directives.o macros.o zobjfile.o model.o hist.o sym.o opcodes.o z80asm.o libfile.o errors.o lib/array.o lib/class.o lib/minunit.o lib/strhash.o lib/alloc.o lib/str.o lib/dbg.o lib/list.o lib/srcfile.o ../common/optparse.o ../common/objfile.o ../common/fileutil.o ../common/strutil.o ../common/die.o ../../ext/regex/regerror.o ../../ext/regex/regexec.o ../../ext/regex/regcomp.o ../../ext/regex/regfree.o  
../../ext/regex/regerror.o: In function `regerror':
../../ext/regex/regerror.c:(.text+0x2b1): warning: sprintf() is often misused, please use snprintf()
options.o: In function `replace_str':
/home/edd/source/z88dk/src/z80asm/options.c:467: warning: strcpy() is almost always misused, please use strlcpy()
make -C dev/z80asm_lib
*** Parse error in /home/edd/source/z88dk/src/z80asm/dev/z80asm_lib: Need an operator in 'MAKE_LIB' (Makefile:19)
*** Parse error: Need an operator in 'endef' (Makefile:30)
*** Parse error: Need an operator in '' (Makefile:35)
*** Parse error: Need an operator in '' (Makefile:36)
*** Parse error: Need an operator in '' (Makefile:37)
*** Parse error: Need an operator in '' (Makefile:38)
*** Parse error: Need an operator in '' (Makefile:39)
*** Parse error: Need an operator in '' (Makefile:40)
*** Parse error: Need an operator in '' (Makefile:41)
*** Parse error: Need an operator in '' (Makefile:42)
*** Parse error: Need an operator in '' (Makefile:43)
*** Parse error: Need an operator in '' (Makefile:44)
gmake[1]: *** [Makefile:74: z80asm_lib] Error 1
gmake[1]: Leaving directory '/home/edd/source/z88dk/src/z80asm'
gmake: *** [Makefile:97: z80asm] Error 2
```

vext01, 2018-09-12:
> I'll sort out those makes this evening

Thanks!

suborb, 2018-09-12:
Yup, that was going to be the fix! I think it's just z80asm that's affected.

If you're doing an OpenBSD package, the z80asm tests require a whole load of perl modules - the .travis.yml lists them.

Or you can skip them with the appropriate option to build.sh

vext01, 2018-09-12:
I might build a package, but for now I just want to assemble some simple Z80 programs to see how it all hangs together :)

closed 2018-09-12 by suborb

suborb, 2018-09-12:
The top level makefile will now build zsdcc. It uses /tmp which isn't great, but means that getting everything built on linux/bsd is a now a bit easier

commit e1264e04 (2018-09-12): Fix #966: Used $(MAKE) not make - BSD compatiblity