z88dk/z88dk #968: [mz2500][rand()] random outputs always zero (in my game)

Fabrizio-Caruso, 2018-09-13:
I am having a weird issue.
In my game rand() always outputs zero.
I compile my game with:
```
mz2500_tiny_buffered:
	$(Z88DK_PATH)$(MYZ88DK) +mz2500 \
	-SO3 \
	-compiler=sdcc \
...
	-vn \
	-lndos \
	-pragma-include:$(CFG_PATH)/z88dk/zpragma_clib.inc \
	$(TINY_FILES) \
	$(CROSS_LIB_PATH)/display/alt_print/memory_mapped_alt_print.c \
	-create-app 
```
I get the same issue with SCCZ80.

This is not reproducible in a simple test because the issue is not present in a trivial example.

Could this come from the pragmas used?
```
// It next exits
#pragma-define:CLIB_EXIT_STACK_SIZE=0
// Do not initialize BSS
#pragma-define:CRT_INITIALIZE_BSS=0
// move the stack to available memory
#pragma output REGISTER_SP = -1      // must be placed somewhere in the 16k ram
// halt program on exit, no need to play well with basic
#pragma output CRT_ON_EXIT = 0x10001
// eliminate the malloc heap
#pragma output CLIB_MALLOC_HEAP_SIZE = 0
// eliminate the stdio heap (no files can be opened)
#pragma output CLIB_STDIO_HEAP_SIZE = 0
// eliminate unused printf converters
#pragma printf = "%0u"    // placed used converters here, empty string if none
```

Fabrizio-Caruso, 2018-09-13:
@zx70 this only happens under some circumstances, which seem unrelated to the problem.
It may be some sort of overflow in memory. I really don't know.

rand() works fine (range 0000-0x7fff) on the mz2500 if you test just test rand().

I am going to test the cpm target with the very same code and see if it behaves strangely.

This is the function that mis-behaves:
```
#if defined(__NCURSES__) || defined(__GCC_BUFFERED__)
	#define GHOST_RANDOM_CONDITION ((rand()&0x7fff)>slowDown)
#else
	#define GHOST_RANDOM_CONDITION (rand()>slowDown)
#endif
void chaseCharacter(unsigned short slowDown)	
{
	unsigned char i;
	
	for(i=0;i<GHOSTS_NUMBER;++i)
	{
                        if((ghosts[i]._status) && GHOST_RANDOM_CONDITION)	
			{
				deleteGhost(&ghosts[i]);
				moveTowardCharacter(&ghosts[i]);	
			}
	}
}
```

suborb, 2018-09-22:
@Fabrizio-Caruso can you try again after the next nightly? Hopefully being able to load larger programs has made this issue go away.

Fabrizio-Caruso, 2018-09-22:
@suborb, sure! I am the tester! That is my self-appointed role! :-P

Fabrizio-Caruso, 2018-09-23:
@suborb I can confirm that indeed it has fixed this issue!

closed 2018-09-23 by Fabrizio-Caruso

zx70, 2018-09-24:
Wonderful  :)
It's probably very tricky to make it right but a warning in case of overflows in the BSS or DATA space could be nice to have.