z88dk/z88dk #970: [vg5k][sdcc] Build broken 20th of August = OK, 28th of August = KO

Fabrizio-Caruso, 2018-09-14:
My game starts with graphics glitches and then crashes if compiled with sdcc.

sdcc/vg5k has been broken after the 20th of August snapshot and it is confirmed broken on the 28th of August snapshot:
20/08/2018: OK
27/08/2018: KO 
10/09/2018: KO
14/09/2018: KO

I compile with:
```
	$(Z88DK_PATH)$(MYZ88DK) +vg5k \
	-compiler=sdcc \
	--opt-code-size \
	--reserve-regs-iy \
	-pragma-include:$(CFG_PATH)/z88dk/zpragma_clib.inc \
	...
	-create-app -o $(BUILD_PATH)/FULL_vg5k_less_text.prg \
	-DCONIO -DNO_INIT_GRAPHICS \
	$(FULL_FILES) \
```

suborb, 2018-09-14:
A disassembly of the generated binary shows a couple of unexpected uses of ix:

- beeper.asm
- ___sdcc_enter_ix

Both of these should be using iy rather than ix. I suspect your problem is the latter (might also explain the zx81 problems)

I'll look into how they're assembled tonight, the latter should have been assembled with iy but appears not to have been.

aralbrec, 2018-09-14:
The newlib separates the two cases:

suborb, 2018-09-14:
This is a bit confusing (if only because the solution looks counterintuitive)

Classic --IXIY targets use libraries that have been assembled with --IXIY. Newlib defines __SDCC_IX and __SDCC_IY to swap them round rather than rely on the assembler flag.

So, for classic IXIY we need to compile newlib routines with SDCC_IX (so the assembler reads ix) but generates object code using iy.

I'll also need to check where we include newlib routines elsewhere to apply the same inverse logic.

closed 2018-09-14 by suborb

aralbrec, 2018-09-15:
SDCC_IX and SDCC_IY choose between two different c interfaces.  SDCC_IY gives IX to the compiler and assumes the library has exclusive use of IY.  It is made with the IXIY flag but it is not equivalent to SDCC_IX built with IXIY.

I would have thought it would work properly if the sdcc_ix library is used with __SDCC and __SDCC_IX defined.  Then you can choose to build with or without IXIY.

Fabrizio-Caruso, 2018-09-15:
I confirm that this issue is solved but not other sdcc issues in my builds.
For example #817 has not been solved (I still need --max-alloc to get a fully working binary).