z88dk/z88dk #972: Hi-Tech C Libraries Source Code - Informational

feilipu, 2018-09-15:
It may not be news for anyone but me, but I've put the Hi-Tech C v3.09 Libraries Source Code in my YAZ180 repository.

It was quite a bit of work to get them into the modern world, via both a DOSBOX and CP/M DEHUFF tools, and I haven't seen these files elsewhere, so I thought it worthwhile to post them.

Why are they interesting?

There is a[ CP/M](https://github.com/feilipu/yaz180/tree/master/docs/Hi-Tech_C_Libraries/CPM) implementation, with [stdio file access](https://github.com/feilipu/yaz180/tree/master/docs/Hi-Tech_C_Libraries/STDIO) complete.
The fast (but inaccurate) [32 bit float](https://github.com/feilipu/yaz180/tree/master/docs/Hi-Tech_C_Libraries/FLOAT) implementation.
And, [generally understanding the Hi-Tech C code generation](https://github.com/feilipu/yaz180/tree/master/docs/Hi-Tech_C_Libraries/GEN), and what the C compiler expects.

I've also [attached a ZIP of the files to this issue](https://github.com/z88dk/z88dk/files/2385343/Hi-Tech_C_Libraries.zip), in case there is interest to store them here in the docs section too.

aralbrec, 2018-09-15:
Where did you find the source code?

feilipu, 2018-09-15:
I understand that the Library source, and the source for the `c.com` tool, was distributed when the Hi-Tech C compiler was put into the public domain.

```
The HI-TECH Z80 CP/M C compiler V3.09 is provided free of charge for any
use, private or commercial, strictly as-is. No warranty or product
support is offered or implied.

You may use this software for whatever you like, providing you acknowledge
that the copyright to this software remains with HI-TECH Software.
```
The file I started with was [`LIBSRC.EXE`](http://www.retroarchive.org/cpm/cdrom/SIMTEL/HITECH-C/). The annoying thing about the file is that it is a DOS self expanding archive. The expanded archive then needs to be `DEHUFF`ed using the tool in the `Z80V309.EXE` file, on a CP/M machine. So there's a bit of shuffling back and forward. Also of note I changed the file extension of the asm files from `.as` to `.asm`, and copied the C header files into the directories from the compiler executable directory for easy reference.

Yes. I hope that some use can be made the source code, particularly of the CP/M stdio library. The only reason I am looking at  the Hi-Tech C compiler is that it can read and write CP/M files from C. It would be great to have that capability in the z88dk CP/M new library.

But as you say, there's a lot that can be done with this source.

zx70, 2018-09-17:
https://github.com/arpruss/ozdev/tree/master/libsrc

zx70, 2018-09-17:
IIRC part of the earlier fcntl lib inspired Dominic in the z88dk CP/M support, which I later improved by picking ASM portions from BDS C.

zx70, 2018-09-17:
I think the char type wasn't very sophisticated. 

https://github.com/arpruss/ozdev/blob/master/libsrc/gen/isdigit.as

aralbrec, 2018-09-17:
The c library requires those "chars" to be int type so you can't read too much into that.  There may be rules for passing values by register but I'm at least not aware of any formal documentation.

feilipu, 2018-09-18:
I find it interesting that the ZAS assembler distributed with Hi-Tech C CP/M-80 chokes on Z180 mnemonics. There are a number of relevant pages blanked out in all of the user manuals. I think something was going on here.

But, for interest some example code.

```C
#include <stdio.h> 
#include <stdlib.h>
  
int main() 
{ 
    FILE *fptr; 
  
    char filename[100], c; 
  
    printf("Enter the filename to open \n"); 
    scanf("%s", filename); 
  
    /* Open file */
    fptr = fopen(filename, "r"); 
    if (fptr == NULL) 
    { 
        printf("Cannot open file \n"); 
        exit(0); 
    }
#asm
testing:
halt
out (c),a
mlt de
#endasm
  
    /* Read contents from file */
    c = fgetc(fptr); 
    while (c != EOF) 
    { 
        printf ("%c", c); 
        c = fgetc(fptr); 
    } 
  
    fclose(fptr); 
    return 0; 
}
```

When compiled produces this z80 assembly code.
The compiler produces forward `nnf` and backwards `nnb` labels to ease relative jumps and references to string bytes.
The resultant code (with the testing: junk removed) works perfectly on a CP/M target.

```
B>TYPE PRNTFILE.AS

;stdio.h: 12: extern    struct  _iobuf {
;stdio.h: 13:   char *          _ptr;
;stdio.h: 14:   int             _cnt;
;stdio.h: 15:   char *          _base;
;stdio.h: 16:   unsigned char           _flag;
;stdio.h: 17:   char            _file;
;stdio.h: 18: } _iob[   8];
;stdio.h: 66: extern int        fclose( struct _iobuf *);
;stdio.h: 67: extern int        fflush( struct _iobuf *);
;stdio.h: 68: extern int        fgetc(  struct _iobuf *);
;stdio.h: 69: extern int        ungetc(int,     struct _iobuf *);
;stdio.h: 70: extern int        fputc(int,      struct _iobuf *);
;stdio.h: 71: extern int        getw(   struct _iobuf *);
;stdio.h: 72: extern int        putw(int,       struct _iobuf *);
;stdio.h: 73: extern char *     gets(char *);
;stdio.h: 74: extern int        puts(char *);
;stdio.h: 75: extern int        fputs(char *,   struct _iobuf *);
;stdio.h: 76: extern int        fread(void *, unsigned, unsigned,       struct _iobuf *);
;stdio.h: 77: extern int        fwrite(void *, unsigned, unsigned,      struct _iobuf *);
[... 174 lines cut ...]
add     hl,sp
ld      sp,hl
;PRNTFILE.C: 37:     return 0; 
ld      hl,0
jp      l2
;PRNTFILE.C: 38: }
l2:
jp      cret
f45     equ     -103
psect   data
19:
defb    69,110,116,101,114,32,116,104,101,32,102,105,108,101,110,97
defb    109,101,32,116,111,32,111,112,101,110,32,10,0
29:
defb    37,115,0
39:
defb    114,0
49:
defb    67,97,110,110,111,116,32,111,112,101,110,32,102,105,108,101
defb    32,10,0
59:
defb    37,99,0

B>
```

feilipu, 2018-09-18:
Page 16 of `hitech_c_compiler_z80_8086_manual.pdf` manual details the stack frame:

> Arguments to function
> Return address
> Saved IY
> Saved IX <--- IX points to here in the function.
> Local data
> <--- SP points to the end of the local data.

The text in the manual is:
> "All references to local data or parameters are made via IX. The first argument is located at (IX+6). Each parameter occupies at least 2 bytes. If a char is passed as an argument, it is expanded to int length. Local variables are accessed at locations (IX+-1) downward. If there is a register variable (there is at most one) it will be in register IY."

To be updated when I get access to the manual.

feilipu, 2018-09-19:
>  If there is a register variable (there is at most one) it will be in register `IY`.

Found an example of this in practice in the libraries.

In `fgetc.asm` this code calls `__filbuf` passing one (and only one) parameter via `IY`.

```
1:
	bit	_IOSTRG_BIT,(iy+flag)	;end of string?
	jr	nz,reteof		;yes, return EOF
	push	de			;save de
	push	iy			;pass iy as argument
	call	__filbuf		;refill the buffer
	ld	a,l			;the returned value
	pop	bc
	pop	de			;return address in de again
	bit	7,h
	jr	nz,reteof		;returned EOF
	jr	2b
```
And in `filbuf.c` the code looks like this.

```C
#include	<stdio.h>
_filbuf(f)
register FILE *	f;
{
	f->_cnt = 0;
	if(!(f->_flag & _IOREAD))
		return(EOF);
	if(f->_base == (char *)NULL) {
		uchar	c;
		f->_cnt = 0;
		if(read(fileno(f), &c, 1) == 1)
			return(c);
		f->_flag |= _IOEOF;
		return(EOF);
	}
	if((f->_cnt = read(fileno(f), f->_base, BUFSIZ)) <= 0) {
		if(f->_cnt == 0)
			f->_flag |= _IOEOF;
		else
			f->_flag |= _IOERR;
		return(EOF);
	}
	f->_ptr = f->_base;
	f->_cnt--;
	return((unsigned)*f->_ptr++);
}
```
I'm still not sure of the reasoning behind writing `fgetc()` in assembly "for speed", and yet having it call `_filbuf()` and `read()` both written in C.

C functions are initialised with `csv` and returned from with `cret`, which implements the previous description of the stack frame.
```
csv:	
	pop hl		;return address
	push iy
	push ix
	ld ix,0
	add ix,sp	;new frame pointer
	jp (hl)

cret:
	ld sp,ix
	pop ix
	pop iy
	ret
```

feilipu, 2018-09-29:
I've written some glue functions to map between the Hi-Tech C calling convention and the ZSDCC convention. The examples in this [yabios.c](https://github.com/feilipu/yaz180/blob/master/mpm21/mpmldr/yabios.c).

The `asm_*` function calls are [simply defined](https://github.com/feilipu/yaz180/blob/master/mpm21/mpmldr/YALIB.AS) as `EQU` entry points.

A bit laborious, but works well. The reverse (calling Hi-Tech C libraries from ZSDCC) would also be pretty straight forward, though perhaps not relevant.

feilipu, 2019-02-25:
Relevant for #120 as this source code could also support the fast 32 bit float implementation.

feilipu, 2019-06-07:
Informational Issue - closing because it can still be searched.

closed 2019-06-07 by feilipu