z88dk/z88dk #974: [c7420][bug] The produced binaries cannot be loaded

Fabrizio-Caruso, 2018-09-15:
The c7420 binaries may be malformed.
I follow the instructions in the doc:
I use the recommended emulator (I follow the link to get O2EM v1.20b5)
I run it with the recommended option ( -config=c7420)
I move the binary _a.bas into the basic directory.
I start the basis by pressing '0'.
I enter CLOAD"_a.bas"
The emulator then prints "Found _a<inverted ?>"

I am building a small binary (about 4k).

I am compiling with
```
	$(Z88DK_PATH)$(MYZ88DK) +c7420 \
	$(SCCZ80_OPTS) \
...
	-create-app \
	$(CROSS_LIB_PATH)/display/alt_print/memory_mapped_alt_print.c \
	$(TINY_FILES)
```
[image]

zx70, 2018-09-17:
It is the normal behavior.
Just keep waiting while the program is being loaded.

Fabrizio-Caruso, 2018-09-17:
@zx70 Thanks! After waiting for about 5 minutes for a 4800 byte file I got an error message. :-(

[image]

Fabrizio-Caruso, 2018-09-17:
@zx70  Not even a simple "hello world" can be run.
I compile the hello world example:
```
#include <stdio.h>
int main(void) {
    printf("Hello world! \n");
    while(1){};
    return 0;
}
```
by
```
	$(Z88DK_PATH)$(MYZ88DK) +c7420  _O3 \
	-lndos \
	$(SOURCE_PATH)/../experiments/hello.c \
	-create-app	
```
I place the "_A" inside the "basic" directory. I start the emulator with the c7420 option.
Then I load and run with
```
CLOAD"_A"
RUN
```
it just clears the screen and return an "OK" message.
If instead I list the BASIC program after CLOAD"_A", I see the program and then the screen flashes and the emulator freezes.

Fabrizio-Caruso, 2018-12-06:
@zx70 if the issue is confirmed, could you please flag this as "bug" in order to properly track it as a bug issue?

zx70, 2019-01-10:
Test is progressing:   I confirm that appmake is correct.
Using a strict loop in ASM it is possible to produce some output, so the program is loaded and run by the emulator.

I'm afraid we're in a situation similar to the X07 one, odd things happening, possibly getting worse due to  the emulators.
In the LIB sources for fputc_cons I found a note I left time ago:

.fputc_cons_native

; ODDLY THIS DOESN'T WORK !   (O2EM emulator problem ?)
;        ld      hl,2
;        add     hl,sp
;        ld      a,(hl)
		
	pop bc
	pop hl
	ld a,l
	push hl
	push bc