z88dk/z88dk #975: [s-os][question] Cannot load .obj file

Fabrizio-Caruso, 2018-09-15:
I cannot load and run the .obj file produced by z88dk.

I build a 4k binary with:
```
	$(Z88DK_PATH)$(MYZ88DK) +sos \
	-O3
	-lndos \
...
	-create-app \
	$(CROSS_LIB_PATH)/display/alt_print/memory_mapped_alt_print.c \
	$(TINY_FILES)
```
I don't know how to enter the Monitor in most emulators.
I can with mz800emu but
#<space> a.obj
does nothing
#L a.obj
opens a menu that only allows loading mzf images.

zx70, 2018-09-17:
s-os is an operating  system.   you need first to boot s-os, then load the z88dk generated program.
an s-os impldmentation including z z80 emulator exists and permits to avoid the hassle of transferring files on the target disk

Fabrizio-Caruso, 2018-09-17:
@zx70  I know. I enter the monitor but the obj file cannot be loaded under mz800emu. So I wonder what is wrong.

zx70, 2018-09-17:
I'm confused.   Are you providing the s-os mzf file or the z88dk obj?    In the latter case the mz800 monitor won't be able to load it:  s-os is mandatory.

Fabrizio-Caruso, 2018-09-17:
@zx70 I would like to load the obj file.

zx70, 2018-09-17:
I never tried with the MZ800, but I suppose you need an S-OS disk to boot.
The obj file should be copied onto it.

Fabrizio-Caruso, 2018-09-17:
I just would like to know how to test the s-os target in its generic brand-less format, i.e., obj

zx70, 2018-09-18:
try by  running the unix or windows port, I think it is  at the end of this section: http://www.retropc.net/ohishi/s-os/#3

Fabrizio-Caruso, 2018-09-21:
@zx70  THANKS! It works fine! One more target!

closed 2018-09-21 by Fabrizio-Caruso