z88dk/z88dk #979: Cannot open map file generating SNA files

jsanjose, 2018-09-17:
I'm using the latest "nighly-build" for Windows: z88dk-win32-20180917-dccd45d-12928.zip

Compiling with the option -create-app and -Cz"--sna" the compiler gives this error:

`zx: Error: Cannot open map file C:\Users\SANJOS~1.map`

The command line is:

`zcc +zx -clib=sdcc_iy -startup=0 -m test.c -o test -create-app -Cz"--sna" -v`

Seems the error is because my user name is a Windows domain user: SANJOSE.DOMODEI. So the appmake utility trunks it when it builds the path to the map file:

`appmake --sna +zx  -b "test" -c "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90383"`

Following is the complete output from the compile command:

```
PROCESSING test.c
zsdcpp -iquote"." -D__Z88DK -D__SPECTRUM -DSPECTRUM -D__SPECTRUM__ -D__Z80 -DZ80 -D__SDCC -D__SDCC_IY -isystem"D:\Datos\JSJ\z88dk\lib\config\..\..\\include\_DEVELOPMENT\sdcc"  -D__SDCC  "test.c" "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90382.i2"
zpragma  < "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90382.i2" > "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90382.i"
zsdcc --constseg rodata_compiler   --reserve-regs-iy -mz80 --no-optsdcc-in-asm --c1mode --emit-externs  --no-c-code-in-asm   --no-peep --peep-file "D:\Datos\JSJ\z88dk\lib\config\..\..\/libsrc/_DEVELOPMENT/sdcc_peeph.2"  < "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90382.i" -o "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90382.opt"
copt "D:\Datos\JSJ\z88dk\lib\config\..\..\/libsrc/_DEVELOPMENT/sdcc_opt.1" < "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90382.op1" > "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90382.opt"
copt "D:\Datos\JSJ\z88dk\lib\config\..\..\/libsrc/_DEVELOPMENT/sdcc_opt.9" < "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90382.opt" > "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90382.op1"
copt "D:\Datos\JSJ\z88dk\lib\config\..\..\/libsrc/_DEVELOPMENT/sdcc_opt.2" < "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90382.op1" > "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90382.asm"
type "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90382.asm" >> "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90382.tmp"
type "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90382.tmp" >> "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90382.asm"
z80asm -I"D:\Datos\JSJ\z88dk\lib\config\..\..\\lib\target\zx\def" -D__SDCC_IY -I"D:\Datos\JSJ\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\target\zx" -D__SDCC  -s  --cpu=z80 -I"D:\Datos\JSJ\z88dk\lib\config\..\..\/lib" "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90382.asm"

PROCESSING D:\Datos\JSJ\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\target\zx\zx_crt.asm.m4
m4 -I "D:\Datos\JSJ\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\target\zx" -I "D:\Datos\JSJ\z88dk\lib\config\..\..\/src/m4"  --define=__STARTUP=0  < "D:\Datos\JSJ\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\target\zx\zx_crt.asm.m4" > "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90383"
type "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90383" >> "D:\Datos\JSJ\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\target\zx\zx_crt.asm"
type "D:\Datos\JSJ\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\target\zx\zx_crt.asm" >> "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90383.asm"
type "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90383.asm" >> "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90383.tmp"
type "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90383.tmp" >> "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90383.asm"
WILL ACT AS CRT

GENERATING OUTPUT

PROCESSING CRT
z80asm -I"D:\Datos\JSJ\z88dk\lib\config\..\..\\lib\target\zx\def" -D__SDCC_IY -I"D:\Datos\JSJ\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\target\zx" -D__SDCC  -s  --cpu=z80 -I"D:\Datos\JSJ\z88dk\lib\config\..\..\/lib" -I"D:\Datos\JSJ\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\target\zx"  "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90383.asm"

z80asm --cpu=z80 -b -d  -o"test" -m -s -L.  -L"D:\Datos\JSJ\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\lib\sdcc_iy"   -D__SDCC_IY -I"D:\Datos\JSJ\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\target\zx" -D__SDCC -izx    "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90383.o" "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90382.o"
type "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90383.map" >> "test.map"
appmake --sna +zx  -b "test" -c "C:\Users\SANJOS~1.DOM\AppData\Local\Temp\zcc90383"
zx: Error: Cannot open map file C:\Users\SANJOS~1.map
Building application code failed
```

suborb, 2018-09-17:
Thank you for the report and diagnosis.

Yes, the code looks forward for '.' when changing the extension. It should of course be searching from the end.

I'll get that fixed for the next nightly.

aralbrec, 2018-09-17:
As an aside the preferred way to select output type is via subtype with -Cz used to pass other appmake options:

zcc +zx -clib=sdcc_iy -startup=0 -m test.c -o test -create-app -subtype=sna -Cz"--clean" -v

The subtypes are defined in [zx.cfg](https://github.com/z88dk/z88dk/blob/master/lib/config/zx.cfg#L24) for the zx target.  Sometimes more options are necessary for a given output type and subtype ensures they are all set.  For sna "+zx --sna" must be passed to appmake.  In your compiler output I see +zx got into the appmake invoke although I am not sure how :) so it worked by passed just "--sna" to appmake.

jsanjose, 2018-09-19:
Perhaps i'm missing something but I've been downloading the last nightly (including the last one present today) and don't see the change applied. Or the change is wrong.

suborb, 2018-09-19:
It looks like I managed to misread both the code and the your report. Sorry.

So I replaced a while () loop with the strrchr so it's functionally the same which is why there's no change.

The actual issue is that the file in question doesn't have an extension but the path has a "." in it.

As a result, I think it's probably best to fix the zcc driver to pass through the complete filename, not the just the stub.

aralbrec, 2018-09-19:
It should be fixed in appmake too.  Searching for the "." separator should be a simple matter of looking at the basename only.  If you don't get around to it, I will have a look tonight.

aralbrec, 2018-09-20:
Let's try this fix.

commit 5012fe2c (2018-09-17): Fix #979: Search backwards for the extension
commit 190529a5 (2018-09-20): appmake: fix #979 look for file extension in basename part of path
closed 2018-09-20 by aralbrec