z88dk/z88dk #980: Porting Hi-Tech C 32 bit Float library to zxn & z180 - question

feilipu, 2018-09-17:
Reading the code discussed in #972, it looks quite simple (but still a lot of work), to convert the Hi-Tech C 32bit floating point libraries to the format required by SDCC, and use them within z88dk.

And, also it should be possible utilise the multiply hardware on the zxn and z180 to speed up [`eval_poly`](https://github.com/feilipu/yaz180/blob/master/docs/Hi-Tech_C_Libraries/FLOAT/evalpoly.c) and other routines, where the hardware is available.

It is an interesting task, since the Hi-Tech C float format also looks like that of the Am9511A-1 APU, except that the Hi-Tech C exponent is biased by 64. So, again it would be quite simple to patch in the APU, where it was available.

This looks like an interesting diversion for a few weeks. But first a question...

Where is the z88dk patching code to match the Gen48 floating point library into the required SDCC 32bit (nearly) IEEE-754 format? I've poked around, but I can't find it.

aralbrec, 2018-09-17:
The root of math48 is here:
https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/math/float/math48

`z80/core` contains the original math48 implementation with some bugfixes.

`z80` contains additional code for meeting what's expected in math.h.  These functions define the asm interface to math48 and still operate on floats in the format math48 defines.  They call into functions in `z80/core`.   Functions are prefaced with "am48_" for "asm math48".

`c` contains the interface expected by math.h and the c compilers.  These functions do the conversion to/from the float format expected by the compiers and call into the "am48_" code to carry out the operations.  All functions are namespaced with "cm48_".

As an example, [cm48_sdcciy_asin_fastcall.asm](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/math48/c/sdcc_iy/cm48_sdcciy_asin_fastcall.asm) executes "call am48_asin" to carry out the actual operation but this call is bracketed by functions that do the floating point conversion "cm48_sdcciyp_dx2m48" and "cm48_sdcciyp_m482d".  They contain "sdcc_iyp" in their names means sdcc_iy library primitive.  Some primitive functions in this directory also implement compiler float primitives expected by the compiler.

At this point math48 is namespaced so that it can exist in all target libraries without interfering with any other float implementations also in the target libraries.  It's possible to call several different float libraries in the same program.

The final link between the compiler and math48 is done in the `lm` directory.  The functions here connect the proper function names to the cm48_ and am48_ namespaced ones.  Everything here will just be defines.  These functions are compiled into a separate library "m.lib" so that linking to it on the compile line with "-lm" will connect the standard float names expected by the compiler to math48's names.  In this way you can select which float library you are using on the compile line.  The float functions themselves are already in every target's library.

Hopefully that can help a bit.  It's a big job to bring another float lib in and it's been a while since I did this one so the details are a bit foggy.

feilipu, 2018-09-17:
Thanks. That makes it clear what is to be done, and how to go about doing it.

The `cm48_sdcciyp_dx2m48` and `cm48_sdcciyp_m482d` functions and their peers are the ones I couldn'd find.

The Hi-Tech C compiler expects on operand on the stack, and the other in HLDE. So, there will be a bit of work to either a) flip it to DEHL' and DEHL, or b) write a shim in a similar manner to the functions above.

feilipu, 2018-09-29:
Found a source for the [Hi-Tech C Compiler Manual](https://github.com/z88dk/z88dk/issues/972#issuecomment-422372099).
Will update this issue once closer to actually doing something.

feilipu, 2019-02-25:
Related to #120.

feilipu, 2019-05-21:
Now in #1155

closed 2019-05-21 by feilipu