z88dk/z88dk #981: [rex][question][error in doc] Where can I get information on the APIs?

Fabrizio-Caruso, 2018-09-18:
The doc says that the rex targets has its own API and LibMain() is used instead of main.
This is not true in the only available example where the standard main() is used.

Where can I get information on how to read the keyboard input and 
print characters on give positions on the screen?

In the example I see DsPrintf(90,20,16,"Hello World");
what are the parameters used for? Is 90 the font size/type? 
I am looking at the Assembly code but I can't figure it out.
The prototypes just have types and no variable names which would have automatically told me how to use the functions.

Fabrizio-Caruso, 2018-09-18:
@suborb Thanks a lot!
I hope I manage to find a usable emulator.

Fabrizio-Caruso, 2018-09-18:
@suborb The doc in 
https://github.com/z88dk/z88dk/wiki/Platform---Xircom-REX6000
seems to still have a little error: LibMain() is not the main program entry from what I can see in the example.
It is not an important issue, though.

suborb, 2018-09-18:
The doc is right. If you want to build a library use libmain

Fabrizio-Caruso, 2018-09-21:
@suborb 
This seems to be the only target that lacks fput_cons.

suborb, 2018-09-26:
I managed to find those emulators - you missed have missed the message, anyway, I put them here: http://nightly.z88dk.org/misc/emulators/rex/ - I can't get them to work sensibly in wine though.

Regarding mame, it might be worth downloading a 3rd party one (there's some on rexextras) and see if they work..preferably one with source and then recompile and see if it still works.