z88dk/z88dk #983: [abc800][bug/question/feature] ABC800 status

Fabrizio-Caruso, 2018-09-21:
@zx70

ABC80SIM (https://www.zytor.com/pub/abc80/) now supports the ABC802.
It is possible to run .bac files with ABC80SOM.

I can build a hello world example (it is supposed to print and then loops forever on an empty while):
```
	$(Z88DK_PATH)$(MYZ88DK) +abc800 -Ca --reloc-info \
	-vn -lndos \
	$(SOURCE_PATH)/../experiments/hello.c
	perl $(TOOLS_PATH)/z88dk/abc80/bin2bac.pl -800 $(SOURCE_PATH)/../a.bin $(SOURCE_PATH)/../a.reloc > a.bac
```
(ABC800 requires relocatable binaries)

I then run abc80sim with:
abc80.exe --abc802 --memfile c:\Retro\CROSS-CHASE\src\a.bac

and run the basic program with:
run mem:

but nothing happens: nothing is printed and the computer returns to the blinking cursor as if it has exited from the program.

In order to exclude problems could you please implement the .ihx output for the ABC800 target?

zx70, 2018-09-21:
I tried to mock-up a 32 bit version from the sources but I only partially succeded.
One of the things I don't understand is how the parameters described in the help file are actually implemented (e.g. --802):

int main(int argc, char **argv)
{
    char  sysfile_name[256];
    FILE *sysfile;
    int   no_device = 0;
    int   c;

    while ((c = getopt(argc, argv, "bdv")) != EOF) {
        switch (c) {

          case 'v':
            printf("ABC80 emulator version %s\n", __version_string);
            exit(0);
            break;

          case 'b':
            no_basic = 1;
            break;

          case 'd':
            no_device = 1;
            break;

          default:
            usage();
            exit(1);
            break;
        }
    }

zx70, 2018-09-24:
I'm still having troubles in finding the updated sources.
Since we are at a "Release Candidate" stage of the emulator I'd wait for the sources to be released or for a 32bit version of the RC builds.

zx70, 2018-09-25:
My MinGw32 env is quite a mess, but I'm slowly proceeding.   One of the oddities I found is that configure.ac uses a folder named "aux" which is forbidden on Windows, but perhaps on a correctly set up machine this would have been automatically fixed.

zx70, 2018-09-26:
I have bad news for you, Fabrizio.   The ihx file support works on the ABC80 but not on the ABC800, you can test it yourself by typing PRINT PEEK (49200)

Fabrizio-Caruso, 2018-09-26:
@zx70  arggg!!! You mean that ABC80SIM does not accept ihx files when emulating the ABC800? The ABC800 requires relocatable binaries. Could this be the cause of the problem?

zx70, 2018-09-26:
[image]

Well, it does, but the ABC800 rom resets the memory on startup, while the ABC80 leaves it intact.
I hacked the sources to force a file reload every time a press a key, and as you can see it works.
Incredibly the same library I wrote for the ABC800 is works also on the ABC80, so it is possible to produce simple universal console programs for the ABC computers choosing the ABC800 target.

closed 2021-06-04 by Fabrizio-Caruso