z88dk/z88dk #984: [TI8X][bug/question] Cannot load the .8xp into an emulated TI83 Plus (Ion 1.5)

Fabrizio-Caruso, 2018-09-21:
@zx70 I am sorry but I need your help again.

I am compiling my game for the TI8X target with the suggested shell (default: ION) with:
```
	$(Z88DK_PATH)$(MYZ88DK) +ti8x \
...
	-vn -lndos \
	-create-app -oxchase \
	$(CROSS_LIB_PATH)/display/alt_print/memory_mapped_alt_print.c \
	$(FULL_FILES)
```
As suggested I use the xchase name to prevent name conflcts with the shell.

I use the recommended emulator: VTI 2.5 Beta 5 with the latest rom for the TI 83 Plus: 1.12.

I manage to load Ion 1.5 for TI83 Plus:
It displays: "Ion Installed"
(Here I have also tried to turn it off and on)
Now press on 2nd link
Drag the .8xp program and quickly press receive
(Here I have also tried to turn i off and on)
Now if I press on PRGM I see:
1: A
2: ION
3: IONZ
4: XCHASE
5: ZMINFO
6:\THETA ION

None of these options allows me to start the game/shell
XCHASE produces:
ERR:UNDEFINED
Trying to select the other options before or after does not help.

@zx70 are the Z88DK binaries only usable with specific rom versions?
I thought using a shell would prevent these problems?
Is there a specific rom version I need to use?
Is there a specific Ion shell version?

@zx70 This target may also be broken. Could you please test it and let us know?

zx70, 2018-09-21:
you must start the shell.  I expect it is the program named 'A', if it wasn't overwritten.
If the Shell exits immediately then it is possible that no valid programs were found.

Fabrizio-Caruso, 2018-09-21:
I know I must start the shell. I fear the binary is not good. None of the options seem to work.

Fabrizio-Caruso, 2018-09-23:
I confirm that no options work.
Moreover the more recent emulator Wabbitemu produces an error message "Invalid File Argument" when I try to transfer my .8xp files.

It looks like the files are really malformed or some non-obvious load operations are necessary.

I have tried to convert the raw binaries by using other tools such as brass.ese (something like devpac8x but more recent and 64-bit-friendly) and I get tons of errors (it cannot understand the content of the bin file).

We need at least one way to test binaries, possibly with a recent emulator.

Fabrizio-Caruso, 2018-09-23:
I confirm that Wabbitemu can digest other .8xp files but cannot take the .8xp files compiled by Z88DK.

zx70, 2018-09-24:
https://www.z88dk.org/forum/viewtopic.php?pid=7261
...this is probably the relevant chang which impacted on VTI.
I'll try to locally revert the configuration locally on my PC to check the VTI behavior and will let you know.

zx70, 2018-09-24:
Edited by Thibault Duponchelle (z88dk version not bin2var) because it
was buggy at least for 8xp (due to the port to z88dk probably or maybe
the bin2var  which was used was simply an old bin2var ressource).
I'm not sure if I haven't added some other issues but I tested it and
now it works for 8xp (so it's better than before).
In fact I just have used vimdiff to correct the bugs, and not checked 
the real meaning of each written byte/word but only checked some of them).
- Tested for ti84.
- Tested for ti83 regular (squished and unsquished).

zx70, 2018-09-24:
No luck, the binary looks still broken even if I revert appmake to the previous release

zx70, 2018-09-24:
Ok. I found the problem.
Now we need to define a solution.

For long time the TI 83 plus calculator's users claimed to have problems with the format produced by appmake.    The culript are 2 bytes that sometimes are added by the transfer/conversion tools, sometimes are missing.

If you are using ION, this is the patch for lib/target/ti83p/classic/ti83p_crt0.asm :

;-----------------
;1 - Ion (default)
;-----------------
IF !NOT_DEFAULT_SHELL
	DEFINE Ion
	;org	$9D95
	org	$9D93
	defb         $BB,$6D
	ret

In my opinion the best thing to do is to avoid the 2 extra bytes on all the ti83p formats and put an option in appmake to add them optionally.    I think they can be documented as binary format variant the users can try if in trouble..

zx70, 2018-09-24:
as said, I've put a new option in appmake to force the two extra bytes at the user's convenience, e.g.

zcc +ti8x -create-app -clib=ansi -subtype=ion -o tic -Cz--altfmt  tic.c

zx70, 2018-09-24:
Wiki docs updated

closed 2018-09-24 by zx70

Fabrizio-Caruso, 2018-09-25:
Wabbitemu still refuses to load the .8xp file (compiled with no options).
So the default compilation seems malformed.

Fabrizio-Caruso, 2018-09-25:
It seems we still have a problem

If I compile my game with  -subtype=ion -Cz--altfmt I can transfer the file to the emulator but the game cannot be started:

1. I drag the Ion shell file onto the emulator window
2. I get the message "ION Installed"
3. I drag the .8xp file  onto the emulator window
4. I press on PRGM
5. I see a list with 6 items where "amXCHASE" is the last one 
6. If I try to start "a", "Ion" I get nothing or errors
7. I cannot select "amXCHASE" and cannot start it with ASM(...

[image]

zx70, 2018-09-25:
[image]

zx70, 2018-09-25:
..runs on VTI with ROM v1.03

Fabrizio-Caruso, 2018-09-25:
It runs indeed on VTI (any rom version) but it fails on Wabbitemu.
Wabbitemu is a modern emulator and it is still maintained. VTI is abandomware.
Nevertheless, VTI could be more accurate than Wabbitemu.
We may still have an issue with the format if Wabbitemu is expecting the correct format and VTI is simply more forgiving...

This is significant progress because we can at least test programs. I have no idea if the format is correct. I will see if Wabbitemu's developers can figure this out...

zx70, 2018-09-25:
The story of the TI83 Plus file format on z88dk is incredibly tricky, but sadly I have no way to confirm what is the right format.   The tools itself was contributed by many people, everyone trying to improve it and solve the oddities, but generally speaking I'd trust the backup tools more than the emulators, be it VTI or Wabbitemu.